Ysurac/openmptcprouter #3804: Snapshot 0.62 bootloop after flashing

darkman1983, 2025-02-18:
## Expected Behavior
OMR boots up successfully

## Current Behavior
After a fresh flashing of 0.62 Snapshot from February 18, it complains about corrupted filesystem / disks on boot and goes into a bootloop.
Vrsion 0.61 is working fine here, even after a fresh flash.

## Possible Solution
You tell me!
Gone back to 0.61, as it still works.

## Steps to Reproduce the Problem

  1. Flash fresh 0.62 Snapshot
  2. Let it boot

## Context (Environment)
Can't get better screenshot as the debug output is scrolling soooo fast! Maybe doesn't contain any helpful information.
[image]

## Specifications

  - OpenMPTCProuter version: openmptcprouter-v0.62-snapshot-6.12-r0+28753-f480fa60d3-x86-64-generic-squashfs-combined-efi
  - OpenMPTCProuter VPS version: Version 0.1032-test 6.12.13-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: 1blue
  - OpenMPTCProuter platform: x86_64
  - Country: DE

Ysurac, 2025-02-18:
6.12 is still experimental, more than 6.6 snapshot.

darkman1983, 2025-02-18:
> 6.12 is still experimental, more than 6.6 snapshot.

As you might think from a Snapshot! But i had it already running in January!
So there has to be something corrupted now.
But hey, i can stop sending bugs, if you want to 😄

darkman1983, 2025-02-18:
To satisfy you, i've also tried with kernel 6.6 -> bootloop
Checking with live ubuntu system, GParted said: "GPT table corrupted" and also wanted to make filesystem checks.

Ysurac, 2025-02-18:
Strange... What image you tested ? squashfs or ext4 ?

darkman1983, 2025-02-18:
Same as with 6.12 kernel. See specifications provided by me.

darkman1983, 2025-02-19:
Hardware:

> 00:00.0 Host bridge: Intel Corporation Device 461c
00:02.0 VGA compatible controller: Intel Corporation Alder Lake-N [UHD Graphics]
00:14.0 USB controller: Intel Corporation Alder Lake-N PCH USB 3.2 xHCI Host Controller
00:14.1 USB controller: Intel Corporation Device 54ee
00:14.2 RAM memory: Intel Corporation Alder Lake-N PCH Shared SRAM
00:16.0 Communication controller: Intel Corporation Alder Lake-N PCH HECI Controller
00:17.0 SATA controller: Intel Corporation Alder Lake-N SATA AHCI Controller
00:1c.0 PCI bridge: Intel Corporation Device 54b9
00:1c.2 PCI bridge: Intel Corporation Device 54ba
00:1c.3 PCI bridge: Intel Corporation Device 54bb
00:1c.6 PCI bridge: Intel Corporation Device 54be
00:1d.0 PCI bridge: Intel Corporation Alder Lake-N PCI Express Root Port #10
00:1f.0 ISA bridge: Intel Corporation Alder Lake-N PCH eSPI Controller
00:1f.3 Audio device: Intel Corporation Alder Lake-N PCH High Definition Audio Controller
00:1f.4 SMBus: Intel Corporation Alder Lake-N SMBus
00:1f.5 Serial bus controller: Intel Corporation Alder Lake-N SPI (flash) Controller
01:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
02:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
03:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
04:00.0 Ethernet controller: Intel Corporation Ethernet Controller I226-V (rev 04)
05:00.0 Non-Volatile memory controller: Samsung Electronics Co Ltd NVMe SSD Controller 980 (DRAM-less)

Ysurac, 2025-03-15:
I think it's due to a driver, was already in a previous release so I removed many drivers but it's not a solution... Need to find which one have really an issue... One by one... I think a realtek module...

x3rx15, 2025-03-16:
Same Problem here, had to upgrade due to cloud vpn frontend update. 

OpenMPTCProuter version: openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined-efi
OpenMPTCProuter platform: x86_64
Country: DE

upgraded from openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5-x86-64-generic-ext4-combined-efi
additional driver for the 10G Intel NIC

Ysurac, 2025-03-16:
I'm compiling a new snapshot with less default drivers. Will be available in a few hours.

Ysurac, 2025-03-17:
It's compiled on https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/ can you try ?

Ysurac, 2025-03-17:
Yes, I removed all realtek drivers. But shouldn't impact 10GB intel if it was working in previous releases.
At least it's really a kernel divers that make bootloop. I will have to tests all one by one... Will take some times...

x3rx15, 2025-03-17:
That sounds semi good. kmod i40e still not there. probably easier to downgrade the vps side. next coffe is on me. Thanks for your work.

MatzDerTrainer, 2025-03-18:
Thank you for your efforts and your work.

MatzDerTrainer, 2025-03-18:
> It's compiled on https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/ can you try ?

I've tested this snapshot and haven't noticed any errors so far. All LAN interfaces are present.

lhc1993, 2025-03-19:
On which branch will this crash not occur now? I want to compile it myself. Also, I would like to ask how to easily upgrade branches locally without affecting local changes，tks!

Ysurac, 2025-03-19:
as always, on develop branch.

darkman1983, 2025-03-19:
Is the crash resolved?

darkman1983, 2025-03-25:
Still waiting for news :-D

Ysurac, 2025-03-25:
As already said here, snapshot image is not crashing but have less drivers. I will test one by one.

3dzug, 2025-03-31:
Same here. I'd really like to test the new image on my intel n100 machine.

Ysurac, 2025-04-01:
It's a joke or ?

3dzug, 2025-04-01:
No,It’s still looping on my intel N 100 computer. Matthias

3dzug, 2025-04-01:
If you want me something new to test let me knowMatthias

Ysurac, 2025-04-01:
So even with snapshot it's still not working ?

3dzug, 2025-04-01:
I tried a week ago. Shall I test again? Maybe I did something wrong on my end.Matthias

3dzug, 2025-04-02:
Hi, Snapshot v[0.63-snapshot-6.6 r0+28431-92e020b50f](https://github.com/ysurac/openmptcprouter) is working

darkman1983, 2025-04-27:
> Hi, Snapshot v[0.63-snapshot-6.6 r0+28431-92e020b50f](https://github.com/ysurac/openmptcprouter) is working

Still workin?

darkman1983, 2025-04-27:
Okay, i'm using it, it works! For other problems i've opened a issue 😄

patimages, 2025-05-07:
I confirm : the latest snapshot (6.6) works great on N100 !

darkman1983, 2025-05-07:
> I confirm : the latest snapshot (6.6) works great on N100 !

v6.6 or 6.12 Kernel, would be good to know :-)

Ysurac, 2025-05-07:
6.12 kernel is still an alpha version, should be better soon.

patimages, 2025-05-07:
> v6.6 or 6.12 Kernel, would be good to know :-)

It's written 😉 : 6.6

patimages, 2025-05-09:
correction : there are disconnection issues with the latest snapshot (6.6) on N100. Also, my orbi (distributing the mesh wifi at home) went burning. I went back to 0.61.

Ysurac, 2025-05-09:
If you want it to be solved, I would need the logs from Status->System log (both System and Kernel logs)

patimages, 2025-05-09:
I gathered but had to have a stable connexion quickly for work (Visio) so I forgot about it. The logs showed lots of connexions issues (cannot connect to server) at a random pace.

darkman1983, 2025-05-10:
I solved the disconnection issues bei setting keepalive from 7200 to 120, on 0.62+ i always had problems with disconnections, that solvd it for me!!

closed 2025-08-14 by github-actions[bot]