Ysurac/openmptcprouter #3807: VPS install script should be aware of kernel packages on check for broken packages

darkman1983, 2025-02-19:
In the current state of the install script it will end execution on broken packages, also if they're the kernel packages it installs. That could render the isntallation impossible, if you accidentally aborted the script in kernel package installing stage.
After that the script exit's at checking for broken dependencies and you will never be able to get it past that stage.
You have to manually download linux-headers and linux-image, if you have the url's for them and install them with dpkg your self.

The VPS-Install script should be aware of packages it installs itself, instead of abort on any dependencie problems.

Ysurac, 2025-02-19:
Current install script is the same since v0.61 release, you are the first issue.
Would be better with some log...

Ysurac, 2025-02-19:
Sorry, I misunderstood the issue.
Yes on broken package detection it fail, I will check if I can do something... not sure... In most case this need a manual operation and to remove broken packages.
It's more related to apt, Debian packet manager, than to my script.

darkman1983, 2025-02-19:
But the operation should be simply, reinstall kernel packages, so on headers and image, it can be ignored and the script can be run.

closed 2025-05-26 by github-actions[bot]