Ysurac/openmptcprouter #3809: Raspberry CM5 install openmptcprouter  error

yutian077, 2025-02-20:
when I boot form emmc  cm5  ，error  ！

373] usb usb1: New USB device found, idvendor=1d6b，idProduct=0002，bcdDevice=6.140] usb usb1: New USB device strings: Mfr=3,Product=2,SerialNunber=1360] usb usb1: Product: xHCI Host Controller
118] usb usb1: Manufacturer:Linux 6.6.50 xhci-hcd?98] usb usb1: SerialMumber: xhci-hcd.0597] hub 1-0:1.0:USB hub found624] hub 1-0:1.0: 2 ports detected
9921 usb usb2: New USB device found, idVendor=1d6b, idProduct=0003, bcdDevice= 6.5471 usb usb2: New USB device strings: Mir=3, Product=2,SerialNunber=1061] usb usb2: Product: xHCI Host Controller
210] usb usb2: Manufacturer: Linux 6.6.50 xhci-hcd878] usb usb2: SerialNumber: xhci-hcd.0(645] hub 2-0:1.0: USB hub found6661] hub 2-0:1.0: 1 port detected
9053] xhci-hcd xhci-hcd.1: xHCI Host Controller
5373] xhci-hcd xhci-hcd.1: new USB bus registered, assigned bus number 3
3305] xhci-hcd xhci-hcd.1: hcc parans 0x0240fe6d hci version 0x110 quirks 0x002559] xhci-hcd xhci-hcd.1: irq 138,io men exif003000008632] xhci-hcd xhci-hcd.1: xHCI Host Controller
3952] xhci-hcd xhci-hcd.1: new USB bus registered, assigned bus number 41458] xhci-hcd xhci-hcd.1: Host supports USB 3.0 SuperSpeed
7861] usb usb3: New USB device found, idvendor=1d6b， idProduct=0002，bcdDevice= 6.6419] usb usb3: New USB device strings: Mfr=3，Product=2,SerialNunber=13929] usb usb3: Product:xHCI Host Controller
9078 usb usb3: Manufacturer:Linux 6.6.50 xhci-hcd14748] usb usb3: SerialNumber: xhci-hcd.119528] hub 3-0:1.0:USB hub found23543] hub 3-0:1.0: 2 ports detected
27893] usb usb4: New USB device found, idVendor=1d6b，idProduct=0003，bcdDevice= 6.036445] usb usb4: New USB device strings: Mfr=3,Product=2,SerialNunber=143943] usb usb4: Product: xHCI Host Controller
49084] usb usb4: Manufacturer:Linux 6.6.50 xhci-hcd54745] usb usb4: SerialMumber: xhci-hcd.1595061 hub 4-0:1.0: USB hub found635211 hub 4-0:1.0: 1 port detected
678021 rpi-gpionen 1feeodoo00.gpiomen: window base exifee0d0000 size 0x0003000075854) rpi-gpionem ifeo0d0000.gpionen: initialised 1 regions as /deu/gpioneno
841091bcn2712-ionnu 1000005100.ionnu:bcn2712_ionnu_init:DEBUG_INPO =0x206047749926201 platforn 1000800000.codec: bcn2712_ionnu_probe_device: MHU 1000005100.ionmu00915] platforn 1000800000.codec: bcn2712_ionnu_device_group: MMU 1000005100.iommu092111 platforn 1000800000.codec: Adding to ionnu group θ
0153271 platforn 1000880000.pisp_be: bcm2712_ionnu_probe_device: nMU 1000005100.ionmu237971 platforn 1000880000.pisp_be: bcn2712_ionnu_device_group:MHU 1000005100.ionmu0322671 platforn 1000080000.pisp_be: Adding to ionnu group 0
0385631 platforn 1000800000.codec: bcn2712_ionnu_attach_dev:MM 1000005100.iomnu0466831 platforn 1000880000.pisp_be: bcn2712_ionnu_attach_deu: MhU 1000005100.ionmu0549761 bcn2712-ionu1000005100.iomnu:bcn2712_ionnu_probe: Success
.0622321 bcn2712-iu 1000005200.iomnu: bcn2712_iomnu_init: DEBUG_INFO = 0x20804774.0707631 bcm2712-10mmu 1000005200.iommu: bcn2712_ionnu_probe: Success
.0780231 bcn2712-u 1000005280.ionnu: bcn2712_ionnu_init: DEBUG_INFO = 0x20804774.0865381 bcn2712-ionu 1000005280.iommu: bcn2712_ionnu_probe:Success.0938671 mnc1:CQHCI uersion 5.10
.0978691 sdhci-brcnstb 1000ffr000.nnc: Got CD GPIO.098054] of _cfs_init
.0982691 nnc0:CQHCI version 5.10.098452] of_cfs_init:OK
1.112886] clk: Disabling unused clocks
3.1512281 nncO:SDHCI controller on 1000fff000.nnc【10eefffee0 .nnc] using ADHA 64-bit
3.3092341 nnc1:SDHCI controller on 1001100000.nnc[1001100000.nnc] using ADMA 64-bit
3.317711] Waiting for root device /deu/nncblk8p2..
3.3497461 mmc1: new ultra high speed DdRS0 SDI0 card at address 0001

## Specifications

  - OpenMPTCProuter version: https://releases.openmptcprouter.com/v0.61-6.6/rpi5/targets/bcm27xx/bcm2712/openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5-bcm27xx-bcm2712-rpi-5-ext4-factory.img.gz
  - 
what's wrong with this?

Ysurac, 2025-02-20:
I don't see error in log

yutian077, 2025-02-21:
Thank you for your reply。
The code stops here. When I plug in the USB keyboard, it prompts that the USB device is connected, but I cannot input any characters.
The device in question is a Raspberry Pi Compute Module 5 (CM5),  onboard eMMC 16G The image used : openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5-bcm27xx-bcm2712-rpi-5-ext4-factory.img.gz.

Ysurac, 2025-02-21:
You tried to press enter ? It answer to ping on 192.168.100.1 ?

yutian077, 2025-02-21:
Pressing the enter key has no response
ping192.168.100.1 Request timeout

yutian077, 2025-02-21:
openmptcprouter kernel is not support Raspberry CM5？

Ysurac, 2025-02-21:
It's supported: you can boot on it and it's not crashed.
Can you try snapshot image ? https://snapshots.openmptcprouter.com/6.6/rpi5/targets/bcm27xx/bcm2712/

yutian077, 2025-02-21:
I try to:
 openmptcprouter-v0.62-snapshot-6.6-r0+28431-92e020b50f-bcm27xx-bcm2712-rpi-5-ext4-factory.img.gz
openmptcprouter-v0.62-snapshot-6.12-r0+28753-f480fa60d3-bcm27xx-bcm2712-rpi-5-ext4-factory.img.gz

the same result！

Ysurac, 2025-02-21:
Can you try this image: https://downloads.openwrt.org/releases/24.10.0/targets/bcm27xx/bcm2712/

Ysurac, 2025-02-21:
True, there is all patches, DTS,... But CM5 DTS is not used for the image. I may try to add and you will have to test if you are ok ?

yutian077, 2025-02-21:
> True, there is all patches, DTS,... But CM5 DTS is not used for the image. I may try to add and you will have to test if you are ok ?
Okay, I'll wait for you to add it， I will test it！

yutian077, 2025-02-21:
True, there is all patches, DTS,... But CM5 DTS is not used for the image. I may try to add and you will have to test if you are ok ?
Okay, I'll wait for you to add it， I will test it！

Ysurac, 2025-02-21:
Can you test https://snapshots.openmptcprouter.com/6.6/rpi5/targets/bcm27xx/bcm2712/ ?

yutian077, 2025-02-22:
Thank！ for your work. It can now start normally. I will test the others,if there are any issues, I will  feedback again

Ysurac, 2025-02-22:
Thanks for the feedback

yutian077, 2025-02-25:
> Thanks for the feedback

Here's a new question about gpio! Can you fix it

yutian077, 2025-02-25:
> Thanks for the feedback
Here's a new question about gpio! Can you fix it
[](https://github.com/Ysurac/openmptcprouter/issues/3820)

yutian077, 2025-02-25:
https://github.com/Ysurac/openmptcprouter/issues/3820

yutian077, 2025-02-26:
> Can you try this image: https://downloads.openwrt.org/releases/24.10.0/targets/bcm27xx/bcm2712/

Hello, this one  Raspberry Pi CM5 device tree was incomplete. I contacted Raspberry Pi and updated it yesterday. Can you make it again！[
](https://github.com/raspberrypi/linux/tree/rpi-6.6.y/arch/arm64/boot/dts/broadcom)

update ： | bcm2712-rpi-cm5.dtsi | dts: Add hogs for RP1 GPIO 46/48 on CM5 | yesterday

Ysurac, 2025-02-27:
I added the patch, it's compiling (take about 8-10h)

yutian077, 2025-02-28:
> I added the patch, it's compiling (take about 8-10h)
thinks ,I need GPIO support, all GPIO are enable in kernel_menconfig or menuconfig?

closed 2025-06-03 by github-actions[bot]