Ysurac/openmptcprouter #3810: glinet  MT3000 ，Can you support it?

yunlai618, 2025-02-20:
There are many users using glinet 3000. Can it be supported? If I want to compile the firmware myself, is there a compilation tutorial?

Ysurac, 2025-02-20:
mt2500 and mt6000, I think I can try mt3000 now

Ysurac, 2025-02-20:
It's compiling, will be available in less than 7h

yunlai618, 2025-02-21:
Thank you very much. Where should I download it for glinet

Ysurac, 2025-02-21:
It's available here: https://snapshots.openmptcprouter.com/6.6/gl-mt3000/targets/mediatek/filogic/

mattfaraday, 2025-03-07:
omg this is super cool. I have the Beryl Ax and I've always wanted it to run OMR! 
are there any special instructions or can we just go via the upgrade option from within the Gl.iNet? 
Thankyou @Ysurac this is a very welcome addition!!

yunlai618, 2025-03-08:
yes  you can  update

closed 2025-06-11 by github-actions[bot]