Ysurac/openmptcprouter #3814: The phenomenon shown in the figure appears when requesting wan_addr

lhc1993, 2025-02-21:
## Expected Behavior

[image]

[image]

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

lhc1993, 2025-02-21:
and this

[image]

lhc1993, 2025-02-21:
I don't know what happened。

Ysurac, 2025-02-21:
Fixed.

closed 2025-02-21 by Ysurac