Ysurac/openmptcprouter #3816: Can't reliably ping OpenMPTCPRouter lan address when it is the default gateway

enricodetoma, 2025-02-23:
## Expected Behavior
I should be able to reliably ping the OpenMPTCPRouter LAN address regardless of whether it is the default gateway.

## Current Behavior
Setting the OpenMPTCPRouter LAN address as the default gateway makes the connection unreliable.
E.g. my OpenMPTCPRouter lan address is 192.168.148.1
If I use 192.168.148.1 as the default gateway and I start pinging 192.168.148.1, it pings for some seconds, then stops for some seconds (e.g. 30 seconds), then pings again, stops again, and so on. And of course, the internet connection doesn't work properly.
If I keep the LAN connection to the OpenMPTCPRouter but also connect to another connection (e.g., a Wi-Fi connection), and the other connection has the default gateway, or if I just remove the default gateway without any other connection, I can ping 192.168.148.1 continuously with no interruptions.
This behavior happens both with a Windows and a Linux client, so it doesn't appear to be a client problem.
This behavior is so strange that I don't really know where to start investigating: what does the default gateway setting have to do with the ability to ping the OpenMPTCPRouter LAN address reliably?

## Specifications

OpenMPTCProuter version: v0.61-6.6 r0+27346-c7ba5574f5
OpenMPTCProuter VPS version: 0.1031
OpenMPTCProuter VPS provider: Aruba (IT)
OpenMPTCProuter platform: x86_64

darkman1983, 2025-02-23:
I dunno if this is directly related. I've set OMR IP in UDM Pro as Internet Source for ping command, should be always green, but it isn't.

[image]

Maybe the LAN-Interface in OMR becomes unreachable, if VPN is down? VPN is often restarted on my OMR router.

Ysurac, 2025-02-23:
There is no firewall on OMR LAN interface and should always answer. Check in Status->Sytem log and Kernel log if there is any restart of LAN interface.

enricodetoma, 2025-02-23:
The problem is that the router reboots as soon as I set it as default gateway. How can I get a log before it reboots?

Ysurac, 2025-02-24:
You can try the latest snapshot 6.6: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

enricodetoma, 2025-02-24:
Latest snapshot 6.6 doesn't have r8169 driver which I need for my 4 ports PCI card for the WANs

Ysurac, 2025-02-24:
It's in latest snapshot

enricodetoma, 2025-02-24:
Never mind, I decided to use only the mainboard ethernet interface both for LAN and for WANs using a switch, and with the latest 6.6 snapshot it doesn't kernel panic any more.

closed 2025-02-24 by enricodetoma