Ysurac/openmptcprouter #3818: Request to add support for Teltonika RUTX12 router with dual modem

giuseppe64, 2025-02-23:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**

Yes, this request is related to the Teltonika RUTX12 router, which features dual modems that can be used simultaneously. The support for older versions is documented, but I would like to see support extended to the latest version.

**Describe why you are asking for this support?**

The Teltonika RUTX12 offers dual modem functionality, but I am seeking support for the latest versions of the device. Ideally, this support would be implemented through an installable package on the original Teltonika firmware, similar to how Bondix handles it, so that users do not need to upgrade the firmware.

**Additional context**

I am willing to test the functionality of this support if more detailed documentation is provided on how it works. This could help in verifying and improving the implementation.

Additionally, providing the support as an external package would avoid the need to create a custom firmware, making the solution more flexible and easier for users to implement.

Ysurac, 2025-02-23:
My RUTX12 is dead.
In any case, this can't be an installable package on Teltonika firmware (not the same UI, not same kernel, not same tools,...)

giuseppe64, 2025-02-23:
why is dead? so they are based on same firmware i think, so openwrt correct?

Ysurac, 2025-02-23:
The NAND is not working anymore, maybe due to maybe flash...
They are both based on OpenWRT, but on RUTXOS there is many change and they are using a very old release.

giuseppe64, 2025-02-23:
Thank you for your response. I understand that an installable package in the traditional sense may not be possible due to the differences in UI, kernel, and tools used by Teltonika firmware. However, I would appreciate it if you could consider alternatives such as:

A separate package that could be integrated into the existing firmware without requiring a full custom firmware build.

An implementation that works with the standard version of the firmware, without needing to perform a complete firmware upgrade.
I believe these solutions could provide flexibility for users and help enable the support for the dual modem feature on the Teltonika RUTX12.

Thank you for considering this, and I look forward to your thoughts on potential alternatives.

Ysurac, 2025-02-23:
It's not possible. At least Linux kernel 6.1 is needed for MPTCP to work correctly.
RUTXOS is really based on an old release of OpenWRT without many needed packages available at that time. So this would be lot of work for a platform with also an old and slow CPU.
Maybe when teltonika will update RUTXOS this will be possible.

closed 2025-05-30 by github-actions[bot]