Ysurac/openmptcprouter #3819: ss-manager: Could not bind

braidiano, 2025-02-24:
Hi,

on the VPS i have this error repeated many and many times:

Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 INFO: using tcp fast open
Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 INFO: using tcp no-delay
Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 INFO: using MPTCP
Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 INFO: running from root user
Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 INFO: working directory points to /root/.shadowsocks
**Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 INFO: try to bind interface: ::0, port: 65101
Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 ERROR: bind: Address already in use
Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 ERROR: Could not bind
Feb 24 09:38:47 mtcp-server ss-manager[253031]:  2025-02-24 09:38:47 ERROR: port is not available, please check.**

what that does means and what I have to check?

thanks

Ysurac, 2025-02-25:
Can you give me the result of ```netstat -laputen | grep 65101``` ?

braidiano, 2025-02-26:
`tcp        0      0 0.0.0.0:65101           0.0.0.0:*               LISTEN      0          6444       632/ss-server
tcp6       0      0 :::65101                :::*                    LISTEN      0          6441       632/ss-server
udp        0      0 0.0.0.0:65101           0.0.0.0:*                           0          6449       632/ss-server
udp6       0      0 :::65101                :::*                                0          6447       632/ss-server
`

now the error seems disappeard...

closed 2025-06-03 by github-actions[bot]