Ysurac/openmptcprouter #3821: the connection gets compleley broken when the master interface gets disconnected.

femrichem, 2025-02-25:
## Expected Behavior
Since OMR does loadbalancing as well as failover on connections it is expected when  the master connection gets disconnected, OMR switch to another connection and prevent the internet from getting blocked. however quite the opposite happens. 

as can be seen in the screenshot below as soon as the master interface runs into connection issues (in this case the TD-LTE modem froze and had to be manually restarted) the whole connection crumbles! what's more if other connections (non-master interfaces) run into similar issues (for example high latency _ over 900ms_ because the antenna is shaking due to wind) the connection slows down to a halt!! this is despite me having defined a BACKUP connection too (in this case it's an ADSL low speed line defined as backup). 
[image]

## Current Behavior
to make a long story short. the failover funtion of OMR does not work/has not worked ever since the installation. it just doesn't failover ___ I always have to manually go to the control panel of OMR and disable the faulty connection and manually change the master for my connection to get resorted !! whichc is EXtremely frustrating. if you are going to monitize this project in the future, which you should, this issue is a big obstacle on your way. I've been meaning to report this for a year now but I couldn't get around to do it. funnily enough previes versions (5.1 something I can't remmber did NOT HAVE THIS issue but 6.1 has it! which is strange) 

## Possible Solution
as for solutions, I haven't been able to come up with one since I am not familar with network programing and all. however I've written a script using node.js and puppeter to monitor OMR's status page and as soon as something goes wrong (which I think it refreshes every 3 or 4 seconds) change the master connection!! this is not a solution it's like putting a bandaid on  a bullet wound but I digress. 

## Steps to Reproduce the Problem

in order to reporduce the problem a couple of things need to be taken into consideration: 
1 - this issue does not occure if you compeletely disable the connection (for example turning off the modem, or disabling the interface) so you wouldn't be able to reporduce it this way 
2- this issue occurs when for example the modem freezes or your simcard runs out of credit or when for whatever reason the connection becomes too slow. in other words I don't think, to my knowledge and experenece, that there is a mechanism in OMR which monitors connection quality and disconnects or halts MPTCP from using it, which there should be! 

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

Ysurac, 2025-02-25:
Please complete specifications part of the template.
You can try snapshot, some fixes have been done when using backup multipath mode: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots
There is in Services->OMR-Tracker Manager, the possibility to check link quality based on latency, packet loss,...

Ysurac, 2025-02-25:
Maybe a bug, can you give the result of ```uci show omr-tracker``` ?

femrichem, 2025-02-26:
what should I do next ? do you need more info?

femrichem, 2025-02-26:
sure I'll get right on that, however I'm asking about omr-tracker? is my omr acting as it should? is there something I can tweak to get it to disconnect or ignore connections that are unstable based on predefined criteria untill they get back to normal?

Ysurac, 2025-02-26:
I make a fix in OMR-tracker, this may solve the issue.

femrichem, 2025-02-27:
[image]
I updated to the snapshot. I'll do more tests and report back.

closed 2025-06-03 by github-actions[bot]