Ysurac/openmptcprouter #3824: Request support for Cudy TR3000 1.0

RabbitJun, 2025-02-26:
Long time no see! I've just acquired a new product, the Cudy TR3000.
Its configuration is similar to the GL-MT3000, featuring an MT7981 CPU, 512MB DDR3 memory, and 128MB NAND flash.
Could you help me create a test firmware .bin file for it?
I'd really appreciate your assistance! :-)

Ysurac, 2025-02-28:
You can follow https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
You can also copy config-gl-mt3000 file to config-cudy-tr3000 and change also in this file glinet_gl-mt3000 to cudy_tr3000-v1 (if it's not a v1... good luck)

RabbitJun, 2025-02-28:
Thanks, I'll give it a try and see if it works.

RabbitJun, 2025-03-01:
I attempted manual compilation, manually selected the device model, and successfully generated a bin image file. However, the image failed to run on the tr3000—the device keeps rebooting repeatedly after powering on. After that, I tried directly flashing the mt3000's image onto the tr3000. Everything works fine except for the WAN port. I'm currently troubleshooting the issue. Theoretically, manual compilation should work, but it failed in this case.

[image]

RabbitJun, 2025-03-03:
The PHY driver for the 2.5G WAN port on the Cudy TR3000 is currently faulty, and there appears to be no immediate need for support at this time.Await subsequent driver updates, which may present opportunities to implement OMR on this device model.

closed 2025-03-03 by RabbitJun

alfindk, 2025-05-26:
hello RabbitJun. did you manage to compile an image for tr3000? i have the same device i want to make it on it openmptcprouter