Ysurac/openmptcprouter #3825: Hello. Tell me how to install OMR on RBM33G?

Zhuk77777, 2025-02-26:
Hello. Tell me how to install OMR on RBM33G?

Ysurac, 2025-02-26:
RBM3G use a slow CPU, same as Ubiquiti edgerouter-x that is supported but not working really well for me. There is also only 16MB flash.
So I will not add support for this platform.
You can compile yourself: https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform
You will have to remove some packages, due to 16MB flash.

Zhuk77777, 2025-02-28:
What exactly is not so good?
I need to combine two LTE channels. Now they are set to backup. Let's say if there is no Wi-Fi RBM 33, will this improve the situation?

Ysurac, 2025-02-28:
As I said: too slow CPU, too small flash. Nothing can improve that.