Ysurac/openmptcprouter #3827: TP-Link ER605 V2 Port Request

MikeKlem, 2025-03-01:
https://openwrt.org/toh/tp-link/er605_v2

- Five Gigabit Ethernet Ports
- Already runs OpenWRT <https://openwrt.org/releases/23.05.0>
- Relatively inexpensive <https://www.amazon.com/dp/B08QTXNWZ1>
- ramips architecture (similar to supported EdgeRouterX)

MikeKlem, 2025-03-01:
'''
[    0.000000] Linux version 5.15.137 (builder@buildhost) (mipsel-openwrt-linux-musl-gcc (OpenWrt GCC 12.3.0 r23619-101988c61a) 12.3.0, GNU ld (GNU Binutils) 2.40.0) #0 SMP Sun Nov 12 15:19:39 2023
[    0.000000] SoC Type: MediaTek MT7621 ver:1 eco:4
[    0.000000] printk: bootconsole [early0] enabled
[    0.000000] CPU0 revision is: 0001992f (MIPS 1004Kc)
[    0.000000] MIPS: machine is TP-Link ER605 v2
[    0.000000] Initrd not found or empty - disabling initrd
[    0.000000] VPE topology {2,2} total 4
[    0.000000] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.000000] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.000000] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000000000000-0x000000000fffffff]
[    0.000000] percpu: Embedded 11 pages/cpu s14736 r8192 d22128 u45056
[    0.000000] pcpu-alloc: s14736 r8192 d22128 u45056 alloc=11*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3 
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 64960
[    0.000000] Kernel command line: console=ttyS0,115200 noinitrd ubi.mtd=3,2048
[    0.000000] Dentry cache hash table entries: 32768 (order: 5, 131072 bytes, linear)
[    0.000000] Inode-cache hash table entries: 16384 (order: 4, 65536 bytes, linear)
[    0.000000] Writing ErrCtl register=00007260
[    0.000000] Readback ErrCtl register=00007260
[    0.000000] mem auto-init: stack:off, heap alloc:off, heap free:off
[    0.000000] Memory: 249108K/262144K available (7299K kernel code, 624K rwdata, 880K rodata, 1232K init, 226K bss, 13036K reserved, 0K cma-reserved)
[    0.000000] SLUB: HWalign=32, Order=0-3, MinObjects=0, CPUs=4, Nodes=1
[    0.000000] rcu: Hierarchical RCU implementation.
[    0.000000] 	Tracing variant of Tasks RCU enabled.
[    0.000000] rcu: RCU calculated value of scheduler-enlistment delay is 10 jiffies.
[    0.000000] NR_IRQS: 256
[    0.000000] clocksource: GIC: mask: 0xffffffffffffffff max_cycles: 0xcaf478abb4, max_idle_ns: 440795247997 ns
[    0.000004] sched_clock: 64 bits at 880MHz, resolution 1ns, wraps every 4398046511103ns
[    0.008058] Calibrating delay loop... 586.13 BogoMIPS (lpj=2930688)
[    0.066214] pid_max: default: 32768 minimum: 301
[    0.071629] Mount-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.078845] Mountpoint-cache hash table entries: 1024 (order: 0, 4096 bytes, linear)
[    0.090600] rcu: Hierarchical SRCU implementation.
[    0.096124] smp: Bringing up secondary CPUs ...
[    0.101590] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.101620] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.101634] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.101680] CPU1 revision is: 0001992f (MIPS 1004Kc)
[    0.160886] Synchronize counters for CPU 1: done.
[    0.193100] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.193127] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.193140] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.193172] CPU2 revision is: 0001992f (MIPS 1004Kc)
[    0.252286] Synchronize counters for CPU 2: done.
[    0.283072] Primary instruction cache 32kB, VIPT, 4-way, linesize 32 bytes.
[    0.283100] Primary data cache 32kB, 4-way, PIPT, no aliases, linesize 32 bytes
[    0.283113] MIPS secondary cache 256kB, 8-way, linesize 32 bytes.
[    0.283150] CPU3 revision is: 0001992f (MIPS 1004Kc)
[    0.337498] Synchronize counters for CPU 3: done.
[    0.367379] smp: Brought up 1 node, 4 CPUs
[    0.376619] clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[    0.386427] futex hash table entries: 1024 (order: 3, 32768 bytes, linear)
[    0.393513] pinctrl core: initialized pinctrl subsystem
[    0.400466] NET: Registered PF_NETLINK/PF_ROUTE protocol family
[    0.407139] thermal_sys: Registered thermal governor 'step_wise'
[    0.412983] FPU Affinity set after 11720 emulations
[    0.433508] clocksource: Switched to clocksource GIC
[    0.440024] NET: Registered PF_INET protocol family
[    0.445100] IP idents hash table entries: 4096 (order: 3, 32768 bytes, linear)
[    0.453345] tcp_listen_portaddr_hash hash table entries: 512 (order: 0, 6144 bytes, linear)
[    0.461670] Table-perturb hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.469378] TCP established hash table entries: 2048 (order: 1, 8192 bytes, linear)
[    0.477042] TCP bind hash table entries: 2048 (order: 2, 16384 bytes, linear)
[... 130 lines cut ...]
[   20.192635] device dsa entered promiscuous mode
[   20.197895] mt7530-mdio mdio-bus:1f eth1: configuring for phy/gmii link mode
[   20.207530] br-lan: port 1(eth1) entered blocking state
[   20.212843] br-lan: port 1(eth1) entered disabled state
[   20.222067] device eth1 entered promiscuous mode
[   20.256395] mt7530-mdio mdio-bus:1f eth2: configuring for phy/gmii link mode
[   20.265993] br-lan: port 2(eth2) entered blocking state
[   20.271287] br-lan: port 2(eth2) entered disabled state
[   20.278805] device eth2 entered promiscuous mode
[... cut ...]
[   20.351103] device eth4 entered promiscuous mode
[   20.376114] mtk_soc_eth 1e100000.ethernet eth0: PHY [mdio-bus:00] driver [MediaTek MT7530 PHY] (irq=POLL)
[   20.385829] mtk_soc_eth 1e100000.ethernet eth0: configuring for phy/rgmii link mode
[   24.768831] mt7530-mdio mdio-bus:1f eth1: Link is Up - 1Gbps/Full - flow control rx/tx
[   24.776870] br-lan: port 1(eth1) entered blocking state
[   24.782086] br-lan: port 1(eth1) entered forwarding state
[   24.788277] IPv6: ADDRCONF(NETDEV_CHANGE): br-lan: link becomes ready

'''

Ysurac, 2025-03-01:
Edgerouter X is really too slow, I don't think it's a good idea to add another device not really usable.