Ysurac/openmptcprouter #3830: v0.62rc1 script does not include kernel 6.12 support

mrkaffeine92, 2025-03-02:
Every time I attempt to run the instructions from v0.62rc1 VPS install:

You can download OpenMPTCprouter v0.62rc1 here: https://download.openmptcprouter.com/release/v0.62rc1-6.6/

You need to also update VPS script: `wget -O - https://www.openmptcprouter.com/server-rc/debian-x86_64.sh | KERNEL="6.12" sh`

The kernel is not supported. I am looking inside the file and there aren't any process paths for 6.12 neither.

Is this the wrong release upload? @Ysurac

Ysurac, 2025-03-02:
CDN cache wasn't purged. It's now working. rc1 is the same as current VPS snapshot script currently.

mrkaffeine92, 2025-03-03:
> CDN cache wasn't purged. It's now working. rc1 is the same as current VPS snapshot script currently.

Alot of issues with GL.iNet MT3000 (interfaces continue to shut down and restart, VPN not wanting to connect to VPS, extremely slow aggregation.

(Ran same profile as my RPI 4 where I got 700Mbps - on the newest RC MT3000 - 350 Mbps max).

mrkaffeine92, 2025-03-03:
> CDN cache wasn't purged. It's now working. rc1 is the same as current VPS snapshot script currently.

Latest build has been 4/4 times positive for becoming read-only after less than an hour of usage (using ext4).
` [base-manager] couldn't check support for device '/sys/devices/platform/soc/fe300000.mmcnr/mmc_host/mmc1/mmc1:0001/mmc1:0001:1': not supported by any plugin`
`Mon Mar  3 06:19:35 2025 daemon.notice procd: /etc/rc.d/S98omr-bypass: /etc/rc.common: eval: line 560: can't create /etc/firewall.omr-bypass: Read-only file system
Mon Mar  3 06:19:35 2025 daemon.info ModemManager[519]: hotplug: Checking if ModemManager is available...
Mon Mar  3 06:19:35 2025 daemon.info ModemManager[518]: hotplug: checking if ModemManager is available...
Mon Mar  3 06:19:35 2025 daemon.info ModemManager[519]: hotplug: ModemManager not yet available (count=26)
Mon Mar  3 06:19:35 2025 daemon.info ModemManager[518]: hotplug: ModemManager not yet available
Mon Mar  3 06:19:35 2025 daemon.notice procd: /etc/rc.d/S98omr-bypass: /etc/rc.common: line 560: can't create /etc/firewall.omr-bypass: Read-only file system
Mon Mar  3 06:19:36 2025 daemon.info ModemManager[519]: hotplug: Checking if ModemManager is available...
Mon Mar  3 06:19:36 2025 daemon.info ModemManager[519]: hotplug: ModemManager not yet available (count=25)
Mon Mar  3 06:19:36 2025 daemon.info ModemManager[518]: hotplug: checking if ModemManager is available...
Mon Mar  3 06:19:36 2025 daemon.info ModemManager[518]: hotplug: ModemManager not yet available
Mon Mar  3 06:19:37 2025 daemon.info ModemManager[519]: hotplug: Checking if ModemManager is available...
Mon Mar  3 06:19:37 2025 daemon.info ModemManager[519]: hotplug: ModemManager not yet available (count=24)
Mon Mar  3 06:19:37 2025 daemon.info ModemManager[518]: hotplug: checking if ModemManager is available...
Mon Mar  3 06:19:37 2025 daemon.info ModemManager[518]: hotplug: ModemManager not yet available
Mon Mar  3 06:19:37 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon Mar  3 06:19:38 2025 daemon.notice procd: /etc/rc.d/S98omr-bypass: /etc/rc.common: line 1000: can't create /usr/share/omr-bypass/omr-bypass-proto.lst: Read-only file system
Mon Mar  3 06:19:38 2025 daemon.notice procd: /etc/rc.d/S98omr-bypass: /etc/rc.common: eval: line 51: can't create /usr/share/omr-bypass/omr-bypass-proto.lst: Read-only file system`
`Mon Mar  3 06:19:25 2025 daemon.warn openvpn(omr)[12429]: Note: enable extended error passing on TCP/UDP socket failed (IP_RECVERR): Not supported (errno=95)`

Ysurac, 2025-03-03:
For MT3000, 350Mbps seems expected with this CPU. For interfaces shutdown and VPN issue I would need some log.
For ext4 i think it's on RPI, check squashfs.

closed 2025-03-04 by mrkaffeine92