Ysurac/openmptcprouter #3832: Glorytun never-ending errors in syslog

enricodetoma, 2025-03-03:
## Current Behavior
I have these never-ending errors in the syslog on the VPS:
```
2025-03-03T23:06:51.095411+01:00 admin166784342 ss-server[781]: failed to handshake with 91.80.29.87: authentication error
2025-03-03T23:06:51.579773+01:00 admin166784342 glorytun-tcp-run[623]: ::ffff:91.80.29.87.7386: connected
2025-03-03T23:06:51.579866+01:00 admin166784342 glorytun-tcp-run[623]: couldn't set socket option `TCP_USER_TIMEOUT'
2025-03-03T23:06:51.579886+01:00 admin166784342 glorytun-tcp-run[623]: peer sends a bad hash!
2025-03-03T23:06:51.579903+01:00 admin166784342 glorytun-tcp-run[623]: ::ffff:91.80.29.87.7386: key exchange failed
2025-03-03T23:06:52.660277+01:00 admin166784342 glorytun-tcp-run[623]: ::ffff:91.80.29.87.7305: connected
2025-03-03T23:06:52.660379+01:00 admin166784342 glorytun-tcp-run[623]: couldn't set socket option `TCP_USER_TIMEOUT'
2025-03-03T23:06:52.660399+01:00 admin166784342 glorytun-tcp-run[623]: peer sends a bad hash!
2025-03-03T23:06:52.660415+01:00 admin166784342 glorytun-tcp-run[623]: ::ffff:91.80.29.87.7305: key exchange failed
```

What can I do to remove them? 91.80.29.87 is not one of the wan addresses (maybe it was before).
I had to setup logrotate because the log filled all the available space on the filesystem.
Can I just disable IPv6 on the VPS?

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.62-snapshot-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: 6.6.36-x64v3-xanmod1 0.1031
  - OpenMPTCProuter VPS provider: Aruba (IT)
  - OpenMPTCProuter platform: x86_64)
  - Country: IT

Ysurac, 2025-03-06:
If it's not your of your IP, can be an IP scan. You can block the IP with ```shorewall drop 91.80.29.87```
Why disable IPv6 ? as it was enabled on installation all is configured with IPv6 support.

enricodetoma, 2025-03-06:
The point is: I don't think it's an IP scan. It is from my same provider, so I guess this was my previous dynamic IP.
For example, now I get this log:
```
2025-03-06T21:41:10.266051+01:00 admin166784342 ss-server[781]: failed to handshake with 37.159.54.129: authentication error
2025-03-06T21:41:11.306797+01:00 admin166784342 glorytun-tcp-run[623]: ::ffff:37.159.54.129.3491: connected
2025-03-06T21:41:11.306904+01:00 admin166784342 glorytun-tcp-run[623]: couldn't set socket option `TCP_USER_TIMEOUT'

```
and again, 37.159.54.129 is from exactly my same provider.
It is strange that I get continuous IP scans to my VPS, all from my VPN1 provider.
I also get "Looping ruote detected" as warning on VPN1 on the router status page.

AndreaLeidi, 2025-03-06:
Origina da te, secondo me il tuo software di trading online (Swiss Exchange?) da app o da un PC dietro il router sta chiamando un indirizzo IP appartenente al pool delle linee residenziali di Vodafone Italia (91.80.29.87) e gli passa la chiave di autenticazione RSA che viene ovviamente rifiutata. Dovresti forse ricontrollare la sua configurazione, OMR non c'entra.

enricodetoma, 2025-03-06:
@AndreaLeidi Non credo sia quello il motivo, perché ho appena provato a
bypassare completamente OMR andando su internet direttamente tramite la
WAN2 (adesso OMR quindi non ha più nessuno dietro di lui) e nonostante
quello, il log sulla VPS continua imperterrito.