Ysurac/openmptcprouter #3834: Can't get public IP from ShadowSocks Rust

mvmaksnet, 2025-03-06:
## Expected Behavior
It works but there is message on vps:  Can't get public IP from ShadowSocks Rust

## Current Behavior
Again like before one ISP can't work with multipath and it won't bond like other operators

## Specifications

  - OpenMPTCProuter version: OpenMPTCProuter v0.62rc1-6.6
  - OpenMPTCProuter VPS version: 0.1032-test 6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: Maksnet
  - OpenMPTCProuter platform:  x86_64

See picture:

[image]

Ysurac, 2025-03-06:
Please the country in the template can also be needed.
"The multipath seems..." is only a warning, if aggregation use this interface no problem, else it's really blocked.
The "Can't get public IP" is a problem. It's working when you change proxy in System,=->OpenMPTCProuter, "Wizard" tab, "Check advanced settings" checkbox ?

Ysurac, 2025-03-06:
When you tried what ? the checkbox only display more option and change nothing by itself.

mvmaksnet, 2025-03-06:
What should I do when go to advanced settings?

mvmaksnet, 2025-03-07:
I have set Shadowsock instead of shadowsock rust. Now everything is working perfectly.

Pajjja, 2025-03-07:
I have the same problem with ShadowSocks Rust - what are the exact instructions to fix the issue? In which place do I need to change the values?

    OpenMPTCProuter version: OpenMPTCProuter v0.62rc1-6.6
    OpenMPTCProuter VPS version: KERNEL="6.6"
    OpenMPTCProuter platform: x86_64

Ysurac, 2025-03-07:
I first need to find the issue...
What do you have in ```uci show shadowsocks-rust``` via SSH on the router (obfuscate the password if here)

Andrik-user, 2025-03-08:
r0+28431-92e020b50f)
------------------------------------------------------------------------------
-----------------------------------------------------
 PACKAGE:     openmptcprouter
 VERSION:     v0.62rc1
 TARGET:      rpi4
 ARCH:        aarch64_cortex-a72

 BUILD REPO:  https://github.com/ysurac/openmptcprouter
 BUILD DATE:  Fri Feb 28 16:36:02 UTC 2025
-----------------------------------------------------
root@OpenMPTCProuter:~# uci show shadowsocks-rust
shadowsocks-rust.hi1=ss_redir
shadowsocks-rust.hi1.server='sss0'
shadowsocks-rust.hi1.local_address='::'
shadowsocks-rust.hi1.local_port='1100'
shadowsocks-rust.hi1.mode='tcp_and_udp'
shadowsocks-rust.hi1.timeout='1000'
shadowsocks-rust.hi1.keep_alive='15'
shadowsocks-rust.hi1.fast_open='0'
shadowsocks-rust.hi1.verbose='0'
shadowsocks-rust.hi1.syslog='1'
shadowsocks-rust.hi1.reuse_port='0'
shadowsocks-rust.hi1.mptcp='1'
shadowsocks-rust.hi1.ipv6_first='1'
shadowsocks-rust.hi1.no_delay='1'
shadowsocks-rust.hi2=ss_redir
shadowsocks-rust.hi2.server='sss1'
shadowsocks-rust.hi2.local_address='::'
shadowsocks-rust.hi2.local_port='1101'
shadowsocks-rust.hi2.mode='tcp_and_udp'
shadowsocks-rust.hi2.timeout='1000'
shadowsocks-rust.hi2.keep_alive='15'
shadowsocks-rust.hi2.fast_open='0'
shadowsocks-rust.hi2.verbose='0'
shadowsocks-rust.hi2.syslog='1'
shadowsocks-rust.hi2.reuse_port='0'
shadowsocks-rust.hi2.mptcp='1'
shadowsocks-rust.hi2.ipv6_first='1'
shadowsocks-rust.hi2.no_delay='1'
shadowsocks-rust.ss_rules=ss_rules
shadowsocks-rust.ss_rules.disabled='0'
shadowsocks-rust.ss_rules.redir_tcp='all'
shadowsocks-rust.ss_rules.src_default='forward'
shadowsocks-rust.ss_rules.dst_default='forward'
shadowsocks-rust.ss_rules.local_default='forward'
shadowsocks-rust.ss_rules.server='sss0'
shadowsocks-rust.ss_rules.ifnames='lo' 'eth0' 'tun0'
shadowsocks-rust.sss0=server
shadowsocks-rust.sss0.disabled='0'
shadowsocks-rust.sss0.server='185.181.8.110'
shadowsocks-rust.sss0.server_port='65280'
shadowsocks-rust.sss0.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss0.obfs='0'
shadowsocks-rust.sss0.password='cmREp6HXyfl+Wvh8Dj689PLCl2JuKVPVXrS7Ff0wcmw=:TVEfUq31v/TIMezatKpUw7U4IH2eH41S2rMJsVMp8SU='
shadowsocks-rust.sss1=server
shadowsocks-rust.sss1.disabled='1'
shadowsocks-rust.sss1.server='185.181.8.110'
shadowsocks-rust.sss1.server_port='65280'
shadowsocks-rust.sss1.method='2022-blake3-aes-256-gcm'
shadowsocks-rust.sss1.obfs='0'
shadowsocks-rust.sss1.password='cmREp6HXyfl+Wvh8Dj689PLCl2JuKVPVXrS7Ff0wcmw=:TVEfUq31v/TIMezatKpUw7U4IH2eH41S2rMJsVMp8SU='
shadowsocks-rust.dns=ss_tunnel
shadowsocks-rust.dns.disabled='1'
shadowsocks-rust.dns.mode='tcp_and_udp'
shadowsocks-rust.dns.server='sss0'
shadowsocks-rust.dns.local_port='5353'
shadowsocks-rust.dns.forward_address='8.8.8.8'
shadowsocks-rust.dns.forward_port='53'
shadowsocks-rust.tracker_sss0=ss_local
shadowsocks-rust.tracker_sss0.server='sss0'
shadowsocks-rust.tracker_sss0.local_address='127.0.0.1'
shadowsocks-rust.tracker_sss0.local_port='1111'
shadowsocks-rust.tracker_sss0.mode='tcp_and_udp'
shadowsocks-rust.tracker_sss0.timeout='600'
shadowsocks-rust.tracker_sss0.keep_alive='15'
shadowsocks-rust.tracker_sss0.fast_open='0'
shadowsocks-rust.tracker_sss0.syslog='0'
shadowsocks-rust.tracker_sss0.reuse_port='1'
shadowsocks-rust.tracker_sss0.mptcp='1'
shadowsocks-rust.tracker_sss0.verbose='0'
root@OpenMPTCProuter:~#

Ysurac, 2025-03-08:
What VPS script did you use ? rc1 URL script or snapshot script ?
I updated shadowsocks-go in snapshot, maybe a problem here... Can you give me the result of ```journalctl -u shadowsocks-go``` on the VPS (about 10 latest lines) ?
And then, can you test rc1 script ?

Andrik-user, 2025-03-08:
[image]

VPS script: wget -O - https://www.openmptcprouter.com/server-rc/debian-x86_64.sh | KERNEL="6.12" sh
OpenMPTCprouter v0.62rc1 here: https://download.openmptcprouter.com/release/v0.62rc1-6.6/

Mar 08 16:11:03 mp12tcp systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:03 mp12tcp systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:03 mp12tcp shadowsocks-go[42167]: 2025-03-08T16:11:03.680-0500 FATAL Failed to load config {"confPath": "/etc/shadowsocks-go/server.json", "error": "json: unknown field \"listeners\""}
Mar 08 16:11:03 mp12tcp systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 16:11:03 mp12tcp systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 08 16:11:03 mp12tcp systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 524.
Mar 08 16:11:03 mp12tcp systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:03 mp12tcp systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:03 mp12tcp shadowsocks-go[42171]: 2025-03-08T16:11:03.931-0500 FATAL Failed to load config {"confPath": "/etc/shadowsocks-go/server.json", "error": "json: unknown field \"listeners\""}
Mar 08 16:11:03 mp12tcp systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 16:11:03 mp12tcp systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 08 16:11:04 mp12tcp systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 525.
Mar 08 16:11:04 mp12tcp systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:04 mp12tcp systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:04 mp12tcp shadowsocks-go[42207]: 2025-03-08T16:11:04.179-0500 FATAL Failed to load config {"confPath": "/etc/shadowsocks-go/server.json", "error": "json: unknown field \"listeners\""}
Mar 08 16:11:04 mp12tcp systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 16:11:04 mp12tcp systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 08 16:11:04 mp12tcp systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 526.
Mar 08 16:11:04 mp12tcp systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:04 mp12tcp systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Mar 08 16:11:04 mp12tcp shadowsocks-go[42243]: 2025-03-08T16:11:04.431-0500 FATAL Failed to load config {"confPath": "/etc/shadowsocks-go/server.json", "error": "json: unknown field \"listeners\""}
Mar 08 16:11:04 mp12tcp systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 08 16:11:04 mp12tcp systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 08 16:11:04 mp12tcp systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 527.
Mar 08 16:11:04 mp12tcp systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.

Ysurac, 2025-03-09:
ok, an error in the config.
Can you give me the result on the server of ```cat /etc/shadowsocks-go/server.json``` ?

Andrik-user, 2025-03-09:
root@mp12tcp:~# cat /etc/shadowsocks-go/server.json
{
    "servers": [
        {
            "name": "ss-2022",
            "protocol": "2022-blake3-aes-256-gcm",
            "tcpListeners": [
            {
                "network": "tcp",
                "address": ":65280",
                "fastOpen": false,
                "reusePort": false,
                "multipath": true
            }
            ],
            "enableTCP": true,
            "listenerTFO": true,
            "enableUDP": true,
            "mtu": 1500,
            "psk": "5Ry4ogZM5WabVthmUFsTBkevO07UlXkpT1OtWGb/RYI=",
            "uPSKStorePath": "/etc/shadowsocks-go/upsks.json"
        }
    ],
    "stats": {
        "enabled": true
    },
    "api": {
        "enabled": true,
        "debugPprof": false,
        "trustedProxies": [],
        "listeners": [
        {
                "network": "tcp",
                "address": "127.0.0.1:65279"
        }
        ]
    }
root@mp12tcp:~#

Ysurac, 2025-03-09:
Should be fixed in rc and snapshot script, you can run it again.

Andrik-user, 2025-03-09:
Now everything works.

mvmaksnet, 2025-03-10:
Should we update VPS to make SC RUST work as it should?

Ysurac, 2025-03-10:
What you mean ?

mvmaksnet, 2025-03-10:
Cant get public ip address for sc rust.
How to solve this?
Shadowssocks works but shadowsocks rust don't work.

Ysurac, 2025-03-10:
As it's indicated, should work with latest VPS script.

mvmaksnet, 2025-03-10:
You want to say I have to run again script on vps to solve issue?

Ysurac, 2025-03-10:
Yes. Same VPS script is used to install and update. Here you have to run it again to fixe issue.

mvmaksnet, 2025-03-11:
I have tested it.  IT WORKS NOW.
Thank you very much.

For one operator Yettel I was forced to enable mptcp over vpn and now this provider work in bonding like other operaors.

[image]

DENZAKIN, 2025-03-14:
I get the same error, how do I fix it? I tried snapshot, same error.

Ysurac, 2025-03-15:
Not the same error, can you copy ```/etc/shadowsocks-go/server.json``` from VPS here ?

DENZAKIN, 2025-03-15:
{
    "servers": [
        {
            "name": "ss-2022",
            "protocol": "2022-blake3-aes-256-gcm",
            "tcpListeners": [
            {
                "network": "tcp",
                "address": ":65280",
                "fastOpen": false,
                "reusePort": false,
                "multipath": true
            }
            ],
            "listenerTFO": true,
            "enableUDP": true,
            "mtu": 1500,
            "psk": "bjKkU/9JVBw1/zH8T2vBUyVK0VuigykRpUyB5ZFUcyw=",
            "uPSKStorePath": "/etc/shadowsocks-go/upsks.json"
        }
    ],
    "stats": {
        "enabled": true
    },
    "api": {
        "enabled": true,
        "debugPprof": false,
        "trustedProxies": [],
        "listen": "127.0.0.1:65279"
    }

Ysurac, 2025-03-15:
Update VPS script.

DENZAKIN, 2025-03-15:
I used the snapshot script and the latest standard script and got the same error.
debian9-x86_64.sh тому що ви його редагували нещодавно

Ysurac, 2025-03-15:
I would need same ```/etc/shadowsocks-go/server.json``` file and ```journalctl -u shadowsocks-go``` on the VPS.

DENZAKIN, 2025-03-17:
I was able to get all the green. :)