Ysurac/openmptcprouter #3837: Request for Orange Pi CM 4

comstant, 2025-03-07:
**Is this request related to a missing driver support for a particular hardware platform, SoC or board? Please describe.**
This request is for adding support for the Orange Pi Compute Module 4, a single-board computer (SBC) based on the Rockchip RK3566 SoC. The board features ARM Cortex-A55 CPU.

**Describe why you are asking for this support?**
Openmptcrouter has no official support for that router.

**Additional context**
http://www.orangepi.org/html/hardWare/computerAndMicrocontrollers/details/Orange-Pi-CM4-1.html

comstant, 2025-03-07:
Or is there anything I can do to help you speed the work up?

Ysurac, 2025-03-11:
There is no support in OpenWRT, I don't have the device and there is now a CM5 board. I don't think I will support it.