Ysurac/openmptcprouter #3839: Currently using version v0.62, but sometimes it restarts irregularly, usually about five or six hours

lhc1993, 2025-03-08:
Currently no useful information is obtained from the log, and the log is gone after the restart. Setting the remote IP or saving the log locally does not work.

router version：v0.62
vps version：6.6
please help，thanks！

## Expected Behavior

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

Ysurac, 2025-03-08:
Please, complete at least the template.

lhc1993, 2025-03-11:
Are you saying that you fixed it in your own compiled template, but the currently compiled version has not been fixed yet? Could you give some suggestions or explain the differences between your template and the current master?

Ysurac, 2025-03-11:
I'm saying that the issue template is not completed, please complete it.

lhc1993, 2025-03-11:
Currently no useful information is obtained from the log, and the log is gone after the restart. Setting the remote IP or saving the log locally does not work.

router version：v0.62
vps version：6.6
please help，thanks！

Expected Behavior：It can run stably without automatic restart.
Current Behavior：It will reboot itself in about 5 to 6 hours, and the reboot interval is also unstable.
Possible Solution：has no solution
Steps to Reproduce the Problem：No specific solution was identified.
Context (Environment)
Specifications
OpenMPTCProuter version:v0.62-snapshot-v0.61rc1
OpenMPTCProuter VPS version:6.6
OpenMPTCProuter VPS provider:aliyun
OpenMPTCProuter platform:AARCH64
Country:CN

Ysurac, 2025-03-11:
I would need the platform more than the arch.

Ysurac, 2025-03-18:
I think the reboot are on the router part only ? Did you try latest snapshot (without many drivers) ?