Ysurac/openmptcprouter #3841: Many files are created in the confige folder

Silvanskii, 2025-03-10:
## Expected Behavior

This bug has been seen in squashfs version on various hardware. If for example one hour equipment works and turn off the power and then turn on, it will turn on for 5-10 minutes and I looked through the UART is restoring files, which can be more than 10000. I did a broadcast and after 8 hours of work was created and inserted more than 23000 files, well there was a lot of memory and my script successfully deleted, but there were cases that the system did not start, because these files clogged all the memory.
Tell me what these files are and how I can get rid of them gradually, this is a real problem.
Here is the log from my script, it deletes hundreds of them every time.
`2025-03-06 05:59:12,795 - Delete 0 hidden files.
2025-03-06 06:09:44,365 - Delete 128 hidden files.
2025-03-06 08:39:09,660 - Executed sync command after waiting for 3 minutes.
2025-03-06 08:44:44,697 - Delete 0 hidden files.
2025-03-06 08:48:14,425 - Executed sync command after waiting for 3 minutes.
2025-03-06 08:55:27,438 - Delete 0 hidden files.
2025-03-06 10:12:07,726 - Delete 2932 hidden files.
2025-03-06 10:17:04,431 - Executed sync command after waiting for 3 minutes.
2025-03-06 10:35:16,257 - Delete 0 hidden files.
2025-03-06 10:49:29,439 - Delete 266 hidden files.
2025-03-10 03:35:28,493 - Executed sync command after waiting for 3 minutes.
2025-03-10 03:38:34,736 - Delete 0 hidden files.
2025-03-10 03:43:30,530 - Delete 28 hidden files.
2025-03-10 08:05:00,737 - Delete 10084 hidden files.
2025-03-10 08:13:32,204 - Executed sync command after waiting for 3 minutes.
2025-03-10 10:12:25,118 - Delete 0 hidden files.
2025-03-10 10:15:52,721 - Executed sync command after waiting for 3 minutes.
2025-03-10 11:12:45,337 - Delete 0 hidden files.
2025-03-10 11:29:18,450 - Delete 456 hidden files.
2025-03-10 11:45:01,304 - Delete 444 hidden files.
2025-03-10 11:52:29,859 - Delete 41 hidden files.
`

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 5.4, 6.1, 6.6
  - OpenMPTCProuter VPS version: i check old an new, bus has all vervison squashfs
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: i chec nanopi r4s, nanipi core and core2

Ysurac, 2025-03-10:
Strange, did you try another sd cards/disk for each ?

Silvanskii, 2025-03-10:
> Strange, did you try another sd cards/disk for each ?

I've tried several different memory cards, mostly working on internal memory with no flash card. It doesn't matter, I tried different devices and all versions have the same bug. And the device that I made 2 years ago on it was almost 100k of such hidden files, they are not visible on this did not notice before. All files are generated in the config folder and are hidden.

Ysurac, 2025-03-10:
In config file it's file where changes are made but not commited. Now it should have far less than before.

Silvanskii, 2025-03-10:
> In config file it's file where changes are made but not commited. Now it should have far less than before.

Do I need to change something on the server or on the device? So that fewer of these files are created.

Maybe I didn't put it right, these files are restored in large numbers only after power off, if I reboot or shut down via browser there are no hidden files. I asked chat gpt about it and gave him the whole log, he said it's a feature of the file system. I was wondering why so many files are generated, the longer I work the more of these files are recovered, sometimes it takes up to 10 minutes after powering off after a long time of working.

closed 2025-06-13 by github-actions[bot]