Ysurac/openmptcprouter #3842: The image for rpi4 is not compiled

dmkuzmin, 2025-03-11:
## Expected Behavior
I'm trying to build an image for raspberry pi 4 from the master branch using the 6.6 kernel

## Current Behavior
I do everything according to the instructions https://github.com/Ysurac/openmptcprouter/wiki/Create-image-for-unsupported-platform . I have a clean Debian 10 OS, downloaded the necessary packages, master the openmptcprouter branch and launched "OMR_TARGET="rpi4" OMR_FEED_SRC="master" OMR_KERNEL="6.6"./build.sh ". As a result, compilation stops at

...
 make[3] -C tools/dwarves clean-build
 make[3] -C tools/dwarves compile
 make[2] toolchain/compile
 make[3] -C toolchain/binutils compile
 make[3] -C toolchain/gcc/initial compile
    ERROR: toolchain/gcc/initial failed to build.
make -r world: build failed. Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on
make: *** [/home/username/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:233: world] Error 1

Ysurac, 2025-03-11:
"Please re-run make with -j1 V=s or V=sc for a higher verbosity level to see what's going on" in rpi4/6.6/source

dmkuzmin, 2025-03-11:
**The answer to make -j1 V=s**

```
make[5]: Entering directory '/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0-initial/gcc'
g++  -fno-PIE -c   -O2 -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include  -pipe   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -Wno-unused -DHAVE_CONFIG_H -I. -I. -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/. -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../include  -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libcpp/include -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libcody -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include  -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libdecnumber -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libbacktrace   -o gimple-match.o -MT gimple-match.o -MMD -MP -MF ./.deps/gimple-match.TPo gimple-match.cc

cc1plus: out of memory allocating 65536 bytes after a total of 881360896 bytes
make[5]: *** [Makefile:1158: gimple-match.o] Error 1
make[5]: Leaving directory '/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0-initial/gcc'
make[4]: *** [Makefile:4605: all-gcc] Error 2
make[4]: Leaving directory '/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0-initial'
make[3]: *** [Makefile:29: /home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0-initial/.built] Error 2
make[3]: Leaving directory '/home/username/openmptcprouter/rpi4/6.6/source/toolchain/gcc/initial'
time: toolchain/gcc/initial/compile#106.36#2.39#109.78
    ERROR: toolchain/gcc/initial failed to build.
make[2]: *** [toolchain/Makefile:93: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/username/openmptcprouter/rpi4/6.6/source'
make[1]: *** [toolchain/Makefile:89: /home/username/openmptcprouter/rpi4/6.6/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/username/openmptcprouter/rpi4/6.6/source'
make: *** [/home/username/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:233: world] Error 2
```

**The answer to make -j1 V=sc**

```
make[5]: Entering directory '/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0-initial/gcc'
g++  -fno-PIE -c   -O2 -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include  -pipe   -DIN_GCC  -DCROSS_DIRECTORY_STRUCTURE   -fno-exceptions -fno-rtti -fasynchronous-unwind-tables -W -Wall -Wno-narrowing -Wwrite-strings -Wcast-qual -Wmissing-format-attribute -Wconditionally-supported -Woverloaded-virtual -pedantic -Wno-long-long -Wno-variadic-macros -Wno-overlength-strings  -Wno-unused -DHAVE_CONFIG_H -I. -I. -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/. -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../include  -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libcpp/include -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libcody -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include -I/home/username/openmptcprouter/rpi4/6.6/source/staging_dir/host/include  -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libdecnumber -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libdecnumber/dpd -I../libdecnumber -I/home/username/openmptcprouter/rpi4/6.6/source/build_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/gcc-13.3.0/gcc/../libbacktrace   -o gimple-match.o -MT gimple-match.o -MMD -MP -MF ./.deps/gimple-match.TPo gimple-match.cc

cc1plus: out of memory allocating 65536 bytes after a total of 876945408 bytes
[... cut ...]
make[2]: *** [toolchain/Makefile:93: toolchain/gcc/initial/compile] Error 1
make[2]: Leaving directory '/home/username/openmptcprouter/rpi4/6.6/source'
make[1]: *** [toolchain/Makefile:89: /home/username/openmptcprouter/rpi4/6.6/source/staging_dir/toolchain-aarch64_cortex-a72_gcc-13.3.0_musl/stamp/.toolchain_compile] Error 2
make[1]: Leaving directory '/home/username/openmptcprouter/rpi4/6.6/source'
make: *** [/home/username/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:233: world] Error 2
```

Ysurac, 2025-03-11:
Seems to be an out of memory error, need more RAM or swap.

dmkuzmin, 2025-03-11:
@Ysurac , got it, thanks

Can you tell me which way is better to compile an image for a custom platform? Copy a ready-made platform and edit it for a custom one? Unfortunately, I don't fully understand the steps in the instructions. Is a "OMR_TARGET="myplatform" OMR_FEED_SRC="master" OMR_KERNEL="6.6"./build.sh" call required if there is no custom platform?

Ysurac, 2025-03-11:
"myplatform" will select the right config file if it exist and set the directory of the platform. OMR_FEED_SRC select the branch you want to use (develop/master) and OMR_KERNEL the Linux Kernel you want to use.
Then you can do exactly what you want.

dmkuzmin, 2025-03-11:
The config-myplatform file is required in the root of openmptcprouter, right?

Ysurac, 2025-03-11:
No, but without it you have to select arch via "make menuconfig"

dmkuzmin, 2025-03-11:
Increased RAM. Now I get this

```
# LD [M]  net/qrtr/qrtr-mhi.o
  aarch64-openwrt-linux-musl-ld  -EL  -maarch64elf -z noexecstack --no-warn-rwx-segments    -r -o net/qrtr/qrtr-mhi.o @net/qrtr/qrtr-mhi.mod
# cmd_modules_order net/qrtr/modules.order
  {   echo net/qrtr/qrtr.o;   echo net/qrtr/qrtr-tun.o;   echo net/qrtr/qrtr-mhi.o; :; } > net/qrtr/modules.order
# CC      net/socket.o
  aarch64-openwrt-linux-musl-gcc -Wp,-MMD,net/.socket.o.d  -nostdinc -I./arch/arm64/include -I./arch/arm64/include/generated  -I./include -I./arch/arm64/include/uapi -I                              ./arch/arm64/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -include                               ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -DKASAN_SHADOW_SCALE_SHIFT= -fmacro-prefix-map=./= -Werror -std=gnu11 -fshort-wchar -funsigned-char -fno-c                              ommon -fno-PIE -fno-strict-aliasing -mgeneral-regs-only -DCONFIG_CC_HAS_K_CONSTRAINT=1 -Wno-psabi -mabi=lp64 -fno-asynchronous-unwind-tables -fno-unwind-tables -mbranch                              -protection=pac-ret -Wa,-march=armv8.5-a -DARM64_ASM_ARCH='"armv8.5-a"' -DKASAN_SHADOW_SCALE_SHIFT= -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races -fst                              ack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -falign-functions=4 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-stack                              -check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security -Wno                              -trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling-poi                              nter -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-pointer-                              types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -Wno-                              format-truncation -Wno-stringop-overflow -Wno-stringop-truncation -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialized -W                              no-sign-compare -g -fmacro-prefix-map=/home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl=target-aarch64_cortex-a72_musl -fno-caller-                              saves -mstack-protector-guard=sysreg -mstack-protector-guard-reg=sp_el0 -mstack-protector-guard-offset=1072    -DKBUILD_MODFILE='"net/socket"' -DKBUILD_BASENAME='"socke                              t"' -DKBUILD_MODNAME='"socket"' -D__KBUILD_MODNAME=kmod_socket -c -o net/socket.o net/socket.c
# CC      net/sysctl_net.o
  aarch64-openwrt-linux-musl-gcc -Wp,-MMD,net/.sysctl_net.o.d  -nostdinc -I./arch/arm64/include -I./arch/arm64/include/generated  -I./include -I./arch/arm64/include/uap                              i -I./arch/arm64/include/generated/uapi -I./include/uapi -I./include/generated/uapi -include ./include/linux/compiler-version.h -include ./include/linux/kconfig.h -incl                              ude ./include/linux/compiler_types.h -D__KERNEL__ -mlittle-endian -DKASAN_SHADOW_SCALE_SHIFT= -fmacro-prefix-map=./= -Werror -std=gnu11 -fshort-wchar -funsigned-char -f                              no-common -fno-PIE -fno-strict-aliasing -mgeneral-regs-only -DCONFIG_CC_HAS_K_CONSTRAINT=1 -Wno-psabi -mabi=lp64 -fno-asynchronous-unwind-tables -fno-unwind-tables -mbr                              anch-protection=pac-ret -Wa,-march=armv8.5-a -DARM64_ASM_ARCH='"armv8.5-a"' -DKASAN_SHADOW_SCALE_SHIFT= -fno-delete-null-pointer-checks -O2 -fno-allow-store-data-races                               -fstack-protector -fno-omit-frame-pointer -fno-optimize-sibling-calls -fno-stack-clash-protection -falign-functions=4 -fstrict-flex-arrays=3 -fno-strict-overflow -fno-s                              tack-check -fconserve-stack -Wall -Wundef -Werror=implicit-function-declaration -Werror=implicit-int -Werror=return-type -Werror=strict-prototypes -Wno-format-security                               -Wno-trigraphs -Wno-frame-address -Wno-address-of-packed-member -Wframe-larger-than=2048 -Wno-main -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-dangling                              -pointer -Wvla -Wno-pointer-sign -Wcast-function-type -Wno-array-bounds -Wno-alloc-size-larger-than -Wimplicit-fallthrough=5 -Werror=date-time -Werror=incompatible-poin                              ter-types -Werror=designated-init -Wenum-conversion -Wno-unused-but-set-variable -Wno-unused-const-variable -Wno-restrict -Wno-packed-not-aligned -Wno-format-overflow -                              Wno-format-truncation -Wno-stringop-overflow -Wno-stringop-truncation -Wno-missing-field-initializers -Wno-type-limits -Wno-shift-negative-value -Wno-maybe-uninitialize                              d -Wno-sign-compare -g -fmacro-prefix-map=/home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl=target-aarch64_cortex-a72_musl -fno-cal                              ler-saves -mstack-protector-guard=sysreg - [... cut ...]
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make[1]: *** [target/Makefile:25: /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:233: world] Error 2
```

dmkuzmin, 2025-03-11:
There are about 80GB free on the disk

dmkuzmin, 2025-03-11:
I think I figured out what the problem was. I did "make -j2 V=s" to split into 2 threads, but apparently it was not applicable for me. I did "make -j1 V=s" and the build went on. The next problem is this:

```
make[3]: Entering directory '/home/debuser/openmptcprouter/feeds/openmptcprouter/mptcp'
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.prepared_b1c26ab213546bc3089e7101368614de_18f1e190c5d53547fed41a3eaa76e9e9_check
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.prepared_b1c26ab213546bc3089e7101368614de_18f1e190c5d53547fed41a3eaa76e9e9
rm -f /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.configured_*
rm -f /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.mptcp_installed
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.configured_68b329da9893e34099c7d8ad5cb9c940
rm -f /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.built
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.built_check
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.built
rm -rf /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.pkgdir/mptcp.installed /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.pkgdir/mptcp
mkdir -p /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.pkgdir/mptcp
install -d -m0755 /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.pkgdir/mptcp
cp -fpR ./files/* /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.pkgdir/mptcp/
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.pkgdir/mptcp.installed
mkdir -p /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/stamp
SHELL= flock /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/.root-copy.flock -c 'cp -fpR /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/.pkgdir/mptcp/. /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/'
touch /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/stamp/.mptcp_installed
rm -rf /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp
mkdir -p /home/debuser/openmptcprouter/rpi4/6.6/source/bin/targets/bcm27xx/bcm2711/packages /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/pkginfo
install -d -m0755 /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp
cp -fpR ./files/* /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp/
find /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/debuser/openmptcprouter/rpi4/6.6/source/scripts/strip-kmod.sh" PATCHELF="/home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/host/bin/patchelf" /home/debuser/openmptcprouter/rpi4/6.6/source/scripts/rstrip.sh /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp
install -d -m0755 /home/debuser/openmptcprouter/rpi4/6.6/source/bin/packages/aarch64_cortex-a72/openmptcprouter/tmp
mkdir -p /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp/CONTROL
(cd /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ] || exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
/home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/host/bin/fakeroot /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/host/bin/bash /home/debuser/openmptcprouter/rpi4/6.6/source/scripts/ipkg-build -m "" /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/mptcp/ipkg-all/mptcp /home/debuser/openmptcprouter/rpi4/6.6/source/bin/packages/aarch64_cortex-a72/openmptcprouter
[... cut ...]
    ERROR: package/kernel/bpf-headers failed to build.
make[2]: *** [package/Makefile:179: package/kernel/bpf-headers/compile] Error 1
make[2]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make[1]: *** [package/Makefile:173: /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:233: world] Error 2

```

Ysurac, 2025-03-13:
Seems that LLVM is not installed, select it in ```make menuconfig```

dmkuzmin, 2025-03-13:
[image]

By default, it means that the llvm of the host is used. LLVM is installed on the host, as it is present in the instructions in the package list. Do I need to choose "Build LLVM toolchain for eBPF"?

dmkuzmin, 2025-03-13:
```
$ dpkg -l | grep llvm
ii  libllvm7:amd64                    1:7.0.1-8+deb10u2             amd64        Modular compiler and toolchain technologies, runtime library
ii  llvm                              1:7.0-47                      amd64        Low-Level Virtual Machine (LLVM)
ii  llvm-7                            1:7.0.1-8+deb10u2             amd64        Modular compiler and toolchain technologies
ii  llvm-7-dev                        1:7.0.1-8+deb10u2             amd64        Modular compiler and toolchain technologies, libraries and headers
ii  llvm-7-runtime                    1:7.0.1-8+deb10u2             amd64        Modular compiler and toolchain technologies, IR interpreter
ii  llvm-runtime                      1:7.0-47                      amd64        Low-Level Virtual Machine (LLVM), bytecode interpreter
$ llvm-config --version
7.0.1
```

Ysurac, 2025-03-13:
Really too old

dmkuzmin, 2025-03-13:
I removed the llvm-7 package and installed llvm-13 - nothing has changed. I will try the "Build LLVM toolchain for eBPF" option or use Debian 12.

```
make[3]: Leaving directory '/home/debuser/openmptcprouter/feeds/openmptcprouter/mptcp'
time: package/feeds/openmptcprouter/mptcp/compile#0.18#0.08#0.23
make[2]: [package/Makefile:176: package/kernel/bpf-headers/compile] Error 2 (ignored)
make[3]: Entering directory '/home/debuser/openmptcprouter/rpi4/6.6/source/package/kernel/bpf-headers'
bash: --: invalid option
Usage:  bash [GNU long option] [option] ...
        bash [GNU long option] [option] script-file ...
GNU long options:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --pretty-print
        --rcfile
        --restricted
        --verbose
        --version
Shell options:
        -ilrsD or -c command or -O shopt_option         (invocation only)
        -abefhkmnptuvxBCHP or -o option
bash: line 0: [: : integer expression expected
/home/debuser/openmptcprouter/rpi4/6.6/source/include/bpf.mk:72: *** ERROR: LLVM/clang version too old. Minimum required: 12, found: .  Stop.
make[3]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source/package/kernel/bpf-headers'
time: package/kernel/bpf-headers/compile#0.02#0.00#0.03
    ERROR: package/kernel/bpf-headers failed to build.
make[2]: *** [package/Makefile:177: package/kernel/bpf-headers/compile] Error 1
make[2]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make[1]: *** [package/Makefile:171: /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:248: world] Error 2
$ exit
root@v2781821:~# dpkg -l | grep llvm
ii  libllvm13:amd64                   1:13.0.1-6~deb10u4            amd64        Modular compiler and toolchain technologies, runtime library
ii  libllvm7:amd64                    1:7.0.1-8+deb10u2             amd64        Modular compiler and toolchain technologies, runtime library
ii  llvm-13                           1:13.0.1-6~deb10u4            amd64        Modular compiler and toolchain technologies
ii  llvm-13-dev                       1:13.0.1-6~deb10u4            amd64        Modular compiler and toolchain technologies, libraries and headers
ii  llvm-13-linker-tools              1:13.0.1-6~deb10u4            amd64        Modular compiler and toolchain technologies - Plugins
ii  llvm-13-runtime                   1:13.0.1-6~deb10u4            amd64        Modular compiler and toolchain technologies, IR interpreter
ii  llvm-13-tools                     1:13.0.1-6~deb10u4            amd64        Modular compiler and toolchain technologies, tools
root@v2781821:~# llvm-config --version
-bash: /usr/bin/llvm-config: No such file or directory
root@v2781821:~#

```

dmkuzmin, 2025-03-14:
This is with option "Build LLVM toolchain for eBPF"

```
make[3]: Entering directory '/home/debuser/openmptcprouter/feeds/openmptcprouter/rust'
CARGO_HOME=/home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/cargo TARGET_CFLAGS="-Os -pipe -fno-caller-saves -fno-plt -fhonour-copts -fmacro-prefix-map=/home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/rust-1.81.0=rust-1.81.0 -Wformat -Werror=format-security -fstack-protector -D_FORTIFY_SOURCE=1 -Wl,-z,now -Wl,-z,relro" OPENWRT_RUSTC_BOOTSTRAP_CACHE=/home/debuser/openmptcprouter/rpi4/6.6/source/dl/rustc python3 /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/host/rustc-1.81.0-src/x.py --build-dir /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/host/rustc-1.81.0-src/build --config /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/host/rustc-1.81.0-src/config.toml dist build-manifest cargo llvm-tools rustc rust-std rust-src
Building bootstrap
    Finished `dev` profile [unoptimized] target(s) in 0.59s
WARNING: The `change-id` is missing in the `config.toml`. This means that you will not be able to track the major changes made to the bootstrap configurations.
NOTE: to silence this warning, add `change-id = 127866` at the top of `config.toml`
downloading https://ci-artifacts.rust-lang.org/rustc-builds/eeb90cda1969383f56a2637cbd3037bdf598841c/rust-dev-1.81.0-x86_64-unknown-linux-gnu.tar.xz
#=O#-   #      #
curl: (22) The requested URL returned error: 404
ERROR: failed to download llvm from ci

    HELP: There could be two reasons behind this:
        1) The host triple is not supported for `download-ci-llvm`.
        2) Old builds get deleted after a certain time.
    HELP: In either case, disable `download-ci-llvm` in your config.toml:

    [llvm]
    download-ci-llvm = false

Build completed unsuccessfully in 0:00:02
make[3]: *** [Makefile:110: /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/host/rustc-1.81.0-src/.built] Error 1
make[3]: Leaving directory '/home/debuser/openmptcprouter/feeds/openmptcprouter/rust'
time: package/feeds/openmptcprouter/rust/host-compile#0.53#0.35#3.06
    ERROR: package/feeds/openmptcprouter/rust [host] failed to build.
make[2]: *** [package/Makefile:175: package/feeds/openmptcprouter/rust/host/compile] Error 1
make[2]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make[1]: *** [package/Makefile:171: /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:248: world] Error 2
```

dmkuzmin, 2025-03-14:
This is happening on Debian 12. LLVM version 14

```
make[3]: Leaving directory '/home/debuser/openmptcprouter/feeds/openmptcprouter/mptcp'
time: package/feeds/openmptcprouter/mptcp/compile#0.18#0.06#0.23
make[2]: [package/Makefile:176: package/kernel/bpf-headers/compile] Error 2 (ignored)
make[3]: Entering directory '/home/debuser/openmptcprouter/rpi4/6.6/source/package/kernel/bpf-headers'
bash: --: invalid option
Usage:  bash [GNU long option] [option] ...
        bash [GNU long option] [option] script-file ...
GNU long options:
        --debug
        --debugger
        --dump-po-strings
        --dump-strings
        --help
        --init-file
        --login
        --noediting
        --noprofile
        --norc
        --posix
        --pretty-print
        --rcfile
        --restricted
        --verbose
        --version
Shell options:
        -ilrsD or -c command or -O shopt_option         (invocation only)
        -abefhkmnptuvxBCEHPT or -o option
bash: line 1: [: : integer expression expected
/home/debuser/openmptcprouter/rpi4/6.6/source/include/bpf.mk:72: *** ERROR: LLVM/clang version too old. Minimum required: 12, found: .  Stop.
make[3]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source/package/kernel/bpf-headers'
time: package/kernel/bpf-headers/compile#0.02#0.01#0.04
    ERROR: package/kernel/bpf-headers failed to build.
make[2]: *** [package/Makefile:177: package/kernel/bpf-headers/compile] Error 1
make[2]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make[1]: *** [package/Makefile:171: /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:248: world] Error 2
$ llvm-config --version
14.0.6
$ dpkg -l | grep llvm
ii  libllvm14:amd64                   1:14.0.6-12                             amd64        Modular compiler and toolchain technologies, runtime library
ii  libllvm15:amd64                   1:15.0.6-4+b1                           amd64        Modular compiler and toolchain technologies, runtime library
ii  llvm                              1:14.0-55.7~deb12u1                     amd64        Low-Level Virtual Machine (LLVM)
ii  llvm-14                           1:14.0.6-12                             amd64        Modular compiler and toolchain technologies
ii  llvm-14-dev                       1:14.0.6-12                             amd64        Modular compiler and toolchain technologies, libraries and headers
ii  llvm-14-linker-tools              1:14.0.6-12                             amd64        Modular compiler and toolchain technologies - Plugins
ii  llvm-14-runtime                   1:14.0.6-12                             amd64        Modular compiler and toolchain technologies, IR interpreter
ii  llvm-14-tools                     1:14.0.6-12                             amd64        Modular compiler and toolchain technologies, tools
ii  llvm-runtime:amd64                1:14.0-55.7~deb12u1                     amd64        Low-Level Virtual Machine (LLVM), bytecode interpreter

```

Ysurac, 2025-03-14:
With "Build LLVM toolchain for eBPF" you seems to go further but you used master branch, and as indicated Rust 1.81 package is not more available.
For llvm error, you don't have clang installed.

dmkuzmin, 2025-03-14:
Branch v0.61 is used with the option "Build LLVM toolchain for eBPF"

For Debian 12, I'm trying to install clang and repeat

Ysurac, 2025-03-14:
v0.61 will not compile due to rust. use v0.62 or develop

dmkuzmin, 2025-03-14:
Debian 12: installed clang, selected the v0.62 branch

```
make[5]: Entering directory '/home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.6.73'
  DESCEND bpf/resolve_btfids
  INSTALL libsubcmd_headers
  CC      net/strparser/strparser.o
  AR      net/strparser/built-in.a
  CC      net/8021q/vlan_core.o
  CC      net/8021q/vlan.o
  CC      net/8021q/vlan_dev.o
  CC      net/8021q/vlan_netlink.o
  CC      net/8021q/vlanproc.o
  AR      net/8021q/built-in.a
  AR      net/wireless/built-in.a
  AR      net/mpls/built-in.a
  CC [M]  net/mpls/mpls_gso.o
  CC [M]  net/mpls/af_mpls.o
  LD [M]  net/mpls/mpls_router.o
  CC [M]  net/mpls/mpls_iptunnel.o
  CC      net/switchdev/switchdev.o
  AR      net/switchdev/built-in.a
  CC      net/l3mdev/l3mdev.o
  AR      net/l3mdev/built-in.a
  CC      net/mptcp/protocol.o
  CC      net/mptcp/subflow.o
  CC      net/mptcp/options.o
  CC      net/mptcp/token.o
  CC      net/mptcp/crypto.o
  CC      net/mptcp/ctrl.o
  CC      net/mptcp/pm.o
  CC      net/mptcp/diag.o
  CC      net/mptcp/mib.o
  CC      net/mptcp/pm_netlink.o
  CC      net/mptcp/sockopt.o
  CC      net/mptcp/pm_userspace.o
  CC      net/mptcp/fastopen.o
  CC      net/mptcp/sched.o
  CC      net/mptcp/syncookies.o
  CC      net/mptcp/mptcp_diag.o
  CC      net/mptcp/bpf.o
  AR      net/mptcp/built-in.a
  CC      net/handshake/alert.o
  CC      net/handshake/genl.o
  CC      net/handshake/netlink.o
  CC      net/handshake/request.o
  CC      net/handshake/tlshd.o
  CC      net/handshake/trace.o
  AR      net/handshake/built-in.a
  CC [M]  net/tls/tls_main.o
  CC [M]  net/tls/tls_sw.o
  CC [M]  net/tls/tls_proc.o
  CC [M]  net/tls/trace.o
  CC [M]  net/tls/tls_strp.o
  CC [M]  net/tls/tls_device.o
  CC [M]  net/tls/tls_device_fallback.o
  LD [M]  net/tls/tls.o
  CC [M]  net/key/af_key.o
  CC [M]  net/appletalk/aarp.o
  CC [M]  net/appletalk/ddp.o
fixdep: error opening file: net/appletalk/.ddp.o.d: No such file or directory
make[9]: *** [scripts/Makefile.build:243: net/appletalk/ddp.o] Error 2
make[9]: *** Deleting file 'net/appletalk/ddp.o'
make[8]: *** [scripts/Makefile.build:480: net/appletalk] Error 2
make[7]: *** [scripts/Makefile.build:480: net] Error 2
make[6]: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.6.73/Makefile:1921: .] Error 2
make[5]: *** [Makefile:234: __sub-make] Error 2
make[5]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.6.73'
make[4]: *** [Makefile:34: /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/linux-6.6.73/.modules] Error 2
make[4]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source/target/linux/bcm27xx'
make[3]: *** [Makefile:12: compile] Error 2
make[3]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source/target/linux'
time: target/linux/compile#68.14#8.69#78.44
    ERROR: target/linux failed to build.
make[2]: *** [target/Makefile:32: target/linux/compile] Error 1
make[2]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make[1]: *** [target/Makefile:25: /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.target_compile] Error 2
make[1]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:233: world] Error 2
```

dmkuzmin, 2025-03-14:
Debian 10: The v0.62 branch is being used. I installed clang-13 and llvm-13 packages on top of the default packages clang-7 and llvm-7. Redirected the links /usr/bin/clang and /usr/bin/llvm to /usr/bin/clang-13 and /usr/bin/llvm-13

```
  HDRINST usr/include/asm/ipcbuf.h
  HDRINST usr/include/asm/bpf_perf_event.h
  HDRINST usr/include/asm/unistd_n64.h
  INSTALL /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/user_headers/include
make[4]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73'
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/.built
rm -rf /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers
mkdir -p /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers/host /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/packages
mkdir -p /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers/bpf-headers/arch /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers/bpf-headers/tools
cp -fpR /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/arch/mips /home/debuser/openmptcprouter/rp                   i4/6.6/source/tmp/stage-bpf-headers/bpf-headers/arch/
cp -fpR /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/tools/lib /home/debuser/openmptcprouter/rp                   i4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/tools/testing /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers/                   bpf-headers/tools/
cp -fpR /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/include /home/debuser/openmptcprouter/rpi4                   /6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/samples /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_corte                   x-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/scripts /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-heade                   rs/linux-6.6.73/user_headers /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers/bpf-headers
cp -fpR /home/debuser/openmptcprouter/rpi4/6.6/source/package/kernel/bpf-headers/files/stdarg.h /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers/bpf-headers/include
find /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers -name '*.la' | xargs -r rm -f;
if [ -f /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/packages/bpf-headers.list ]; then /home/debuser/openmptcprouter/rpi4/6.6/source/scripts                   /clean-package.sh "/home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/packages/bpf-headers.list" "/home/debuser/openmptcprouter/rpi4/6.6/source/st                   aging_dir/target-aarch64_cortex-a72_musl"; fi
if [ -d /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers ]; then (cd /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers; find ./ > /home/debuser/ope                   nmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers.files);      SHELL= flock /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/.staging-dir.flock -c ' mv /home/debuser/openmptcprouter/                   rpi4/6.6/source/tmp/stage-bpf-headers.files /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/packages/bpf-headers.list && cp -fpR /home/debuser/                   openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers/* /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/; '; fi
rm -rf /home/debuser/openmptcprouter/rpi4/6.6/source/tmp/stage-bpf-headers
touch /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.bpf-headers_installed
touch -r /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/.built /home/debuser/openmptcprouter/rpi4                   /6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73/.autoremove 2>/dev/null >/dev/null
find /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/bpf-headers/linux-6.6.73 -mindepth 1 -maxdepth 1 -not '(' -type f -and                    -name '.*' -and -size 0 ')' -and -not -name '.pkgdir'  -print0 | xargs -r -0 rm -rf
make[3]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source/package/kernel/bpf-headers'
time: package/kernel/bpf-headers/compile#35.28#18.31#48.44
make[3]: Entering directory '/home/debuser/openmptcprouter/feeds/openmptcprouter/mptcp-bpf-bkup'
touch /home/debuser/openmptcprouter/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/mptcp-bpf-bkup/.prepared_541629326a71a7ab02eb7882c643b19f_18f1e190c5d                   53547fed41a3eaa76e9e9_check
[... cut ...]
make[2]: *** [package/Makefile:179: package/feeds/openmptcprouter/mptcp-bpf-bkup/compile] Error 1
make[2]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make[1]: *** [package/Makefile:173: /home/debuser/openmptcprouter/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/stamp/.package_compile] Error 2
make[1]: Leaving directory '/home/debuser/openmptcprouter/rpi4/6.6/source'
make: *** [/home/debuser/openmptcprouter/rpi4/6.6/source/include/toplevel.mk:233: world] Error 2
```

Ysurac, 2025-03-14:
As you can see, there is a conflict between LLVM 13 and LLVM 7.
I don't understand how you can have a problem in fact, it's compiled for each change using this: https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml so quite the same as the doc.

dmkuzmin, 2025-03-19:
As a result, I took ubuntu 24.04, since ubuntu-latest is specified in the file https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml, and followed the following steps:
1) I installed the same packages as in the file https://github.com/Ysurac/openmptcprouter-feeds/blob/develop/.github/workflows/main.yml
2) OMR_KERNEL="6.6" OMR_FEED_SRC="develop" sh build.sh prepare tools/install -j1 V=s [SUCCESS]
3) OMR_KERNEL="6.6" OMR_FEED_SRC="develop" sh build.sh prepare toolchain/install -j1 V=s [SUCCESS]
4) make IGNORE_ERRORS=m -C rpi4/6.6/source package/compile -j1 V=s [FAIL]

I get a rust unpacking error.

```
mkdir -p /home/debuser/omr/rpi4/6.6/source/bin/targets/bcm27xx/bcm2711/packages /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2 /home/debuser/omr/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/pkginfo
mkdir -p /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2/etc/modules.d; ( echo "ovpn-dco-v2"; ) > /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2/etc/modules.d/30-ovpn-dco-v2;
find /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2 -name 'CVS' -o -name '.svn' -o -name '.#*' -o -name '*~'| xargs -r rm -rf
export CROSS="aarch64-openwrt-linux-musl-"  NO_RENAME=1 ; NM="aarch64-openwrt-linux-musl-nm" STRIP="/home/debuser/omr/rpi4/6.6/source/staging_dir/host/bin/sstrip -z" STRIP_KMOD="/home/debuser/omr/rpi4/6.6/source/scripts/strip-kmod.sh" PATCHELF="/home/debuser/omr/rpi4/6.6/source/staging_dir/host/bin/patchelf" /home/debuser/omr/rpi4/6.6/source/scripts/rstrip.sh /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2
rstrip.sh: /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2/lib/modules/6.6.73/ovpn-dco-v2.ko: relocatable
install -d -m0755 /home/debuser/omr/rpi4/6.6/source/bin/targets/bcm27xx/bcm2711/packages/tmp
mkdir -p /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2/CONTROL
(cd /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2/CONTROL; ( echo "$CONTROL"; printf "Description: "; echo "$DESCRIPTION" | sed -e 's,^[[:space:]]*, ,g'; ) > control; chmod 644 control; ( echo "#!/bin/sh"; echo "[ \"\${IPKG_NO_SCRIPT}\" = \"1\" ] && exit 0"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ]  exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_postinst \$0 \$@"; ) > postinst; ( echo "#!/bin/sh"; echo "[ -s "\${IPKG_INSTROOT}/lib/functions.sh" ]  exit 0"; echo ". \${IPKG_INSTROOT}/lib/functions.sh"; echo "default_prerm \$0 \$@"; ) > prerm; chmod 0755 postinst prerm;  )
/home/debuser/omr/rpi4/6.6/source/staging_dir/host/bin/fakeroot /home/debuser/omr/rpi4/6.6/source/staging_dir/host/bin/bash /home/debuser/omr/rpi4/6.6/source/scripts/ipkg-build -m "" /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2 /home/debuser/omr/rpi4/6.6/source/bin/targets/bcm27xx/bcm2711/packages
Packaged contents of /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/ipkg-aarch64_cortex-a72/kmod-ovpn-dco-v2 into /home/debuser/omr/rpi4/6.6/source/bin/targets/bcm27xx/bcm2711/packages/kmod-ovpn-dco-v2_6.6.73.0.2.20241216-r1_aarch64_cortex-a72.ipk
echo "kmod-ovpn-dco-v2" >> /home/debuser/omr/rpi4/6.6/source/staging_dir/target-aarch64_cortex-a72_musl/pkginfo/ovpn-dco.default.install
touch -r /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/.built /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216/.autoremove 2>/dev/null >/dev/null
find /home/debuser/omr/rpi4/6.6/source/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/ovpn-dco-0.2.20241216 -mindepth 1 -maxdepth 1 -not '(' -type f -and -name '.*' -and -size 0 ')' -and -not -name '.pkgdir'  -print0 | xargs -r -0 rm -rf
make[2]: Leaving directory '/home/debuser/omr/feeds/openmptcprouter/ovpn-dco'
time: package/feeds/openmptcprouter/ovpn-dco/compile#15.55#3.32#19.53
make[2]: Entering directory '/home/debuser/omr/feeds/openmptcprouter/rust'
mkdir -p /home/debuser/omr/rpi4/6.6/source/dl
SHELL= flock /home/debuser/omr/rpi4/6.6/source/tmp/.rustc-1.85.0-src.tar.gz.flock -c '          /home/debuser/omr/rpi4/6.6/source/scripts/download.pl "/home/debuser/omr/rpi4/6.6/source/dl" "rustc-1.85.0-src.tar.gz" "2f4f3142ffb7c8402139cfa0796e24baaac8b9fd3f96b2deec3b94b4045c6a8a" "" "https://static.rust-lang.org/dist/"    '
+ curl -f --connect-timeout 20 --retry 5 --location https://static.rust-lang.org/dist/rustc-1.85.0-src.tar.gz
[... cut ...]
    ERROR: package/feeds/openmptcprouter/rust [host] failed to build.
make[1]: *** [package/Makefile:177: package/feeds/openmptcprouter/rust/host/compile] Error 1
make[1]: Leaving directory '/home/debuser/omr/rpi4/6.6/source'
make: *** [/home/debuser/omr/rpi4/6.6/source/include/toplevel.mk:233: package/compile] Error 2
make: Leaving directory '/home/debuser/omr/rpi4/6.6/source'
```

_**Another option. I have made a fork of the required repositories - the build is successful.**_

I don't understand why difficulties arise when working with a VPS machine.

Ysurac, 2025-03-19:
You have a problem on file download. Try again.
Bad VPS here I think...

dmkuzmin, 2025-03-19:
I tried it three times and there is an error in the same place. There is a problem with unzipping. The file itself downloads 100% successfully. Maybe the file on the remote repository is corrupted.

I would like to understand why the chosen VPS is bad. I don't rule it out, but what should I pay attention to later?

closed 2025-06-23 by github-actions[bot]