Ysurac/openmptcprouter #3846: I frequently refresh this log, but I don't know what the problem is.

lhc1993, 2025-03-14:
Fri Mar 14 20:39:06 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:12 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:18 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:23 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:29 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:34 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:40 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:46 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:51 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:39:57 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:02 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:08 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:14 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:19 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:25 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:30 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:36 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:42 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:47 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:53 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:40:59 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:41:04 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:41:10 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 14 20:41:15 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )

Ysurac, 2025-03-14:
Wrong category and I need the template.

lhc1993, 2025-03-14:
router version: v0.62-snapshot-v0.62rc1
vpsversion:6.6.36-x64v3-xanmod1
platform: qualcommax ipq8074(New device compiled firmware)

lhc1993, 2025-03-15:
So what kind of classification error is this, and how should it be corrected?

Ysurac, 2025-03-15:
issue and use template.

closed 2025-03-21 by Ysurac