Ysurac/openmptcprouter #3848: No ipv6 default route after basic configuration

rdmitry0911, 2025-03-14:
## Expected Behavior

After enabling ipv6 in OMR Adv. Settings, disabling 6in4 and filling ipv6 addr/router for both wan1 and wan2 there is no default route in ip -6 ro show ta all
## Current Behavior

ip -6 ro show ta all:
```
default from fe80::a00:2 dev 6in4-omr6in4 table 61201 proto static metric 1201 pref medium
fe80::a00:2 dev 6in4-omr6in4 table 61201 proto static metric 1201 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 table 61201 proto static metric 1201 pref medium
2a0d:3344:3343:7114::/62 dev wan1 table 69 proto static metric 9 pref medium
fd24:16af:d717:4::/64 dev wan2 table 610 proto static metric 10 pref medium
fdee:39db:141f::/64 dev eth0 table 69999 proto static metric 9999 pref medium
unreachable fdee:39db:141f::/48 dev lo proto static metric 2147483647 pref medium
fe80::/64 dev eth1 proto kernel metric 256 pref medium
fe80::/64 dev wan1 proto kernel metric 256 pref medium
fe80::/64 dev wan2 proto kernel metric 256 pref medium
fe80::/64 dev eth0 proto kernel metric 256 pref medium
fe80::/64 dev 6in4-omr6in4 proto kernel metric 256 pref medium
local ::1 dev lo table local proto kernel metric 0 pref medium
anycast 2a0d:3344:3343:7114:: dev wan1 table local proto kernel metric 0 pref medium
local 2a0d:3344:3343:7114::100 dev wan1 table local proto kernel metric 0 pref medium
anycast fd24:16af:d717:4:: dev wan2 table local proto kernel metric 0 pref medium
local fd24:16af:d717:4::100 dev wan2 table local proto kernel metric 0 pref medium
anycast fdee:39db:141f:: dev eth0 table local proto kernel metric 0 pref medium
local fdee:39db:141f::1 dev eth0 table local proto kernel metric 0 pref medium
anycast fe80:: dev 6in4-omr6in4 table local proto kernel metric 0 pref medium
anycast fe80:: dev wan2 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth0 table local proto kernel metric 0 pref medium
anycast fe80:: dev wan1 table local proto kernel metric 0 pref medium
anycast fe80:: dev eth1 table local proto kernel metric 0 pref medium
local fe80::a00:2 dev 6in4-omr6in4 table local proto kernel metric 0 pref medium
local fe80::aff:ff02 dev 6in4-omr6in4 table local proto kernel metric 0 pref medium
local fe80::52ce:3aff:fe18:3e41 dev wan2 table local proto kernel metric 0 pref medium
local fe80::52dd:7dff:fe40:8f8c dev wan1 table local proto kernel metric 0 pref medium
local fe80::be24:11ff:fe66:c764 dev eth1 table local proto kernel metric 0 pref medium
local fe80::be24:11ff:fe77:3fa2 dev eth0 table local proto kernel metric 0 pref medium
multicast ff00::/8 dev eth1 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev wan1 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev wan2 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev eth0 table local proto kernel metric 256 pref medium
multicast ff00::/8 dev 6in4-omr6in4 table local proto kernel metric 256 pref medium

```
## Possible Solution

## Steps to Reproduce the Problem

  1.Install last OMR [from here](https://releases.openmptcprouter.com/v0.62-6.6/x86_64/targets/x86/64/openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-x86-64-generic-squashfs-combined.img.gz) 
  2.Make basic ipv6 config for both wan1 and wan2
  3.Try to ping6 google.com

## Context (Environment)

I have VPS with public ipv6 address and private ipv4 address. So it can be accessed from outside only via ipv6 

## Specifications

  - OpenMPTCProuter version:  openmptcprouter v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: Not used
  - OpenMPTCProuter VPS provider:Not used
  - OpenMPTCProuter platform: x86/x86_64
  - Country: Greece

Ysurac, 2025-03-15:
There is a bug on the interface on 6in4 disabling, try via SSH: ```uci set openmptcprouter.settings.disable_6in4="1" && uci commit openmptcprouter``` and to be sure reboot.

rdmitry0911, 2025-03-15:
> uci set openmptcprouter.settings.disable_6in4="1" && uci commit openmptcprouter

Didn't help. After reboot:
```
root@OpenMPTCProuter:~# ip -6 ro sh ta all|grep default
default from fe80::a00:2 dev 6in4-omr6in4 table 61201 proto static metric 1201 pref medium
default via fe80::a00:1 dev 6in4-omr6in4 table 61201 proto static metric 1201 pref medium
root@OpenMPTCProuter:~#
```

Ysurac, 2025-03-17:
I would need the result of ```uci show openmptcprouter.settings``` via SSH on the router please

Ysurac, 2025-03-20:
I'm not able to reproduce the issue, when 6in4 is disabled I don't have issue with local IPv6 connections.
You can try to delete omr6in4 interface in Network->Interfaces

closed 2025-07-18 by github-actions[bot]