Ysurac/openmptcprouter #3849: VPS install failing at iperf3

therestlesstexan, 2025-03-14:
## Expected Behavior

Full, complete installation on VPS (debian 12)

## Current Behavior

Install fails when you reach iperf3. See below relevant output:

Selecting previously unselected package libssl-dev:amd64.
(Reading database ... 74621 files and directories currently installed.)
Preparing to unpack .../libssl-dev_3.0.15-1~deb12u1_amd64.deb ...
Unpacking libssl-dev:amd64 (3.0.15-1~deb12u1) ...
Selecting previously unselected package libsctp1:amd64.
Preparing to unpack .../libsctp1_1.0.19+dfsg-2_amd64.deb ...
Unpacking libsctp1:amd64 (1.0.19+dfsg-2) ...
Selecting previously unselected package libsctp-dev:amd64.
Preparing to unpack .../libsctp-dev_1.0.19+dfsg-2_amd64.deb ...
Unpacking libsctp-dev:amd64 (1.0.19+dfsg-2) ...
Setting up libssl-dev:amd64 (3.0.15-1~deb12u1) ...
Setting up libsctp1:amd64 (1.0.19+dfsg-2) ...
Setting up libsctp-dev:amd64 (1.0.19+dfsg-2) ...
Setting up iperf3-build-deps (3.18-2) ...
Processing triggers for man-db (2.11.2-2) ...
Processing triggers for libc-bin (2.36-9+deb12u9) ...
Build iperf3 package....
Install iperf3 package...
-                              92%[==========================================>    ]  95.37K  5.94KB/s    in 16s     

Cannot write to ‘-’ (Broken pipe).

## Possible Solution

## Steps to Reproduce the Problem

  1. Spin up a fresh Debian 12 VM
  2. sudo to root
  3. Follow guide online to install VPS.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

therestlesstexan, 2025-03-14:
Just to make certain is was NOT my environment (I have a server colocated in a data center running ProxMox VE) I spun up a Debian 12 VM on VULTR and attempted the install there, I got the same result, fails at the same place.

Also attempted an ubuntu 20.04 box with same result - fails at iperf3.

Ysurac, 2025-03-14:
Should be fixed.

Ysurac, 2025-03-14:
It's here: https://github.com/Ysurac/openmptcprouter-vps/commit/3f3962b50f022b7762c898a744afdce46f035c9c
It the last changes: I forgot to correctly rename the iperf3 package due to iperf3 debian update.

therestlesstexan, 2025-03-14:
D'oh! LOL

Thanks for the quick fix!

mdp18, 2025-03-15:
same thing happens now with iperf 13.18-2 on BuyVM.

Ysurac, 2025-03-15:
@mdp18 what is the error ? Did you try new script ? What script are you using ?

TFleury, 2025-03-15:
It seems i'm facing the same issue on fresh install : Ubuntu 22.04.5 LTS on OVH VPS
The error I get is `mk-build-deps: You must have equivs installed to use this program.`

TFleury, 2025-03-15:
PR here : Ysurac/openmptcprouter-vps#104

Ysurac, 2025-03-15:
It's a new error, fixed now thanks. Ubuntu is not supported anymore, only Debian.

xNagatsu, 2025-03-19:
Same here on AWS t4g (ARM) : 

Build iperf3 package....
Install iperf3 package...
-                                                           92%[=============================================================================================================================>          ]  96.00K  11.5KB/s    in 8.3s    

Cannot write to ‘-’ (Broken pipe).

Ysurac, 2025-03-19:
I think it's only an error on ARM. I will test.

xNagatsu, 2025-03-19:
i've found a "fix" 

Line 591 
`dpkg -i iperf3_3.18-2_amd64.deb libiperf0_3.18-2_amd64.deb >/dev/null 2>&1`

i've change to 
`dpkg -i iperf3_3.18-2_arm64.deb libiperf0_3.18-2_arm64.deb >/dev/null 2>&1`

It's not clean but it's work

Ysurac, 2025-03-19:
Yes it was the issue, I fixed it in snapshot VPS script (with a "*")

closed 2025-06-23 by github-actions[bot]

Ysurac, 2026-01-13:
What parameters to the script you are using ? It should not compile it, only install Debian package

qsoMaster, 2026-01-14:
> Please use snapshot script: https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

All ok use snapshot script:[https://github.com/Ysurac/openmptcprouter/wiki/Snapshots](url)