Ysurac/openmptcprouter #3852: 0.62rc1  remove master still have some problem

fareign, 2025-03-15:
If  I removed master interface ,
the ESTAB link still is master.
wait for about a miniute, another interface come in . and the master  still at here.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: 0.62rc2
  - OpenMPTCProuter VPS version: 0.1032test
  - OpenMPTCProuter VPS provider: hk aliyun
  - OpenMPTCProuter platform: x86_64
  - Country: hk

fareign, 2025-03-15:
[image]
the streaming  is not stable

fareign, 2025-03-15:
another problem is  OPENVPN seems can't work.

Ysurac, 2025-03-15:
How you removed the interface ?
For OpenVPN, what do you have in Status->System log ?

fareign, 2025-03-15:
Unplug the network cable

fareign, 2025-03-15:
Sat Mar 15 06:32:28 2025 daemon.err openvpn(omr)[13105]: TCP: connect to [AF_INET]47.83.114.149:65301 failed: Operation timed out
Sat Mar 15 06:32:28 2025 daemon.notice openvpn(omr)[13105]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Sat Mar 15 06:32:29 2025 daemon.warn openvpn(omr)[13105]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sat Mar 15 06:32:29 2025 daemon.warn openvpn(omr)[13105]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Mar 15 06:32:29 2025 daemon.notice openvpn(omr)[13105]: TCP/UDP: Preserving recently used remote address: [AF_INET]47.83.114.149:65301
Sat Mar 15 06:32:29 2025 daemon.notice openvpn(omr)[13105]: Attempting to establish TCP connection with [AF_INET]47.83.114.149:65301
Sat Mar 15 06:32:49 2025 daemon.err openvpn(omr)[13105]: TCP: connect to [AF_INET]47.83.114.149:65301 failed: Operation timed out
Sat Mar 15 06:32:49 2025 daemon.notice openvpn(omr)[13105]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Sat Mar 15 06:32:50 2025 daemon.warn openvpn(omr)[13105]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sat Mar 15 06:32:50 2025 daemon.warn openvpn(omr)[13105]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Mar 15 06:32:50 2025 daemon.notice openvpn(omr)[13105]: TCP/UDP: Preserving recently used remote address: [AF_INET]47.83.114.149:65301
Sat Mar 15 06:32:50 2025 daemon.notice openvpn(omr)[13105]: Attempting to establish TCP connection with [AF_INET]47.83.114.149:65301
Sat Mar 15 06:33:01 2025 user.notice omr-schedule-010-services: Set firewall on server vps
Sat Mar 15 06:33:09 2025 daemon.err openvpn(omr)[13105]: TCP: connect to [AF_INET]47.83.114.149:65301 failed: Operation timed out
Sat Mar 15 06:33:09 2025 daemon.notice openvpn(omr)[13105]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Sat Mar 15 06:33:10 2025 daemon.warn openvpn(omr)[13105]: WARNING: No server certificate verification method has been enabled.  See http://openvpn.net/howto.html#mitm for more info.
Sat Mar 15 06:33:10 2025 daemon.warn openvpn(omr)[13105]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Mar 15 06:33:10 2025 daemon.notice openvpn(omr)[13105]: TCP/UDP: Preserving recently used remote address: [AF_INET]47.83.114.149:65301
Sat Mar 15 06:33:10 2025 daemon.notice openvpn(omr)[13105]: Attempting to establish TCP connection with [AF_INET]47.83.114.149:65301

fareign, 2025-03-15:
SS-rust can't get public IP

Ysurac, 2025-03-15:
On the VPS:
* ```journalctl -u openvpn@tun0```
* ```journalctl -u shadowsocks-go```

Make also sure that you are using server or server-test script and latest release (there is no 0.62rc2)

fareign, 2025-03-15:
rC1 ,all is the latest.
Mar 15 16:16:39 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:25295
Mar 15 16:16:39 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:39 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:40 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:25297
Mar 15 16:16:40 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:40 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:43 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:25299
Mar 15 16:16:43 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:43 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:45 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:25313
Mar 15 16:16:45 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:45 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:48 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:59771
Mar 15 16:16:48 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:48 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:49 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:59787
Mar 15 16:16:49 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:49 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:49 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:59791
Mar 15 16:16:49 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:49 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:50 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:59797
Mar 15 16:16:50 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:50 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:16:50 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:59807
Mar 15 16:16:50 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:16:50 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:17:03 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:14769
Mar 15 16:17:03 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:17:03 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected
Mar 15 16:17:23 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client connected from [AF_INET]127.0.0.1:43835
Mar 15 16:17:23 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: CMD 'status'
Mar 15 16:17:23 iZj6c1yoltwixb045jitd8Z ovpn-tun0[2295]: MANAGEMENT: Client disconnected

Mar 15 14:42:06 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 15 14:42:06 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 61.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z shadowsocks-go[832694]: 2025-03-15T14:42:07.103+0800 FATAL Failed to load config {"confPath": "/etc/shadowsocks>
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 62.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z shadowsocks-go[832778]: 2025-03-15T14:42:07.356+0800 FATAL Failed to load config {"confPath": "/etc/shadowsocks>
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 63.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.
Mar 15 14:42:07 iZj6c1yoltwixb045jitd8Z shadowsocks-go[832795]: 2025-03-15T14:42:07.606+0800 FATAL Failed to load config {"confPath": "/etc/shadowsocks>
[... cut ...]
Mar 15 14:42:08 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Main process exited, code=exited, status=1/FAILURE
Mar 15 14:42:08 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Failed with result 'exit-code'.
Mar 15 14:42:08 iZj6c1yoltwixb045jitd8Z systemd[1]: shadowsocks-go.service: Scheduled restart job, restart counter is at 66.
Mar 15 14:42:08 iZj6c1yoltwixb045jitd8Z systemd[1]: Stopped shadowsocks-go.service - shadowsocks-go Service.
Mar 15 14:42:08 iZj6c1yoltwixb045jitd8Z systemd[1]: Started shadowsocks-go.service - shadowsocks-go Service.

lhc1993, 2025-03-15:
I have the same problem. Have you solved it?

lhc1993, 2025-03-15:
Sat Mar 15 06:32:28 2025 daemon.err openvpn(omr)[13105]: TCP: connect to [AF_INET]47..11*.149:65301 failed: Operation timed out
Sat Mar 15 06:32:28 2025 daemon.notice openvpn(omr)[13105]: SIGUSR1[connection failed(soft),connection-failed] received, process restarting
Sat Mar 15 06:32:29 2025 daemon.warn openvpn(omr)[13105]: WARNING: No server certificate verification method has been enabled. See http://openvpn.net/howto.html#mitm for more info.
Sat Mar 15 06:32:29 2025 daemon.warn openvpn(omr)[13105]: NOTE: the current --script-security setting may allow this configuration to call user-defined scripts
Sat Mar 15 06:32:29 2025 daemon.notice openvpn(omr)[13105]: TCP/UDP: Preserving recently used remote address: [AF_INET]47.83.114.149:65301
Sat Mar 15 06:32:29 2025 daemon.notice openvpn(omr)[13105]: Attempting to establish TCP connection with [AF_INET]47.83.114.149:65301

lhc1993, 2025-03-15:
This issue occurs when OMR is connected and then disconnected, and the tun0 device cannot come up.

Ysurac, 2025-03-15:
For Shadowsocks-go, it should be solved in lastest VPS script (release or snapshot)