Ysurac/openmptcprouter #3853: v0.62 rc "Kernel Panic" - Fatal Exception

3dzug, 2025-03-15:
## Expected Behavior

The latest image should start as usual without reboot.

## Current Behavior

After flashing the latest release (also beta from 11.03.2025) on my intel N 100 minicomputer openmtcprouter fails to boot and reboots. Error Image is attached. 

## Possible Solution

Installing v.6.1 again fixed the issue 

## Steps to Reproduce the Problem

  1. Get a Minicomputer with intel N 100 chip
  2. Flash openmptcprouter-v0.62rc1-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined-efi.img.gz
  or

openmptcprouter-v0.62rc1-6.6-r0+28431-92e020b50f-x86-64-generic-squashfs-combined-efi.img.gz 

## Context (Environment)

I try to run the openmtcp sofftware.

## Specifications

  - OpenMPTCProuter version: 

  - OpenMPTCProuter VPS version: openmptcprouter-v0.62rc1-6.6-r0+28431-92e020b50f
  - OpenMPTCProuter VPS provider: 0.1031 6.6.36-x64v3-xanmod1

[image]

  - OpenMPTCProuter platform: x86_64
  - Country:  Germany

darkman1983, 2025-03-15:
I have N100 Hardware too, i also had this panic :-D

Ysurac, 2025-03-15:
Duplicate of #3804

closed 2025-03-15 by Ysurac