Ysurac/openmptcprouter #3855: Build fails on unsupported platform - GW Venice 7300

dekashio, 2025-03-16:
When building Image for Gatework Venice 7300 - unsupported platform (but openwrt supported) the build fails with the following error: 

1 error; 209 MiB in 1218 packages
make[2]: *** [package/Makefile:99: package/install] Error 1
make[2]: Leaving directory '/home/xxxx/projects/openmptcp/openmptcprouter/venice/6.12/source'
make[1]: *** [package/Makefile:184: /home/xxxx/projects/openmptcp/openmptcprouter/venice/6.12/source/staging_dir/target-aarch64_cortex-a53_musl/stamp/.package_install] Error 2
make[1]: Leaving directory '/home/xxxx/projects/openmptcp/openmptcprouter/venice/6.12/source'
make: *** [/home/xxxx/projects/openmptcp/openmptcprouter/venice/6.12/source/include/toplevel.mk:233: world] Error 2

I tried to run with extra verbosity, also added logs folder within make menuconfig, i can't get to where it fails. 

The build system is ubuntu 22.04.
OMR_TARGET="venice" OMR_FEED_SRC="develop" OMR_KERNEL="6.12" ./build.sh 

  - OpenMPTCProuter version: develop branch (git pull from last week). 
  - OpenMPTCProuter platform: Gateworks Venice 7301 - IMX8MM

Any help is much appreciated.

Ysurac, 2025-03-16:
6.12 kernel is not supported on many devices for now.

dekashio, 2025-03-16:
Thanks!, trying 6.6, will update soon.

dekashio, 2025-03-16:
Finished building kernel 6.6, Thanks for the help!

closed 2025-03-16 by dekashio