Ysurac/openmptcprouter #3856: 0.62-6.6 X86-64 seems to miss RTL1811 drivers

KoulEl, 2025-03-16:
## Expected Behavior

The network cards should start at the launch of OpenMPTCP Router

## Current Behavior

No network card is seen by the system. eth0, eth1 and eth2 are non existent.

## Possible Solution

Maybe check if RTL1811 drivers are still present in the kernel.

## Steps to Reproduce the Problem

  1. Use OpenMPTCP Router 0.61-6.6 x86-64 NON EFI
  2. Update the router to 0.62-6.6 using the update fonctionnality
  2. The PC will then reboot but won't see any of the network card
  3.

## Context (Environment)

I updated the router using the fonctionnality included and after reboot, nothing worked. Cheking directly with a display connected to the PC and using ifconfig to check network : no eth0, eth1 nor eth2.

## Specifications

  - OpenMPTCProuter version:  0.62-6.6
  - OpenMPTCProuter VPS version: 5.4.207-mptcp 0.1031
  - OpenMPTCProuter VPS provider: MilkyWan
  - OpenMPTCProuter platform: x86_64
  - Country: France

Ysurac, 2025-03-16:
Are you sure it's a RTL1811 ? I don't know about this card. Do you known the correct driver ?
You must update VPS.

KoulEl, 2025-03-16:
I meant this (retrieved from lspci with 0.61-6.6), 8111 no 1811 as I wrote (typo) : 

01:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Eth
ernet Controller (rev 06)
03:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Eth
ernet Controller (rev 06)
04:00.0 Ethernet controller: Realtek Semiconductor Co., Ltd. RTL8111/8168/8211/8411 PCI Express Gigabit Eth
ernet Controller (rev 06)

I will ask Milkywan to update my VPS.

KoulEl, 2025-03-22:
Back on this topic, I tried to download the ext4 image: **openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined.img.gz** to install it by hand on my router PC and it gave me errors while trying to unarchive it (I tried many times with different unarchivers).

Maybe this archive is broken.

KoulEl, 2025-03-22:
I tried both archives (EFI and non EFI) since I managed to unarchive them (even with the errors) and wrote them directly on the HDD. I can confirm these lack RTL 8111/8169 ethernet drivers.

So far, I'm stuck with 0.61-6.6.

Ysurac, 2025-03-22:
The error is normal, it's a header added but  you can uncompress it.
I will check to add correct driver in snapshot (strange because there is more drivers in 0.62 than 0.61, maybe a name change...)

intersvod, 2025-03-23:
I confirm, but 0.62 network interfaces on these Realtek chipsets do not work

nix48, 2025-04-10:
### **Can confirm**

### r8169 modul is missing in v0.62
### 

 **openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined-efi.img.gz:**

lsmod:  

qcserial               16384  0 
qmi_wwan               28672  0 
r8125                 241664  0 
r8152                  86016  0 
rndis_host             12288  0 

**openmptcprouter-v0.61-6.6-r0+27346-c7ba5574f5-x86-64-generic-ext4-combined-efi.img.gz:**

lsmod:

qcserial               16384  0 
qmi_wwan               28672  0 
r8152                  86016  0 
r8169                  73728  0 
realtek                28672  1 
rndis_host             12288  0 

dmesg:

[   13.523832] r8169 0000:01:00.0 eth0: RTL8168g/8111g, 00:8c:fa:d4:a9:1b, XID 4c0, IRQ 33
[   13.533204] r8169 0000:01:00.0 eth0: jumbo features [frames: 9194 bytes, tx checksumming: ko]

[   18.573902] r8169 0000:01:00.0 eth0: Link is Down
[   18.582680] r8169 0000:01:00.0 eth0: entered promiscuous mode
[   18.596651] r8169 0000:01:00.0 eth0: Link is Down
[   18.603907] r8169 0000:01:00.0 eth0: Link is Down
[   18.610187] r8169 0000:01:00.0 eth0: Link is Down
[   18.616399] r8169 0000:01:00.0 eth0: Link is Down
[   21.670448] r8169 0000:01:00.0 eth0: Link is Up - 1Gbps/Full - flow control rx/tx

helixcn2024, 2025-05-22:
Your problem does exist. You need to install the driver yourself. I have installed it now and it works normally.

iPowerTech, 2025-07-31:
Same issue, I mean:
Version 0.61 loads the r8169 module (RTL8211C Gigabit Ethernet) successfully.
Version 0.62, it gets stuck on kmodloader, then i can not connet to the server.

Is it possible to install the module manually from the local terminal?  (no network)

this build does not work on my server:
openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined.img.gz