Ysurac/openmptcprouter #3857: We encountered an issue during compilation where the libmbim library cannot be downloaded. Could you please provide us with the patch path

lhc1993, 2025-03-16:
## Expected Behavior

time: package/utils/f2fs-tools/default/compile#0.15#0.02#0.17
fatal: unable to access 'https://gitlab.freedesktop.org/mobile-broadband/libmbim.git/': The requested URL returned error: 50

## Current Behavior

## Possible Solution

## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.62rc1
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: 
  - Country:

lhc1993, 2025-03-16:
Do you have a repair plan

lhc1993, 2025-03-16:
Or how can I adapt locally

lhc1993, 2025-03-16:
[image]

The master branch is also like this

Ysurac, 2025-03-16:
The reason is here: https://github.com/Ysurac/openmptcprouter/issues/3857
I will maybe use a mirror, I will see next week.

lhc1993, 2025-03-16:
I don't understand what it means, does it mean that the problem category should not be mentioned? I just want to know how to choose the mptcp scheduler and blocking control when the network is busy. For example, is the combination of mptcp-bpf_rr and cubic suitable for scenarios with busy networks or poor mobile card signals

Ysurac, 2025-03-20:
I don't understand how this comment is related to libmbim issue (now solved).
On bad network, only OMR-Tracker Manager can really help by configuring lower timeout for example.

lhc1993, 2025-03-21:
On bad network, only OMR-Tracker Manager can really help by configuring lower timeout for example.
--- Can you give me a practical configuration example? I can't find a specific configuration guide in the official wiki, and I don't know how to set these parameters.

closed 2025-06-25 by github-actions[bot]