Ysurac/openmptcprouter #3858: Netgear R7800 new releases are broken

UDNLore23, 2025-03-17:
## Expected Behavior
After installing OpenMPTCProuter, I should be able to connect WAN ports and see the interfaces correctly displayed on the Interfaces page.

## Current Behavior
Interfaces are not functioning as expected. I tested kernel versions from 6.1 to 6.6, but there was no change.
The last working version is 0.61 on kernel 5.4 (I had to manually install the ath10k WiFi drivers, but at least the interfaces worked on this version).

## Steps to Reproduce the Problem

  1. Install any of the mentioned versions (6.1 to 6.6).
  2. Check the Interfaces page to see if interfaces are recognized and there is no error.
 

## Context (Environment)
This issue prevents the use of newer OpenMPTCProuter versions, requiring a downgrade to an older release for proper interface functionality.

## Specifications

  - OpenMPTCProuter version:  v60-6.1 / v61-6.6 (I tried also many snapshots...no way)
  - OpenMPTCProuter platform: Netgear R7800

[image]

Ysurac, 2025-03-17:
I would need the result, via SSH, of ```ip a```

UDNLore23, 2025-03-17:
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 brd 127.255.255.255 scope global lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1502 qdisc mq state UP group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2e30:33ff:fe9c:b3a0/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST> mtu 1502 qdisc mq state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:9f brd ff:ff:ff:ff:ff:ff
4: lan4@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
5: lan3@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
6: lan2@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
7: lan1@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 2000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global wan
       valid_lft forever preferred_lft forever
9: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 521e:a0e6:65de::
10: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
11: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
12: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
13: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
14: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr eac3:3762:5c3::
15: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
root@OpenMPTCProuter:~# oot@OpenMPTCProuter:~# ip a
-ash: oot@OpenMPTCProuter:~#: not found
root@OpenMPTCProuter:~# 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
-ash: can't open LOOPBACK,UP,LOWER_UP: no such file
root@OpenMPTCProuter:~#     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
-ash: link/loopback: not found
root@OpenMPTCProuter:~#     inet 127.0.0.1/8 brd 127.255.255.255 scope global lo
-ash: inet: not found
root@OpenMPTCProuter:~#        valid_lft forever preferred_lft forever
-ash: valid_lft: not found
root@OpenMPTCProuter:~#     inet6 ::1/128 scope host proto kernel_lo 
-ash: inet6: not found
root@OpenMPTCProuter:~#        valid_lft forever preferred_lft forever
-ash: valid_lft: not found
root@OpenMPTCProuter:~# 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1502 qdisc mq state UP group default qlen 1000
-ash: can't open BROADCAST,MULTICAST,UP,LOWER_UP: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~#     inet6 fe80::2e30:33ff:fe9c:b3a0/64 scope link proto kernel_ll 
-ash: inet6: not found
root@OpenMPTCProuter:~#        valid_lft forever preferred_lft forever
-ash: valid_lft: not found
root@OpenMPTCProuter:~# 3: eth1: <BROADCAST,MULTICAST> mtu 1502 qdisc mq state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:9f brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 4: lan4@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 5: lan3@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 6: lan2@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 7: lan1@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 2000
-ash: can't open BROADCAST,MULTICAST,UP,LOWER_UP: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found

UDNLore23, 2025-03-17:
[out.txt](https://github.com/user-attachments/files/19288403/out.txt)

UDNLore23, 2025-03-17:
```
root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 brd 127.255.255.255 scope global lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1502 qdisc mq state UP group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
    inet6 fe80::2e30:33ff:fe9c:b3a0/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
3: eth1: <BROADCAST,MULTICAST> mtu 1502 qdisc mq state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:9f brd ff:ff:ff:ff:ff:ff
4: lan4@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
5: lan3@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
6: lan2@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
7: lan1@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 2000
    link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
    inet 192.168.100.1/24 brd 192.168.100.255 scope global wan
       valid_lft forever preferred_lft forever
9: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 521e:a0e6:65de::
10: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
11: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
12: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
13: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
14: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr eac3:3762:5c3::
15: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
root@OpenMPTCProuter:~# oot@OpenMPTCProuter:~# ip a
-ash: oot@OpenMPTCProuter:~#: not found
root@OpenMPTCProuter:~# 1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
-ash: can't open LOOPBACK,UP,LOWER_UP: no such file
root@OpenMPTCProuter:~#     link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
-ash: link/loopback: not found
root@OpenMPTCProuter:~#     inet 127.0.0.1/8 brd 127.255.255.255 scope global lo
-ash: inet: not found
root@OpenMPTCProuter:~#        valid_lft forever preferred_lft forever
-ash: valid_lft: not found
root@OpenMPTCProuter:~#     inet6 ::1/128 scope host proto kernel_lo 
-ash: inet6: not found
root@OpenMPTCProuter:~#        valid_lft forever preferred_lft forever
-ash: valid_lft: not found
root@OpenMPTCProuter:~# 2: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1502 qdisc mq state UP group default qlen 1000
-ash: can't open BROADCAST,MULTICAST,UP,LOWER_UP: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~#     inet6 fe80::2e30:33ff:fe9c:b3a0/64 scope link proto kernel_ll 
-ash: inet6: not found
root@OpenMPTCProuter:~#        valid_lft forever preferred_lft forever
-ash: valid_lft: not found
root@OpenMPTCProuter:~# 3: eth1: <BROADCAST,MULTICAST> mtu 1502 qdisc mq state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:9f brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 4: lan4@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 5: lan3@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 6: lan2@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 7: lan1@eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noqueue state DOWN group default qlen 1000
-ash: can't open BROADCAST,MULTICAST: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 8: wan@eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 2000
-ash: can't open BROADCAST,MULTICAST,UP,LOWER_UP: no such file
root@OpenMPTCProuter:~#     link/ether 2c:30:33:9c:b3:a0 brd ff:ff:ff:ff:ff:ff
-ash: link/ether: not found
root@OpenMPTCProuter:~# 
```

Ysurac, 2025-03-17:
Strange configuration here, we should have all "lan*" ports on eth1 instead of eth0.
It's an update or a new install ?

UDNLore23, 2025-03-17:
Upgrade from latest stable OpenWRT version. OpenWRT works fine.

Ysurac, 2025-03-17:
You installed with "keep settings" option or not ? With OpenWRT settings, OpenMPTCProuter can't work.

Lore239, 2025-03-17:
It is always installed without any options. I also tried installing it via TFTP using the BIN file, but as I mentioned, nothing works—I keep encountering the same issue. Additionally, WiFi doesn’t work, as there is no wireless option in the menu bar.

Ysurac, 2025-03-17:
I'm building a new snapshot that should save some issues (at least wifi) will be ok in a few hours.

UDNLore23, 2025-03-18:
Hi, I'm testing the new release, and it seems to be working perfectly now. Thanks for your work! I just sent you a little something for a coffee.

closed 2025-03-18 by UDNLore23