Ysurac/openmptcprouter #3866: I keep refreshing Replace default route by 10.255.252.1 dev tun0 (was ), and the aggregation is connected for a while and disconnected for a while.

lhc1993, 2025-03-21:
## Expected Behavior

I have been refreshing this log, and the aggregation is connected for a while and disconnected for a while, please help.
proxy: shdowsocks-rust 2022
vpn: openvpn tcp

has used template.

Fri Mar 21 14:01:30 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:01:35 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:01:41 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:01:47 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:01:52 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:01:58 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:02:03 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:02:09 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 14:02:14 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
## Current Behavior

openmptcprouter version:v0.62rc1
vps version:6.6.36-x64v4-xanmod1

## Specifications

  - OpenMPTCProuter version: 
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2025-03-21:
I would need System->Status log, and via SSH from the router ```ip r``` and ```ip a```
And please set specification liake versions in correct place

lhc1993, 2025-03-21:
[iproute.log](https://github.com/user-attachments/files/19393103/iproute.log)

[syslog.log](https://github.com/user-attachments/files/19393133/syslog.log)

[ipa.log](https://github.com/user-attachments/files/19393104/ipa.log)
[ipr.log](https://github.com/user-attachments/files/19393105/ipr.log)

lhc1993, 2025-03-22:
Please help me take a look

Ysurac, 2025-03-22:
Can you copy/paste files instead of attaching them ?

lhc1993, 2025-03-22:
ip a：
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
2: miireg: <> mtu 0 qdisc noop state DOWN group default qlen 1000
    link/netrom 
3: eth0: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether 2e:7b:fa:d7:24:5c brd ff:ff:ff:ff:ff:ff
4: eth1: <BROADCAST,MULTICAST> mtu 1500 qdisc noop state DOWN group default qlen 1000
    link/ether ae:3a:a9:49:53:a2 brd ff:ff:ff:ff:ff:ff
5: eth2: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq state DOWN group default qlen 1000
    link/ether d2:cb:42:9a:bf:f4 brd ff:ff:ff:ff:ff:ff
6: eth3: <NO-CARRIER,BROADCAST,MULTICAST,UP> mtu 1500 qdisc mq master br-lan state DOWN group default qlen 1000
    link/ether 88:12:3d:07:c2:9f brd ff:ff:ff:ff:ff:ff
7: eth4: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc mq master br-lan state UP group default qlen 1000
    link/ether 88:12:3d:07:c2:a0 brd ff:ff:ff:ff:ff:ff
8: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr 8af9:7bb0:8bab::
9: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
10: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
11: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
12: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
13: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr c279:e46e:f6ff::
14: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
15: wwan0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/ether a2:39:86:40:c2:6f brd ff:ff:ff:ff:ff:ff
    inet6 fe80::a039:86ff:fe40:c26f/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
16: wwan1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/none 
    inet 10.207.7.183/28 brd 10.207.7.191 scope global wwan1
       valid_lft forever preferred_lft forever
    inet6 fe80::4787:b8e1:7da4:70c3/64 scope link stable-privacy proto kernel_ll 
       valid_lft forever preferred_lft forever
17: pcie_mhi0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/ether 72:7a:14:25:06:15 peer ff:ff:ff:ff:ff:ff
    inet 10.171.147.123/29 brd 10.171.147.127 scope global pcie_mhi0
       valid_lft forever preferred_lft forever
    inet6 fe80::707a:14ff:fe25:615/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
18: pcie_mhi1: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 1000
    link/ether 02:13:52:5c:22:b8 peer ff:ff:ff:ff:ff:ff
    inet6 fe80::13:52ff:fe5c:22b8/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
19: br-lan: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue state UP group default qlen 2000
    link/ether 88:12:3d:07:c2:9f brd ff:ff:ff:ff:ff:ff
    inet 192.168.9.1/24 brd 192.168.9.255 scope global br-lan
       valid_lft forever preferred_lft forever
    inet6 fdd7:43b6:3dac::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::8a12:3dff:fe07:c29f/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
24: phy1-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 2000
    link/ether 8c:53:c3:dc:4d:dd brd ff:ff:ff:ff:ff:ff
28: ifb4eth3: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 3a:f0:eb:fd:21:11 brd ff:ff:ff:ff:ff:ff
29: phy0-ap0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc noqueue master br-lan state UP group default qlen 2000
    link/ether 8c:53:c3:dc:4d:de brd ff:ff:ff:ff:ff:ff
32: ifb4eth4: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 36:89:f4:9e:be:90 brd ff:ff:ff:ff:ff:ff
35: ifb4eth1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 0e:3d:35:cc:98:af brd ff:ff:ff:ff:ff:ff
38: ifb4eth2: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether be:6f:47:9c:de:7b brd ff:ff:ff:ff:ff:ff
39: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 500
    link/none 
    inet 10.255.252.2/24 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::64f5:a40f:e735:8004/64 scope link stable-privacy proto kernel_ll 
       valid_lft forever preferred_lft forever

lhc1993, 2025-03-22:
IP r：
default dev wwan1 proto static scope link 
default via 10.255.252.1 dev tun0 
default via 10.207.7.184 dev wwan1 metric 3 
default via 10.171.147.124 dev pcie_mhi0 metric 11 
10.171.147.120/29 dev pcie_mhi0 scope link metric 11 
10.171.147.124 dev pcie_mhi0 scope link metric 11 
10.207.7.176/28 dev wwan1 scope link metric 3 
10.207.7.184 dev wwan1 scope link metric 3 
10.255.252.0/24 dev tun0 proto kernel scope link src 10.255.252.2 
123.57.0.66 via 10.207.7.184 dev wwan1 metric 3 
123.57.0.66 via 10.171.147.124 dev pcie_mhi0 metric 11 
192.168.9.0/24 dev br-lan scope link metric 9999

lhc1993, 2025-03-22:
ip router get 1.1.1.1:
1.1.1.1 dev wwan1 src 10.207.7.183 uid 0 
    cache

lhc1993, 2025-03-22:
syslog:
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.671225] [I][mhi_ready_state_transition] Device in READY State
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.678237] mhi_pci_probe pci_dev->name = 0001:01:00.0, domain=1, bus=1, slot=0, vendor=17CB, device=0306
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.682937] [I][mhi_init_mmio] Initializing MMIO
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.688959] [I][mhi_alloc_controller] 1114 size = 760
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.698479] [I][mhi_init_mmio] iova_start = 0, iova_stop = fffffffff
Fri Mar 21 16:01:41 2025 kern.err kernel: [    4.703183] debugfs: File 'debug_mode' in directory 'mhi' already present!
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.708119] [I][mhi_init_mmio] cmd_ctxt_addr = 439d5f00
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.714545] mhi_pci_probe liuhucheng mhi_platform_probe
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.721214] [I][mhi_init_mmio] chan_ctxt_addr = 439d4800
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.726344] mhi_pci_probe liuhucheng mhi_controller_get_devdata ret: 0
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.731545] [I][mhi_init_mmio] er_ctxt_addr = 439d5e00
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.737111] mhi_pci_probe liuhucheng mhi_arch_pcie_init ret: 0
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.743440] [I][mhi_init_mmio] CHDBOFF:0x300
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.748655] mhi_pci_probe liuhucheng mhi_arch_iommu_init ret: 0
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.754474] [I][mhi_init_mmio] ERDBOFF:0x700
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.758903] mhi 0001:01:00.0: BAR 0: assigned [mem 0x10300000-0x10300fff 64bit]
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.764539] [I][mhi_init_mmio] Programming all MMIO values.
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.769153] mhi 0001:01:00.0: enabling device (0000 -> 0002)
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.776142] [I][mhi_ready_state_transition] mhi_db_sw_ring
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.781705] [I][mhi_init_pci_dev] mhi_cntrl->regs = (____ptrval____)
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.793249] [I][mhi_init_pci_dev] msi_required = 4, msi_allocated = 4, msi_irq = 42
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.799360] mhi_pci_probe liuhucheng mhi_init_pci_dev ret: 0
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.806729] [I][mhi_async_power_up] Requested to power on
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.812626] [I][mhi_async_power_up] mhi_cfg = 0x800080
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.883753] [I][mhi_async_power_up] mhi_state = READY
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.883826] [I][mhi_alloc_coherent] size = 98304, dma_handle = 439e0000
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.887778] [I][mhi_init_dev_ctxt] mhi_ctxt->ctrl_seg = (____ptrval____)
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.894565] [I][mhi_async_power_up] bhi_offset = 0x100
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.901147] [I][mhi_async_power_up] current_ee = AMSS
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.906227] [I][mhi_pm_st_worker] Transition to state:READY
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.906327] [I][mhi_async_power_up] Power on setup success
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.911214] [I][mhi_ready_state_transition] Waiting to enter READY state
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.916614] mhi_pci_probe liuhucheng mhi_async_power_up ret: 0
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.922160] [I][mhi_ready_state_transition] Device current EE:AMSS, M:READY
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.929062] [I][mhi_pci_probe] Return successful
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.934665] [I][mhi_ready_state_transition] Device in READY State
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.934671] [I][mhi_init_mmio] Initializing MMIO
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.941590] liuhucheng init successmhi_init Fibocom_Linux_PCIE_MHI_Driver_V1.0.3
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.946380] [I][mhi_init_mmio] iova_start = 0, iova_stop = fffffffff
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.964451] [I][mhi_init_mmio] cmd_ctxt_addr = 439f5f00
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.970761] [I][mhi_init_mmio] chan_ctxt_addr = 439f4800
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.975735] [I][mhi_init_mmio] er_ctxt_addr = 439f5e00
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.981268] [I][mhi_init_mmio] CHDBOFF:0x300
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.986242] [I][mhi_init_mmio] ERDBOFF:0x700
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.990640] [I][mhi_init_mmio] Programming all MMIO values.
Fri Mar 21 16:01:41 2025 kern.info kernel: [    4.994949] [I][mhi_ready_state_transition] mhi_db_sw_ring
Fri Mar 21 16:01:41 2025 kern.info kernel: [    5.004118] NET: Registered PF_INET6 protocol family
Fri Mar 21 16:01:41 2025 kern.info kernel: [    5.006900] Segment Routing with IPv6
Fri Mar 21 16:01:41 2025 kern.info kernel: [    5.010905] In-situ OAM (IOAM) with IPv6
Fri Mar 21 16:01:41 2025 kern.info kernel: [    5.014529] NET: Registered PF_PACKET protocol family
[... cut ...]
Fri Mar 21 16:06:13 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 16:06:18 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 16:06:24 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )
Fri Mar 21 16:06:29 2025 user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was )

Ysurac, 2025-03-22:
I would need the result of ```uci show network.4G-3```
It's used as default gateway and shouldn't.

lhc1993, 2025-03-22:
root@OpenMPTCProuter:~# uci show network | grep wwan1
network.wwan1=interface
network.wwan1.proto='dhcp'
network.wwan1.device='wwan1'
network.wwan1.multipath='on'
network.wwan1.label='4G-3'
network.wwan1.metric='3'
network.wwan1.ip4table='3'
network.wwan1.ip6table='63'
network.wwan1.defaultroute='0'
network.wwan1.peerdns='0'
network.wwan1.ipv6='0'
network.wwan1.delegate='0'
network.wwan1.modalias='usb:v1508p1001d0318dc00dsc00dp00icFFiscFFipFFin04-'
network.wwan1.product='1508/1001/318'
network.wwan1_dev=device
network.wwan1_dev.name='wwan1'
network.wwan1_fw_rule6=rule6
network.wwan1_fw_rule6.priority='1'
network.wwan1_fw_rule6.mark='0x65393'
network.wwan1_fw_rule6.lookup='3'
network.wwan1_fw_rule=rule
network.wwan1_fw_rule.priority='1'
network.wwan1_fw_rule.mark='0x45393'
network.wwan1_fw_rule.lookup='3'

lhc1993, 2025-03-22:
Any more information needed?

Ysurac, 2025-03-22:
Not for now, I don't understand how ```default dev wwan1 proto static scope link``` route can exist... wwan1 seems to force default gateway but there is no reason.

lhc1993, 2025-03-22:
Please help me check it out. If you need any information, please contact me. Thank you.

lhc1993, 2025-03-22:
Before 0.62rc1 was updated, it was OK.

Ysurac, 2025-03-22:
Maybe a ```uci show network```, something may conflict.

lhc1993, 2025-03-22:
root@OpenMPTCProuter:~# uci show network
network.loopback=interface
network.loopback.device='lo'
network.loopback.proto='static'
network.loopback.ipaddr='127.0.0.1'
network.loopback.netmask='255.0.0.0'
network.loopback.multipath='off'
network.loopback.metric='15'
network.loopback.ip4table='15'
network.loopback.ip6table='615'
network.globals=globals
network.globals.ula_prefix='fdd7:43b6:3dac::/48'
network.globals.packet_steering='0'
network.globals.multipath='enable'
network.globals.mptcp_path_manager='fullmesh'
network.globals.mptcp_scheduler='blest'
network.globals.congestion='bbr'
network.globals.mptcp_checksum='0'
network.globals.mptcp_debug='0'
network.globals.mptcp_syn_retries='4'
network.globals.mptcp_subflows='8'
network.globals.mptcp_add_addr_accepted='8'
network.globals.mptcp_add_addr_timeout='120'
network.globals.mptcp_pm_type='0'
network.globals.mptcp_disable_initial_config='0'
network.globals.mptcp_force_multipath='1'
network.globals.mptcpd_enable='0'
network.globals.mptcp_fullmesh_num_subflows='1'
network.globals.mptcp_fullmesh_create_on_err='1'
network.globals.mptcp_ndiffports_num_subflows='1'
network.lan=interface
network.lan.device='br-lan'
network.lan.proto='static'
network.lan.ipaddr='192.168.9.1'
network.lan.netmask='255.255.255.0'
network.lan.ip6assign='60'
network.lan.metric='9999'
network.lan.ipv6='0'
network.lan.delegate='0'
network.lan.addlatency='0'
network.lan.txqueuelen='2000'
network.lan.multipath='off'
network.lan.ip4table='9999'
network.lan.defaultroute='0'
network.lan.peerdns='0'
network.lan.ip6table='69999'
network.lan_rule=rule
network.lan_rule.lookup='lan'
network.lan_rule.priority='100'
network.wan1_dev=device
network.wan1_dev.name='eth1'
network.wan1_dev.txqueuelen='1000'
network.wan2_dev=device
network.wan2_dev.name='eth2'
network.wan2_dev.txqueuelen='1000'
network.wan3=interface
network.wan3.device='eth2'
network.wan3.proto='dhcp'
network.wan3.ip4table='13'
network.wan3.multipath='master'
network.wan3.defaultroute='0'
network.wan3.delegate='0'
network.wan3.addlatency='0'
network.wan3.label='WAN'
network.wan3.metric='13'
network.wan3.ip6table='613'
network.wan3.peerdns='0'
network.wan3.ipv6='0'
network.wan3_dev=device
network.wan3_dev.name='eth3'
[... 175 lines cut ...]
network.pcie_mhi0_fw_rule.lookup='11'
network.pcie_mhi1_fw_rule=rule
network.pcie_mhi1_fw_rule.priority='1'
network.pcie_mhi1_fw_rule.mark='0x45399'
network.pcie_mhi1_fw_rule.lookup='9'
network.wwan0_fw_rule=rule
network.wwan0_fw_rule.priority='1'
network.wwan0_fw_rule.mark='0x453914'
network.wwan0_fw_rule.lookup='14'
network.wwan1_fw_rule=rule
network.wwan1_fw_rule.priority='1'
network.wwan1_fw_rule.mark='0x45393'
network.wwan1_fw_rule.lookup='3'
network.eth5_fw_rule=rule
network.eth5_fw_rule.priority='1'
network.eth5_fw_rule.mark='0x453912'
network.eth5_fw_rule.lookup='12'
network.eth6_fw_rule=rule
network.eth6_fw_rule.priority='1'
network.eth6_fw_rule.mark='0x453910'
network.eth6_fw_rule.lookup='10'
network.all_fw_rule=rule
network.all_fw_rule.priority='1'
network.all_fw_rule.mark='0x4539'
network.all_fw_rule.lookup='991337'

lhc1993, 2025-03-22:
can you have some ideas

lhc1993, 2025-03-26:
tks

closed 2025-03-26 by lhc1993