Ysurac/openmptcprouter #3867: A

JulienHG65, 2025-03-21:
Hello,

Since upgrade of my router and VPS, I've unstable connection.

Can we downgrade ?

Thanks

## Expected Behavior
I've upgrade VPS and router. After this, i've a lot of connection interruption

## Current Behavior

## Possible Solution
downgrade

## Steps to Reproduce the Problem

Upgrade router and VPS to latest version

## Context (Environment)
I try to install a new VPS, with new VPS, same problem.

## Specifications

  - OpenMPTCProuter version: 0.62-6.6
  - OpenMPTCProuter VPS version: 0.1032.6.1
  - OpenMPTCProuter VPS provider: AWS Paris FR
  - OpenMPTCProuter platform: RPI4b
  - Country: France

lhc1993, 2025-03-22:
I also had this problem after upgrading. The phenomenon was the same as yours, and it was also unstable. It was connected for a while and then not connected for a while.

Ysurac, 2025-03-22:
Would be better to solve the issue...
Can I get Status->System log ?

closed 2025-06-25 by github-actions[bot]