Ysurac/openmptcprouter #3868: No access tomy webserver snce server upgrade

KoulEl, 2025-03-22:
Can't access my web servers from external.

I should be able to access my web sites (eg. https://video.amiga-ng.org). I can access them from my internal network.
## Current Behavior

I try to access my websites but get stuck until timeout. Nothing revelant in the System log (except this : Mar 22 09:46:36 OpenMPTCProuter daemon.err dnsmasq[1]: nftset inet fw4 omr_dscp_cs2_6 Error: Could not resolve hostname: Name has no usable address).

It used to work correctly. I even recovered my saved configuration from the server to be sure I had not changed anything. Still the same problem. Everything ik in Firewall rules and OMR-Bypass.

## Possible Solution

Maybe downgrade the VPS?

## Steps to Reproduce the Problem

  1. Ask Milkywan to upgrade the VPS.
  2. Try to access auto-hosting sites.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version:  v0.61-6.6 r0+27346-c7ba5574 (not the 0.62-66 since I had troubles with network cards recognition as explained in another issue)
  - OpenMPTCProuter VPS version:  v0.61-6.6 r0+27346-c7ba5574
  - OpenMPTCProuter VPS provider: Milkywan
  - OpenMPTCProuter platform:  x86_64
  - Country:  France

Ysurac, 2025-03-22:
Follow https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding#debug and give me result.

KoulEl, 2025-03-22:
Thanks, I cannot (or don't know how to) access the VPS, I'll ask Milkywan for help and will come back to you.

Ysurac, 2025-03-22:
In fact just the tcpdump part on the router will tell if firewall config work or not

KoulEl, 2025-03-22:
Here are the results:

root@OpenMPTCProuter:~# tcpdump -i tun0 port 80
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on tun0, link-type RAW (Raw IP), snapshot length 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel

root@OpenMPTCProuter:~# tcpdump -i tun0 port 443
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on tun0, link-type RAW (Raw IP), snapshot length 262144 bytes
^C
0 packets captured
0 packets received by filter
0 packets dropped by kernel

Ysurac, 2025-03-22:
When you tried to reach website ?
Can you give me result of ```uci show firewall``` ?

KoulEl, 2025-03-22:
Here it is :

root@OpenMPTCProuter:~# uci show firewall
firewall.@defaults[0]=defaults
firewall.@defaults[0].syn_flood='1'
firewall.@defaults[0].input='REJECT'
firewall.@defaults[0].output='REJECT'
firewall.@defaults[0].forward='REJECT'
firewall.@defaults[0].fullcone='0'
firewall.@defaults[0].flow_offloading='0'
firewall.@defaults[0].flow_offloading_hw='0'
firewall.zone_lan=zone
firewall.zone_lan.name='lan'
firewall.zone_lan.network='lan'
firewall.zone_lan.input='ACCEPT'
firewall.zone_lan.output='ACCEPT'
firewall.zone_lan.forward='ACCEPT'
firewall.zone_lan.auto_helper='1'
firewall.zone_lan.mtu_fix='1'
firewall.zone_wan=zone
firewall.zone_wan.name='wan'
firewall.zone_wan.input='REJECT'
firewall.zone_wan.output='ACCEPT'
firewall.zone_wan.forward='REJECT'
firewall.zone_wan.fullcone4='0'
firewall.zone_wan.fullcone6='0'
firewall.zone_wan.masq='1'
firewall.zone_wan.mtu_fix='1'
firewall.zone_wan.auto_helper='1'
firewall.zone_wan.network='wan1' 'wan2'
firewall.@forwarding[0]=forwarding
firewall.@forwarding[0].src='lan'
firewall.@forwarding[0].dest='wan'
firewall.@rule[0]=rule
firewall.@rule[0].name='Allow-DHCP-Renew'
firewall.@rule[0].src='wan'
firewall.@rule[0].proto='udp'
firewall.@rule[0].dest_port='68'
firewall.@rule[0].target='ACCEPT'
firewall.@rule[0].family='ipv4'
firewall.@rule[1]=rule
firewall.@rule[1].name='Allow-Ping'
firewall.@rule[1].src='wan'
firewall.@rule[1].proto='icmp'
firewall.@rule[1].icmp_type='echo-request'
firewall.@rule[1].family='ipv4'
firewall.@rule[1].target='ACCEPT'
firewall.@rule[2]=rule
firewall.@rule[2].name='Allow-IGMP'
firewall.@rule[2].src='wan'
firewall.@rule[2].proto='igmp'
firewall.@rule[2].family='ipv4'
firewall.@rule[2].target='ACCEPT'
firewall.@rule[3]=rule
firewall.@rule[3].name='Allow-DHCPv6'
firewall.@rule[3].src='wan'
firewall.@rule[3].proto='udp'
firewall.@rule[3].dest_port='546'
firewall.@rule[3].family='ipv6'
firewall.@rule[3].target='ACCEPT'
firewall.@rule[4]=rule
firewall.@rule[4].name='Allow-MLD'
firewall.@rule[4].src='wan'
firewall.@rule[4].proto='icmp'
firewall.@rule[4].src_ip='fe80::/10'
firewall.@rule[4].icmp_type='130/0' '131/0' '132/0' '143/0'
firewall.@rule[4].family='ipv6'
firewall.@rule[4].target='ACCEPT'
firewall.@rule[5]=rule
firewall.@rule[5].name='Allow-ICMPv6-Forward'
[... 1793 lines cut ...]
firewall.omr_dst_bypass_all_srcport_tcp_6.set_xmark='0x6539'
firewall.omr_dst_bypass_all_srcport_udp_6=rule
firewall.omr_dst_bypass_all_srcport_udp_6.name='omr_dst_bypass_all_srcport_udp'
firewall.omr_dst_bypass_all_srcport_udp_6.proto='udp'
firewall.omr_dst_bypass_all_srcport_udp_6.src='lan'
firewall.omr_dst_bypass_all_srcport_udp_6.dest='*'
firewall.omr_dst_bypass_all_srcport_udp_6.target='MARK'
firewall.omr_dst_bypass_all_srcport_udp_6.enabled='0'
firewall.omr_dst_bypass_all_srcport_udp_6.set_xmark='0x6539'
firewall.omr_dst_bypass_all_dstport_tcp_6=rule
firewall.omr_dst_bypass_all_dstport_tcp_6.name='omr_dst_bypass_all_dstport_tcp'
firewall.omr_dst_bypass_all_dstport_tcp_6.src='lan'
firewall.omr_dst_bypass_all_dstport_tcp_6.dest='*'
firewall.omr_dst_bypass_all_dstport_tcp_6.target='MARK'
firewall.omr_dst_bypass_all_dstport_tcp_6.proto='tcp'
firewall.omr_dst_bypass_all_dstport_tcp_6.enabled='0'
firewall.omr_dst_bypass_all_dstport_tcp_6.set_xmark='0x6539'
firewall.omr_dst_bypass_all_dstport_udp_6=rule
firewall.omr_dst_bypass_all_dstport_udp_6.name='omr_dst_bypass_all_dstport_udp'
firewall.omr_dst_bypass_all_dstport_udp_6.src='lan'
firewall.omr_dst_bypass_all_dstport_udp_6.dest='*'
firewall.omr_dst_bypass_all_dstport_udp_6.proto='udp'
firewall.omr_dst_bypass_all_dstport_udp_6.target='MARK'
firewall.omr_dst_bypass_all_dstport_udp_6.enabled='0'
firewall.omr_dst_bypass_all_dstport_udp_6.set_xmark='0x6539'

KoulEl, 2025-03-23:
And what Milkywan sent me (it seems ok to me):

cat /etc/shorewall/rules
#
# Shorewall version 4.0 - Sample Rules File for two-interface configuration.
# Copyright (C) 2006-2014,2007 by the Shorewall Team
#
# This library is free software; you can redistribute it and/or
# modify it under the terms of the GNU Lesser General Public
# License as published by the Free Software Foundation; either
# version 2.1 of the License, or (at your option) any later version.
#
# See the file README.txt for further details.
#------------------------------------------------------------------------------
# For information about entries in this file, type "man shorewall-rules"
######################################################################################################################################################################################################

#ACTION         SOURCE          DEST            PROTO   DEST    SOURCE          ORIGINAL        RATE            USER/   MARK    CONNLIMIT       TIME            HEADERS         SWITCH          HELPER
#                                                       PORT    PORT(S)         DEST            LIMIT           GROUP
?SECTION ALL
?SECTION ESTABLISHED
?SECTION RELATED
?SECTION INVALID
?SECTION UNTRACKED
?SECTION NEW

#       Don't allow connection pickup from the net
#
Invalid(DROP)   net             all             tcp
#
#       Accept DNS connections from the firewall to the network
#
DNS(ACCEPT)     $FW             net
#
#       Allow Ping from/to the VPN
#
Ping(ACCEPT)    vpn             $FW
Ping(ACCEPT)    $FW             vpn
#
#       Allow Ping from the firewall to the network
#
Ping(ACCEPT)    $FW             net
#
# Drop Ping from the "bad" net zone.. and prevent your log from being flooded..
#
#Ping(DROP)     net             $FW
Ping(ACCEPT)    net             $FW
#
# Accept connection from port > 65000 for shadowsocks and glorytun on the firewall
#
ACCEPT          net             $FW             tcp     65000-65535
ACCEPT          net             $FW             udp     65000-65535
#
# Accept connection from SSH to the firewall
#
ACCEPT          net             $FW             tcp     65222
#
# DHCP forward to the VPN from the firewall
#
DHCPfwd(ACCEPT) $FW vpn
#
# Redirect all port from 1 to 64999 to the VPN client from the network
#
#DNAT net vpn:$OMR_ADDR tcp 1-64999
#DNAT net vpn:$OMR_ADDR udp 1-64999
DNAT net vpn:$OMR_ADDR tcp 25 # OMR openmptcprouter redirect router 25 port tcp
DNAT net vpn:$OMR_ADDR tcp 21 # OMR openmptcprouter redirect router 21 port tcp
DNAT net vpn:$OMR_ADDR udp 20 # OMR openmptcprouter redirect router 20 port udp
DNAT net vpn:$OMR_ADDR tcp 443 # OMR openmptcprouter redirect router 443 port tcp
DNAT net vpn:$OMR_ADDR tcp 4433 # OMR openmptcprouter redirect router 4433 port tcp
DNAT net vpn:$OMR_ADDR tcp 4443 # OMR openmptcprouter redirect router 4443 port tcp
DNAT net vpn:$OMR_ADDR udp 10000 # OMR openmptcprouter redirect router 10000 port udp
DNAT net vpn:$OMR_ADDR tcp 2222 # OMR openmptcprouter redirect router 2222 port tcp
DNAT net vpn:$OMR_ADDR tcp 4435 # OMR openmptcprouter redirect router 4435 port tcp
DNAT net vpn:$OMR_ADDR tcp 4455 # OMR openmptcprouter redirect router 4455 port tcp
DNAT net vpn:$OMR_ADDR tcp 4466 # OMR openmptcprouter redirect router 4466 port tcp
DNAT net vpn:$OMR_ADDR tcp 4436 # OMR openmptcprouter redirect router 4436 port tcp
DNAT net vpn:$OMR_ADDR tcp 110 # OMR openmptcprouter redirect router 110 port tcp
DNAT net vpn:$OMR_ADDR tcp 143 # OMR openmptcprouter redirect router 143 port tcp
DNAT net vpn:$OMR_ADDR tcp 587 # OMR openmptcprouter redirect router 587 port tcp
DNAT net vpn:$OMR_ADDR tcp 993 # OMR openmptcprouter redirect router 993 port tcp
DNAT net vpn:$OMR_ADDR tcp 995 # OMR openmptcprouter redirect router 995 port tcp
DNAT net vpn:$OMR_ADDR tcp 20 # OMR openmptcprouter redirect router 20 port tcp
DNAT net vpn:$OMR_ADDR tcp 10090-10100 # OMR openmptcprouter redirect router 10090-10100 port tcp
DNAT net vpn:$OMR_ADDR tcp 22 # OMR openmptcprouter redirect router 22 port tcp
DNAT net vpn:$OMR_ADDR tcp 8080 # OMR openmptcprouter redirect router 8080 port tcp
DNAT net vpn:$OMR_ADDR tcp 63241 # OMR openmptcprouter redirect router 63241 port tcp
DNAT net vpn:$OMR_ADDR tcp 22027 # OMR openmptcprouter redirect router 22027 port tcp
DNAT net vpn:$OMR_ADDR tcp 22028 # OMR openmptcprouter redirect router 22028 port tcp
DNAT net vpn:$OMR_ADDR tcp 22029 # OMR openmptcprouter redirect router 22029 port tcp
DNAT net vpn:$OMR_ADDR tcp 4445-4446 # OMR openmptcprouter redirect router 4445-4446 port tcp
ACCEPT net $FW tcp 65301 # OMR openmptcprouter open openvpn port tcp
ACCEPT net $FW udp 65311 # OMR openmptcprouter open wireguard port udp
DNAT net vpn:$OMR_ADDR tcp 16881 # OMR openmptcprouter redirect router 16881 port tcp
DNAT net vpn:$OMR_ADDR udp 16881 # OMR openmptcprouter redirect router 16881 port udp
DNAT net vpn:$OMR_ADDR tcp 22026 # OMR openmptcprouter redirect router 22026 port tcp
DNAT net vpn:$OMR_ADDR tcp 80 # OMR openmptcprouter redirect router 80 port tcp

cat /etc/shorewall/params.vpn

VPS_ADDR=10.255.252.1
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
VPS_IFACE=tun0
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6
OMR_ADDR=10.255.252.6

Ysurac, 2025-03-23:
Can you try, if you have access on VPS, to do a ```ping 10.255.252.6``` if yes then all data (when you try to reach from external source the server) should be forwarded on router and you should see them in tcpdump.
If not, I would need ```iptables-save``` on VPS.

KoulEl, 2025-03-23:
I will ask Milkywan if they can do this for me since I have no access to the VPS, it is fully managed by them.

On my side, nothing has been changed on the router. It stopped working after the VPS update (if there's no solution, I'll ask Milkywan to downgrade to the former one in order to get access to my sites again).

Ysurac, 2025-03-24:
What is the result of ```ip r``` on the router ?
Not sure downgrade will help as I made no changes on this part.

KoulEl, 2025-03-24:
Here it is :

root@OpenMPTCProuter:~# ip r
default via 10.255.252.1 dev tun0 
default via 192.168.0.1 dev eth1 metric 6 
default via 192.168.1.1 dev eth2 metric 7 
default via 10.255.252.1 dev tun0 metric 1500 
10.255.252.0/24 dev tun0 proto kernel scope link src 10.255.252.3 
10.255.252.0/24 dev tun0 scope link metric 1500 
45.13.104.49 metric 1 
	nexthop via 192.168.0.1 dev eth1 weight 100 
	nexthop via 192.168.1.1 dev eth2 weight 1 
127.0.0.0/8 dev lo proto static scope link metric 5 
192.168.0.0/24 dev eth1 scope link metric 6 
192.168.1.0/24 dev eth2 scope link metric 7

Ysurac, 2025-03-24:
And ```ip a``` please ? to check if it's the correct VPN config set on VPS

KoulEl, 2025-03-24:
An here it is:

root@OpenMPTCProuter:~# ip a
1: lo: <LOOPBACK,UP,LOWER_UP> mtu 65536 qdisc noqueue state UNKNOWN group default qlen 1000
    link/loopback 00:00:00:00:00:00 brd 00:00:00:00:00:00
    inet 127.0.0.1/8 scope host lo
       valid_lft forever preferred_lft forever
    inet6 ::1/128 scope host proto kernel_lo 
       valid_lft forever preferred_lft forever
2: ip6tnl0@NONE: <NOARP> mtu 1452 qdisc noop state DOWN group default qlen 1000
    link/tunnel6 :: brd :: permaddr a268:3177:5eb4::
3: sit0@NONE: <NOARP> mtu 1480 qdisc noop state DOWN group default qlen 1000
    link/sit 0.0.0.0 brd 0.0.0.0
4: gre0@NONE: <NOARP> mtu 1476 qdisc noop state DOWN group default qlen 1000
    link/gre 0.0.0.0 brd 0.0.0.0
5: gretap0@NONE: <BROADCAST,MULTICAST> mtu 1462 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
6: erspan0@NONE: <BROADCAST,MULTICAST> mtu 1450 qdisc noop state DOWN group default qlen 1000
    link/ether 00:00:00:00:00:00 brd ff:ff:ff:ff:ff:ff
7: ip6gre0@NONE: <NOARP> mtu 1448 qdisc noop state DOWN group default qlen 1000
    link/gre6 :: brd :: permaddr 1a65:39f7:a9a4::
8: teql0: <NOARP> mtu 1500 qdisc noop state DOWN group default qlen 100
    link/void 
9: eth0: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:0a:cd:31:d8:66 brd ff:ff:ff:ff:ff:ff
    inet 192.168.5.1/24 brd 192.168.5.255 scope global eth0
       valid_lft forever preferred_lft forever
    inet6 fd3e:c459:2457::1/60 scope global noprefixroute 
       valid_lft forever preferred_lft forever
    inet6 fe80::20a:cdff:fe31:d866/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
10: eth1: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:18:7d:38:f0:9d brd ff:ff:ff:ff:ff:ff
    inet 192.168.0.20/24 brd 192.168.0.255 scope global eth1
       valid_lft forever preferred_lft forever
    inet6 fe80::218:7dff:fe38:f09d/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
11: eth2: <BROADCAST,MULTICAST,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UP group default qlen 1000
    link/ether 00:18:7d:38:f0:9e brd ff:ff:ff:ff:ff:ff
    inet 192.168.1.20/24 brd 192.168.1.255 scope global eth2
       valid_lft forever preferred_lft forever
    inet6 fe80::218:7dff:fe38:f09e/64 scope link proto kernel_ll 
       valid_lft forever preferred_lft forever
13: 6in4-omr6in4@NONE: <POINTOPOINT,NOARP,UP,LOWER_UP> mtu 1280 qdisc noqueue state UNKNOWN group default qlen 1000
    link/sit 10.255.252.6 peer 10.255.252.1
    inet6 fd00::a00:2/126 scope global 
       valid_lft forever preferred_lft forever
    inet6 fe80::aff:fc06/64 scope link 
       valid_lft forever preferred_lft forever
26: ifb4eth1: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 66:f7:2b:1b:21:e8 brd ff:ff:ff:ff:ff:ff
29: ifb4eth2: <BROADCAST,NOARP> mtu 1500 qdisc noop state DOWN group default qlen 32
    link/ether 9a:32:59:85:18:5f brd ff:ff:ff:ff:ff:ff
31: tun0: <POINTOPOINT,MULTICAST,NOARP,UP,LOWER_UP> mtu 1500 qdisc fq_codel state UNKNOWN group default qlen 100
    link/none 
    inet 10.255.252.3/24 scope global tun0
       valid_lft forever preferred_lft forever
    inet6 fe80::3497:7cf0:f87e:fc4d/64 scope link stable-privacy proto kernel_ll 
       valid_lft forever preferred_lft forever

KoulEl, 2025-03-25:
Here are the results from the VPS (hanks Milkywan):

Le ping ne passe pas

ping 10.255.252.6

PING 10.255.252.6 (10.255.252.6) 56(84) bytes of data.

^C

--- 10.255.252.6 ping statistics ---

16 packets transmitted, 0 received, 100% packet loss, time 15355ms

iptables-save

# Generated by iptables-save v1.8.9 (nf_tables) on Mon Mar 24 20:06:26 2025

*mangle

:PREROUTING ACCEPT [97659619:54649998883]

:INPUT ACCEPT [76730665:45965484733]

:FORWARD ACCEPT [20779327:8662818063]

:OUTPUT ACCEPT [74380277:80920123570]

:POSTROUTING ACCEPT [95159603:89582941589]

:tcfor - [0:0]

:tcin - [0:0]

:tcout - [0:0]

:tcpost - [0:0]

:tcpre - [0:0]

-A PREROUTING -j tcpre

-A INPUT -j tcin

-A FORWARD -j MARK --set-xmark 0x0/0xff

-A FORWARD -j tcfor

-A OUTPUT -j tcout

-A POSTROUTING -j tcpost

COMMIT

# Completed on Mon Mar 24 20:06:26 2025

# Generated by iptables-save v1.8.9 (nf_tables) on Mon Mar 24 20:06:26 2025

*raw

:PREROUTING ACCEPT [97659619:54649998883]

:OUTPUT ACCEPT [74380277:80920123570]

-A PREROUTING -p udp -m udp --dport 10080 -j CT --helper amanda

-A PREROUTING -p tcp -m tcp --dport 21 --tcp-flags FIN,SYN,RST,ACK SYN -j CT --helper ftp

-A PREROUTING -p udp -m udp --dport 1719 -j CT --helper RAS

[... 788 lines cut ...]
-A net_dnat -p tcp -m tcp --dport 22 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 8080 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 63241 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 22027 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 22028 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 22029 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 4445:4446 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 16881 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p udp -m udp --dport 16881 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 22026 -j DNAT --to-destination 10.255.252.6

-A net_dnat -p tcp -m tcp --dport 80 -j DNAT --to-destination 10.255.252.6

COMMIT

# Completed on Mon Mar 24 20:06:26 2025

Ysurac, 2025-03-25:
Oui l'IP VPN côté client a changé et le VPS n'est pas à jour. J'ai fait un fix que je test avant de le mettre dans la branche develop.

KoulEl, 2025-03-27:
Waiting for fixes (VPS and x86_64 router), I had to revert to using my Dual Wan router for the time being (I need to access my websites).

No hurry, I can wait :)

Ysurac, 2025-03-27:
It's in current snapshot.
Else you can run ```/etc/init.d/openmptcprouter-vps restart``` via SSH on the router. If it's not working then the problem is something else...

KoulEl, 2025-03-27:
VPS has been rebooted by Milkywan but still no luck (still no access to my sites). Furthermore, I keep getting these errors now:

I think I will ask MilkyWan to donwgrade to the initial version in order to check if I can access my sites again (and if yes, we'll have to find what is wrong).

KoulEl, 2025-03-28:
Reverted to VPS version 0.1031 5.4.207 and all my sites are now working.