Ysurac/openmptcprouter #3869: Port Forwarding - So many questions, but only so little documentation

MRedZac, 2025-03-22:
Good day,

first of all, a small description, on how everything here is setup:

I have a couple of websites hosted here locally on different linux servers. 1 server per website. 

Previously, I just had a Teltonika RUTX50 router behind a FTTH connection with fixed ip address. The domains are hosted with Cloudflare, in the DNS, the fixed IP from the FTTH was setup as an A-Record and for www. Everything was working correctly, as expected. 
Now I changed the router with an OpenMPTCPRouter based on an x86-64 computer, setup a VPS server with IONOS and bonded for now the Fiber with an o2 LTE/5G connection. So far, so good, internet is working, bonding is working, VPN from local router to VPS is established fine. Both is working: OpenVPN or GloryTun - I can switch these in the advanced settings. I changed DNS Records at Cloudflare to the fixed IP of the Ionos VPS Server. The LAN IP of OpenMPTCProuter has been set to 192.168.178.1 to match the network address range.
The issue now comes with the Port forwarding setup.

## Expected Behavior
I expect to have ports 80 and 443 TCP forwarded to 192.168.178.246 in the internal network, where an NGINX is waiting to handle these requests. But with the existing procedure (https://github.com/Ysurac/openmptcprouter/wiki/Port-forwarding), I am not able to get this to work.

## Current Behavior
Websites are not shown, as the hosts cannot be found. I suppose due to requests are not reaching the NGINX on internal IP 192.168.178.246. 

## Specifications

  - OpenMPTCProuter version: v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: 0.1032.6.12.15-x64v3-xanmod1
  - OpenMPTCProuter VPS provider: IONOS
  - OpenMPTCProuter platform: x86-64

Can somebody please make a YouTube video on this topic ? - Following the mentioned procedure, it leaves too many question open. The debug part is soooo short and for non-linux users somewhat undefined, that I doubt most of people do not even know, where to do what and especially what to look for !

For example, this is my rule set on VPS, but how shall I know, what has to be in there, so that the ports needed would be forwarded...?

[image]

It is also not clear, what has to be the result of the param.vpn... Does it matter, WHICH VPN connection is established (OpenVPN or GloryTun) ? The procedure mentions, that GloryTun is the standard, but after installation, I think, that the default setting was an OMR connection with OpenVPN TCP. 

[image]

And why in the procedure Port Forwarding for port 8000 instead of 80 ? - And which Port to choose, when 443 is needed to be forwarded ?

And what to do, or better how to do the port forwarding, when NGINX is already setup within the network and the requests have to go to this internal server ? I don't want to setup NGINX again on the VPS, as it has dozens of entries and certificates...

And what to setup to forward UDP Port 5000 to another internal host for a streaming connection on an SRTLA Relay Server ? 

And what to set for the standard proxy in VPN Settings, when UDP is needed besides TCP ?

I really think, it needs a much more detailed procedure for this port forwarding topic than what is out there now. As requested, I think the best would be a YouTube video, where everything would be shown how it has to be done and especially how a proper debug is done to see which entries have to be there, etc.

Anybody ?

Thanks in advance for sharing your precious knowlede ! :-)

Cheers
Marco

KoulEl, 2025-03-22:
Maybe you have the same problem I have at the moment (see previous issue). It used to worked but not anymore since last VPS upgrade (I think).

I have 2x PeerTube instances, 1x Jitsi instance and 1x Owncloud instance. All were working fine with the correct NAT/FireWall settings but not anymore (without changing anything in my OpenMPTCP configuration, only the server has been upgraded).

MRedZac, 2025-03-22:
@KoulEl: I read somewhere, that local Router and VPS Server always have to be updated at the same time. Updating only one would lead to issues... But for me, I installed everything fresh 2 days ago with the latest available versions... If there is a bug in the lastest version, then yes, that would explain the behavior, but then maybe this would have been mentioned already more often, no ?

KoulEl, 2025-03-22:
@MRedZac Only if we are many hosting our own websites but I don't think it's the case. Let's wait for Ycarus to tell us what he needs to check whether everything is ok or not.

Ysurac, 2025-03-22:
Documentation show an example to open port 80 of a LAN server from port 8000 of the VPS.
A bug is possible, but for that I need to know where. It's why there is a debug section to help find what is not working.
Did you run tcpdump command on the router ? In the config here I can see only port 5000 UDP forwarded.

Every forwarded port use the VPN (if proxy is not V2Ray or XRay and the case not used). Can work with any VPN.

MRedZac, 2025-03-22:
tcpdump on router for port 443 brings a ton of activity, More than 40000 packets in under 30 seconds... For port 80 there are no packages... 

To me it is pretty obvious that the bug is, that it does not reflect the rules in local router on the server in the rules file... I just opened ALL ports tcp/udp manually in this file now by removing the # in front of the DNAT at the bottom where it was for TCP and UDP for all Ports from 1-64999... I changed 64999 to 59999, saved, restarted the server and tadaaaa, all of a sudden, the port forwards set up on local router were also reflected in this file. So it looks like, the VPS or local router blocks itself somewhere on the way for the rules exchange, so that the local rules cannot be written into the rules file on VPS. Only when you manually open all the ports, TCP and UDP from 1-59999, the rules are transfered from local to vps.

[image]

Ysurac, 2025-03-22:
Seems to be an error in API, I will check.

MRedZac, 2025-03-26:
Well, after days of trying, finally, I was able to make it. I reformated the VPS server and installed it completely fresh, I wiped the SSD in the x86-64 computer and started completely from the beginning, only making those changes which were really nescessary and TADAAAA !!! - Works. Don't even need an snat in the firewall rules, just opened all ports to be let trough to the router and it is fine. That is not problematic, I think, as IONOS puts a firewall in front of the VPS as well and only the ports above 60000 and those really needed below are open there... It was really hard to find out what to configure and how, but finally, I was able to bond a FTTH, a Starlink and a 5G connection together as expected. Thanks for your support on that case ! :-)

closed 2025-03-26 by MRedZac