Ysurac/openmptcprouter #3870: Hide.me VPN Service to be added to OMR exceptions list

MRedZac, 2025-03-22:
**Is your feature request related to a problem? Please describe.**
Hide.me VPN Service not connecting.

**Describe the solution you'd like**
Add hide.me VPN Service to the list of OMR Exeptions

Ysurac, 2025-03-26:
Only hide.me domain is blocking VPS or all hide.me VPN access are blocking VPS IP ?

MRedZac, 2025-03-26:
No no, it is the whole service which is blocked to connect behind the VPN Tunnel to VPS... Not just the hide.me domain... If I define an exeption by source mac and choose one network adapter with internet directly it is connecting, but behind the tunnel to VPS, it is not able to make ANY connection to any of their servers... As all their servers are something like XX.hideservers.net, maybe it would already be enough to define this domain somehow to be let through...

Here is the list of their servers:

Albania	 	al.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Argentina	 	ar.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Österreich	 	at.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Australien	 	au.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Australia - Melbourne	 	melbourne.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Australia - Perth	 	perth.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Australia - Sydney	 	sydney.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Bangladesh	 	bd.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Belgium	 	be.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Brasil	 	br.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Bulgaria	 	bg.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Cambodia	 	kh.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Kanada	 	ca.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Canada - Calgary	 	calgary.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Canada - Montreal	 	montreal.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Canada - Toronto	 	toronto.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Canada - Vancouver	 	vancouver.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Colombia	 	co.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Chile	 	cl.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Croatia	 	hr.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Czech Republic	 	cz.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Denmark	 	dk.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Egypt	 	eg.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Finland	 	fi.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Frankreich	 	fr.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
France - Marseille	 	marseille.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
France - Paris	 	paris.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Deutschland	 	de.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Germany - Frankfurt #1	 	frankfurt1.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Germany - Frankfurt #2	 	frankfurt2.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Greece	 	gr.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Hong-Kong	 	hk.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Hungary	 	hu.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Island	 	is.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Indonesia	 	id.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Ireland	 	ie.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Italien	 	it.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Italy - Milan	 	milan.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Italy - Palermo	 	palermo.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Japan	 	jp.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Litauen	 	lt.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Luxembourg	 	lu.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Malaysia	 	my.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Mexiko	 	mx.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Moldova	 	md.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Marokko	 	ma.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Niederlande	 	nl.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Netherlands - Amsterdam #1	 	amsterdam1.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Netherlands - Amsterdam #2	 	amsterdam2.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
New Zealand	 	nz.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Norway	 	no.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Peru	 	pe.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
[... cut ...]
United States - Texas	 	houston.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
United States - Utah	 	saltlakecity.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
United States - Washington	 	washington-dc.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
United States - Washington	 	seattle.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)
Vietnam	 	vn.hideservers.net 	[+ Mehr Details](https://member.hide.me/de/server-status#)

MRedZac, 2025-03-26:
I've entered *.hideservers.net as a domain in the OMR Exceptions and can connect to hide.me service now without issues (except it is very slow...)

Ysurac, 2025-03-26:
Simply put hideservers.net, not need for the *. Should be at the speed of your selected connection. There is no aggregation when bypassing. Bad choice of hide.me to bans some IPs here...

MRedZac, 2025-03-26:
Hide.me is not banning IPs, it is just recognizing the VPN already in place and does not tunnel a vpn through an existing vpn...