Ysurac/openmptcprouter #3871: Disabling automatic server updates

intersvod, 2025-03-23:
## Expected Behavior

My VPS OpenMPTCProuter server updates itself

## Current Behavior

[image]
[image]

Instead of just working on version 0.1032 5.4.207-mptcp, the server after 1-2 days automatically updates to version 0.1032 6.1.0-32-amd64. As a result, I get the message: "MPTCP kernel on router and VPS doesn't match" and the system does not work properly. I did not press the update button in the router's WEB interface. I tried to reflash the router - it does not help. There is nothing special in the router logs. I also tried to reinstall the server many times. Nothing helps.
How can I disable or prohibit OpenMPTCProuter server updates?

[image]
[image]

## Specifications

  - OpenMPTCProuter version: 0.61-5.4
  - OpenMPTCProuter VPS version: 0.1032 5.4.207-mptcp
  - OpenMPTCProuter VPS provider: Hetzner
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-03-23:
Automatic router or VPS update is not implemented.

intersvod, 2025-03-23:
Then maybe update-grub.sh does not work correctly and the MPTCP kernel changes to the standard Debian 12 system after rebooting the VPS? I forgot to mention that the problem seems to occur after rebooting the VPS. What are your suggestions on how the version 0.1032 5.4.207-mptcp could spontaneously change to 0.1032 6.1.0-32-amd64?

intersvod, 2025-03-23:
I definitely don't take any action and I encounter this problem: "MPTCP kernel on router and VPS does not match"

Ysurac, 2025-03-23:
A Debian update of the kernel, so 6.1.0 was used after reboot.
You can install again script with KERNEL="5.4" parameter (script 0.1032 was never tested with kernel 5.4).

intersvod, 2025-03-23:
So this is a spontaneous update of the Debian kernel? But, as usual, I installed the server part with the command:
wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="5.4" sh

If I reuse this command, the kernel will not change spontaneously anymore?
When was the script version 0.1032 released?

I have many OpenMPTCProuter devices and servers in work and I have never encountered such a problem before. But on March 4, having installed the server part, as I usually did, I encountered such a problem on a new server.
Could this be due to a bug with the 5.4 kernel in the script version 0.1032?

How can I force the script to use version 0.1028, for example? Could you give an example command?
Thanks in advance.

Ysurac, 2025-03-23:
This may be updated if you have set ```unattended-upgrades``` package.
Nothing related to my script. You may remove debian kernel packages via apt-get.

As 5.4 is not supported anymore no tests and nothing is done to support it, server script may also stop supporting 5.4.

All scripts (current and olders) are available here: https://github.com/Ysurac/openmptcprouter-vps (tagged on version) but old script can not work anymore (if a link is not working for example).

intersvod, 2025-03-23:
Do I understand correctly that if I want to install the server with the v0.1026 script, the command will look like this: wget -O https://github.com/Ysurac/openmptcprouter-vps/tree/v0.1026/debian-x86_64.sh | KERNEL="5.4" sh
?

vempire-ghost, 2025-03-23:
> So this is a spontaneous update of the Debian kernel? But, as usual, I installed the server part with the command: wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="5.4" sh
> 
> If I reuse this command, the kernel will not change spontaneously anymore? When was the script version 0.1032 released?
> 
> I have many OpenMPTCProuter devices and servers in work and I have never encountered such a problem before. But on March 4, having installed the server part, as I usually did, I encountered such a problem on a new server. Could this be due to a bug with the 5.4 kernel in the script version 0.1032?
> 
> How can I force the script to use version 0.1026, for example? Could you give an example command? Thanks in advance.

You need to disable automatic updates on Debian, otherwise, it will keep updating the kernel.  

Run this command:  
```bash
sudoedit /etc/apt/apt.conf.d/20auto-upgrades
```  

Replace its contents with the following:  
```plaintext
APT::Periodic::Update-Package-Lists "0";  
APT::Periodic::Download-Upgradeable-Packages "0";  
APT::Periodic::AutocleanInterval "0";  
APT::Periodic::Unattended-Upgrade "0";  
```  

Then, execute this command:  
```bash
sudo systemctl disable --now unattended-upgrades
```  

This way, Debian will not automatically perform any updates. It will be up to you to manually update whatever you find necessary.

intersvod, 2025-03-24:
> > То есть это спонтанное обновление ядра Debian? Но, как обычно, я установил серверную часть командой: wget -O - https://www.openmptcprouter.com/server/debian-x86_64.sh | KERNEL="5.4" sh
> > Если я повторно использую эту команду, ядро ​​больше не будет меняться спонтанно? Когда была выпущена версия скрипта 0.1032?
> > У меня в работе много устройств и серверов OpenMPTCProuter и раньше я с такой проблемой не сталкивался. Но 4 марта, установив серверную часть, как я обычно это делаю, я столкнулся с такой проблемой на новом сервере. Может ли это быть связано с ошибкой с ядром 5.4 в версии скрипта 0.1032?
> > Как заставить скрипт использовать версию 0.1026, например? Можете привести пример команды? Заранее спасибо.
> 
> Вам необходимо отключить автоматические обновления в Debian, в противном случае ядро ​​будет постоянно обновляться.
> 
> Выполните эту команду:
> 
> sudoedit /etc/apt/apt.conf.d/20auto-upgrades
> Замените его содержимое следующим:
> 
> ```
> APT::Periodic::Update-Package-Lists "0";  
> APT::Periodic::Download-Upgradeable-Packages "0";  
> APT::Periodic::AutocleanInterval "0";  
> APT::Periodic::Unattended-Upgrade "0";  
> ```
> 
> Затем выполните эту команду:
> 
> sudo systemctl disable --now unattended-upgrades
> Таким образом, Debian не будет автоматически выполнять никаких обновлений. Вам придется вручную обновлять все, что вы считаете необходимым.

Is it not enough to simply remove the unattended-upgrades package?

intersvod, 2025-03-24:
I also removed all unnecessary kernels by doing the following:

uname -r

dpkg --list | grep linux-image

sudo apt purge linux-image-6.1.0-29-amd64

sudo apt purge linux-image-6.1.0-31-amd64

sudo apt purge linux-image-6.1.0-22-amd64

sudo apt purge linux-image-amd64

sudo apt autoremove

sudo update-grub

dpkg --list | grep linux-image

reboot

uname -r

I will monitor the further behavior of the system. Thanks .

Ysurac, 2025-03-26:
Maybe a good idea to add in VPS script if I can add this without major problems.

intersvod, 2025-03-26:
> Возможно, было бы неплохо добавить скрипт VPS, если это удастся сделать без особых проблем.

It would be great. Since the problem was solved, the kernel no longer updates spontaneously and the MPTCP server works stably.