Ysurac/openmptcprouter #3872: How it works

hidemyname5, 2025-03-23:
Good afternoon. I would like to understand a few things. by default, OMR uses Shadowsocks + MPTCP for channel aggregation. this only works for TCP. all traffic goes through Shadowsocks. This achieves the best performance. if the traffic contains UDP, then you need to configure Glorytun UDP or MLVPN. Aggregation will also work, only by means of these protocols. will be less efficient. 
Is this correct?

Ysurac, 2025-03-23:
There is Shadowsocks-Rust, Shadowsocks-libev, V2Ray or XRay as proxy for TCP and Glotytun-TCP/UDP, OpenVPN, MLVPN, DSVPN and V2Ray/Xray for UDP (in some protocol they can do UDP over TCP).
UDP over TCP is slower than TCP.

hidemyname5, 2025-03-23:
I am using 3 WAN connection. On the VPS side I see 3 connections from GloryTun, but none from Shadowsocks. Is it supposed to be like that? Maybe there is a diagnostic that will show that everything is fine? P.S. I don't see any problems or errors.

Ysurac, 2025-03-24:
No problem so no issue. Status page is the diagnostic page.

closed 2025-03-26 by Ysurac