Ysurac/openmptcprouter #3873: Speed issue with 2+ 4G/5G router

VisionLabGroup, 2025-03-24:
Hello, 
I have 3 internet lines connected, but when all three are connected the speed is lower than when 2 are connected. They are all three in the same position, same sims, same latency.

How can I solve this? With 2 lines the download speed is 600 Mbps, with 2 130 Mbps.

## Specifications

  - OpenMPTCProuter version: openmptcprouter v0.61-6.6 r0+27346-c7ba5574f5
  - OpenMPTCProuter VPS version: 6.6.36-x64v2-xanmod1 0.1032
  - OpenMPTCProuter VPS provider: Host.it
  - OpenMPTCProuter platform: x86_64

Ysurac, 2025-03-24:
If you use same SIMs, so same provider, the bandwidth is shared between modems, never a good idea.
For the speed with 3 modems, shouldn't be lower, try again after some times.

sympavl, 2025-03-24:
@Ysurac To not create a new topic I'd like to ask. I'm running openmptcp on rpi5 with two 5g modems connected via usb 3.0 adapter. I'm using sim cards of 2 different providers but based on the same tower. With omr-test-speed for a single of modems I'm getting 200Mbps minimum so in teory I should get around 400 mbit and in reality around 300+ mbit? I'm getting 230 Mbps max using a vps with close location capable of 3gbit/s. To verify that I'm not limited by vps I've added additional wifi connection that gives clear 300mbit/s and I was seeing speed of 500 Mbps/s. Any advices on that?

Ysurac, 2025-03-26:
Are you using shadowsocks-libev or Shadowsocks-rust ?
Did you try omr-test-speed on both modems at the same time ?