Ysurac/openmptcprouter #3876: segfaulting libcrypto.so on x86_64 vm

Maverick78de, 2025-03-26:
Copy from #3762 

I have the issue segfaulting libcrypto.so and sometime crash the whole machine.
I installed it (new installation) in a VM withing proxmox used the vmdk openmptcprouter-v0.62-6.6-r0+28431-92e020b50f-x86-64-generic-ext4-combined.vmdk

(cpuinfo: Intel Celeron N Series 3160
flags : fpu vme de pse tsc msr pae mce cx8 apic sep mtrr pge mca cmov pat pse36 clflush dts acpi mmx fxsr sse sse2 ss ht tm pbe syscall nx rdtscp lm c onstant_tsc arch_perfmon pebs bts rep_good nopl xtopology tsc_reliable nonstop_t sc cpuid aperfmperf tsc_known_freq pni pclmulqdq dtes64 monitor ds_cpl vmx est t m2 ssse3 cx16 xtpr pdcm sse4_1 sse4_2 movbe popcnt tsc_deadline_timer aes rdrand lahf_lm 3dnowprefetch epb pti ibrs ibpb stibp tpr_shadow flexpriority ept vpid tsc_adjust smep erms dtherm ida arat vnmi md_clear
vmx flags : vnmi preemption_timer invvpid ept_x_only flexpriority tsc_offs et vtpr mtf vapic ept vpid unrestricted_guest
bugs : cpu_meltdown spectre_v1 spectre_v2 mds msbds_only mmio_unknown
)

I tried different types of virtual CPU for the VM, without any luck. After spending several hours I disabled AES support for the VM and now it is working without error or crash. But, a big but, the throughput over Shadowsocks and other VPN Type is really really poor, I can't get not more then 80Mbps because of high CPU usage. (a little bit more with chacha20 used). My old rpi4 (1st series) is 2 times faster.

So there seems to be an issue with libcrypto.so and AES support in the current stable version for x86_64.

Would be great to get it fixed because I want to relace the rpi4 soon.

- OpenMPTCProuter version: v0.62-6.6
- OpenMPTCProuter platform: x86_64

Next step trying snapshot version coming days.

Ysurac, 2025-03-26:
What proxy and VPN are you using when AES enabled ?

Maverick78de, 2025-03-27:
I've tried both Shadowsocks, OpenVPN and GloryVPN.

Maverick78de, 2025-04-02:
similar issue with snapshot release -v0.63-snapshot-6.12-r0+28753-f480fa60d3-x86-64-generic-ext4-combined.vmdk (new setup)
After setting up WAN connectivity and VPS configuration the whole VM is going in a hard reboot loop, so I'm unable to make any screenshots or get any logs, because the system restarts immediately. 
If I disable AES support on CPU, everything is working, but slow.

closed 2025-07-06 by github-actions[bot]