Ysurac/openmptcprouter #3879: After unplugging the network cable and plugging it back in, it did not join the aggregation path. After closing the current application and restarting it, it was able to join again. Please help me figure out what the problem is.

lhc1993, 2025-03-28:
## Expected Behavior

Unplug the network cable and then plug it back in
## Current Behavior

Mar 28 15:28:29 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 28 15:28:31 OpenMPTCProuter user.notice post-tracking-003-up: Tunnel up : Replace default route by 10.255.252.1 dev tun0 (was 10.180.140.70)
Mar 28 15:28:32 OpenMPTCProuter user.notice post-tracking-020-status: New public ip detected for omrvpn (tun0): 123.57.0.66 (First public IP detected)
Mar 28 15:28:33 OpenMPTCProuter daemon.notice openvpn(omr)[8136]: Initialization Sequence Completed
Mar 28 15:34:58 OpenMPTCProuter daemon.err uhttpd[11496]: [info] luci: accepted login on /admin/network/mptcp/bandwidth for admin from 192.168.9.134
Mar 28 15:35:15 OpenMPTCProuter daemon.notice netifd: Network device 'eth2' link is up
Mar 28 15:35:15 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' has link connectivity 
Mar 28 15:35:15 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is setting up now
Mar 28 15:35:15 OpenMPTCProuter daemon.notice netifd: wan3 (29878): udhcpc: started, v1.36.1
Mar 28 15:35:15 OpenMPTCProuter daemon.notice netifd: wan3 (29878): udhcpc: broadcasting discover
Mar 28 15:35:18 OpenMPTCProuter daemon.notice netifd: wan3 (29878): udhcpc: broadcasting discover
Mar 28 15:35:19 OpenMPTCProuter daemon.notice netifd: wan3 (29878): udhcpc: broadcasting select for 192.168.100.120, server 192.168.100.1
Mar 28 15:35:19 OpenMPTCProuter daemon.notice netifd: wan3 (29878): udhcpc: lease of 192.168.100.120 obtained from 192.168.100.1, lease time 43200
Mar 28 15:35:19 OpenMPTCProuter daemon.notice netifd: Interface 'wan3' is now up
Mar 28 15:35:19 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='ifb4eth2'
Mar 28 15:35:19 OpenMPTCProuter daemon.info ModemManager[31229]: hotplug: remove network interface ifb4eth2: event processed
Mar 28 15:35:19 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_25d9d'
Mar 28 15:35:19 OpenMPTCProuter daemon.info ModemManager[31431]: hotplug: add network interface SQM_IFB_25d9d: event processed
Mar 28 15:35:19 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_25d9d'
Mar 28 15:35:19 OpenMPTCProuter daemon.info ModemManager[31573]: hotplug: remove network interface SQM_IFB_25d9d: event processed
Mar 28 15:35:19 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='SQM_IFB_c4ce9'
Mar 28 15:35:20 OpenMPTCProuter user.notice firewall: Reloading firewall due to ifup of wan3 (eth2)
Mar 28 15:35:20 OpenMPTCProuter daemon.info ModemManager[31634]: hotplug: add network interface SQM_IFB_c4ce9: event processed
Mar 28 15:35:20 OpenMPTCProuter user.notice NET: hotplug (iface): action='remove' interface='SQM_IFB_c4ce9'
Mar 28 15:35:20 OpenMPTCProuter daemon.info ModemManager[31808]: hotplug: remove network interface SQM_IFB_c4ce9: event processed
Mar 28 15:35:20 OpenMPTCProuter user.notice NET: hotplug (iface): action='add' interface='ifb4eth2'
Mar 28 15:35:20 OpenMPTCProuter daemon.info ModemManager[31836]: hotplug: add network interface ifb4eth2: event processed
Mar 28 15:35:21 OpenMPTCProuter user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mar 28 15:35:26 OpenMPTCProuter user.notice post-tracking-003-up: wan3 (eth2) switched up
## Specifications

  - OpenMPTCProuter version: 0.61rc1
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

vempire-ghost, 2025-03-29:
I have had this issue with the 5.4 kernel for years. Apparently, it is a problem with MPTCP and not OMR, as you can see in this issue:  
[https://github.com/Ysurac/openmptcprouter/issues/3122](https://github.com/Ysurac/openmptcprouter/issues/3122).

Apparently, the issue persists in the current kernel. Try these commands to see if they help. In kernel 5.4, they restore subflows that were broken due to a drop or timeout.  

```bash
multipath eth off
# Wait a few seconds
multipath eth on
```  

Replace `eth` with the corresponding interface that you disconnected, where the subflows were not recreated.

And for this issue, I believe the team at [https://github.com/multipath-tcp/mptcp_net-next](https://github.com/multipath-tcp/mptcp_net-next) will have more possibilities to solve the problem of subflows not being recreated.

lhc1993, 2025-03-30:
Thank you for your answer. I will try your method to see if it can solve the problem that the substream cannot be recreated.

mvmaksnet, 2025-03-31:
I also had this problem in past and also in present with the newest version of router and vps.

Ysurac, 2025-03-31:
It's possible there is multiples issues, can you check latest snapshot and give me logs ?
In current log, it's missing some lines after "post-tracking-003-up..." like a "Multipath eth2 (wan3) switched to on..." (and if this line is not here, it's a problem)

lhc1993, 2025-03-31:
Mon Mar 31 17:28:10 2025 daemon.notice netifd: Network device 'wwan0' link is up
Mon Mar 31 17:28:10 2025 daemon.notice netifd: Interface 'wwan0' has link connectivity
Mon Mar 31 17:28:10 2025 daemon.notice netifd: Interface 'wwan0' is setting up now
Mon Mar 31 17:28:10 2025 daemon.notice netifd: wwan0 (17056): udhcpc: started, v1.36.1
Mon Mar 31 17:28:10 2025 daemon.notice netifd: wwan0 (17056): udhcpc: broadcasting discover
Mon Mar 31 17:28:10 2025 daemon.notice netifd: wwan0 (17056): udhcpc: broadcasting select for 10.70.47.240, server 10.70.47.241
Mon Mar 31 17:28:11 2025 daemon.notice netifd: wwan0 (17056): udhcpc: lease of 10.70.47.240 obtained from 10.70.47.241, lease time 7200
Mon Mar 31 17:28:11 2025 daemon.notice netifd: Interface 'wwan0' is now up
Mon Mar 31 17:28:11 2025 user.notice firewall: Reloading firewall due to ifup of wwan0 (wwan0)
Mon Mar 31 17:28:12 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon Mar 31 17:28:13 2025 user.notice mptcp: Reloading mptcp config due to ifup of wwan0 (wwan0)
Mon Mar 31 17:28:14 2025 user.notice post-tracking-003-up: wwan0 (wwan0) switched up
Mon Mar 31 17:28:15 2025 user.notice post-tracking-003-up: Status change, reload MPTCP config for wwan0
Mon Mar 31 17:28:20 2025 user.notice post-tracking-003-up: Local IP changed for wwan0 (wwan0) reload Multipath
Mon Mar 31 17:28:20 2025 user.notice post-tracking-003-up: Reload MPTCP config for wwan0
Mon Mar 31 17:28:22 2025 user.notice post-tracking-020-status: New public ip detected for wwan0 (wwan0): 124.127.70.49 (previous: 124.127.66.107)
Mon Mar 31 17:28:23 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon Mar 31 17:28:23 2025 user.notice omr-bypass: Starting OMR-ByPass...
Mon Mar 31 17:28:31 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon Mar 31 17:28:32 2025 user.notice omr-bypass: OMR-ByPass is running
Mon Mar 31 17:28:36 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Mon Mar 31 17:29:01 2025 user.notice omr-schedule-010-services: Set firewall on server vps

Ysurac, 2025-04-08:
@mvmaksnet please top that without putting any log and first testing snapshot (and wait...).

mvmaksnet, 2025-04-08:
Sorry, I will do that. When you asked in previous messages for logs I taught that you wanted logs from @lhc1993 user.

Ysurac, 2025-04-08:
In fact both, more logs is better.
Both of you, Can you also give the result of "multipath" and "ip mptcp endpoint" output via SSH on the router when you have the issue ? (on latest snapshot)

lhc1993, 2025-04-08:
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.023074] dp5: Switch attached to macid 5 status: 0
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.028511] Generic PHY 90000.mdio-1:04: attached PHY driver (mii_bus:phy_addr=90000.mdio-1:04, irq=POLL)
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.033680] **********************************************************
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.042639] * NSS Data Plane driver
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.049035] **********************************************************
Tue Apr  8 16:00:53 2025 kern.debug kernel: [    9.061636] libata version 3.00 loaded.
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.084911] xhci-hcd xhci-hcd.1.auto: xHCI Host Controller
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.084960] xhci-hcd xhci-hcd.1.auto: new USB bus registered, assigned bus number 1
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.089418] xhci-hcd xhci-hcd.1.auto: hcc params 0x0220fe65 hci version 0x110 quirks 0x0000008002000010
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.096903] xhci-hcd xhci-hcd.1.auto: irq 58, io mem 0x08a00000
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.106333] xhci-hcd xhci-hcd.1.auto: xHCI Host Controller
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.112125] xhci-hcd xhci-hcd.1.auto: new USB bus registered, assigned bus number 2
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.117735] xhci-hcd xhci-hcd.1.auto: Host supports USB 3.0 SuperSpeed
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.125728] hub 1-0:1.0: USB hub found
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.131856] hub 1-0:1.0: 1 port detected
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.136006] usb usb2: We don't know the algorithms for LPM for this host, disabling LPM.
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.140079] hub 2-0:1.0: USB hub found
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.147761] hub 2-0:1.0: 1 port detected
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.151662] xhci-hcd xhci-hcd.2.auto: xHCI Host Controller
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.155398] xhci-hcd xhci-hcd.2.auto: new USB bus registered, assigned bus number 3
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.160773] xhci-hcd xhci-hcd.2.auto: hcc params 0x0220fe65 hci version 0x110 quirks 0x0000008002000010
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.168245] xhci-hcd xhci-hcd.2.auto: irq 59, io mem 0x08c00000
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.177677] xhci-hcd xhci-hcd.2.auto: xHCI Host Controller
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.183479] xhci-hcd xhci-hcd.2.auto: new USB bus registered, assigned bus number 4
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.189046] xhci-hcd xhci-hcd.2.auto: Host supports USB 3.0 SuperSpeed
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.197234] hub 3-0:1.0: USB hub found
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.203224] hub 3-0:1.0: 1 port detected
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.207379] usb usb4: We don't know the algorithms for LPM for this host, disabling LPM.
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.211364] hub 4-0:1.0: USB hub found
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.219129] hub 4-0:1.0: 1 port detected
Tue Apr  8 16:00:53 2025 kern.err kernel: [    9.224314] leds-gpio leds: LED color identifier out of range
Tue Apr  8 16:00:53 2025 kern.warn kernel: [    9.226718] leds-gpio leds: LED system color identifier out of range
Tue Apr  8 16:00:53 2025 kern.err kernel: [    9.232431] leds-gpio leds: LED color identifier out of range
Tue Apr  8 16:00:53 2025 kern.warn kernel: [    9.238795] leds-gpio leds: LED wlan2g color identifier out of range
Tue Apr  8 16:00:53 2025 kern.err kernel: [    9.244488] leds-gpio leds: LED color identifier out of range
Tue Apr  8 16:00:53 2025 kern.warn kernel: [    9.250840] leds-gpio leds: LED wlan5g color identifier out of range
Tue Apr  8 16:00:53 2025 user.info kernel: [    9.257619] kmodloader: done loading kernel modules from /etc/modules-boot.d/*
Tue Apr  8 16:00:53 2025 user.info kernel: [    9.263827] init: - preinit -
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.433890] usb 1-1: new high-speed USB device number 2 using xhci-hcd
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.503868] usb 3-1: new high-speed USB device number 2 using xhci-hcd
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.740832] hub 3-1:1.0: USB hub found
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.740935] hub 3-1:1.0: 7 ports detected
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.863922] usb 4-1: new SuperSpeed USB device number 2 using xhci-hcd
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.932816] hub 4-1:1.0: USB hub found
Tue Apr  8 16:00:53 2025 kern.info kernel: [    9.932913] hub 4-1:1.0: 6 ports detected
Tue Apr  8 16:00:53 2025 kern.info kernel: [   10.533859] usb 3-1.1: new high-speed USB device number 3 using xhci-hcd
Tue Apr  8 16:00:53 2025 user.info kernel: [   10.657952] mount_root: loading kmods from internal overlay
Tue Apr  8 16:00:53 2025 user.info kernel: [   10.680331] kmodloader: loading kernel modules from //etc/modules-boot.d/*
Tue Apr  8 16:00:53 2025 user.info kernel: [   10.683283] kmodloader: done loading kernel modules from //etc/modules-boot.d/*
Tue Apr  8 16:00:53 2025 kern.info kernel: [   10.933871] usb 3-1.7: new high-speed USB device number 4 using xhci-hcd
[... cut ...]
Tue Apr  8 16:22:03 2025 user.notice post-tracking-003-up: wan3 (eth2) switched up
Tue Apr  8 16:22:04 2025 user.notice post-tracking-003-up: Status change, reload MPTCP config for eth2
Tue Apr  8 16:22:08 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue Apr  8 16:22:09 2025 user.notice omr-bypass: OMR-ByPass is running
Tue Apr  8 16:22:15 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Tue Apr  8 16:23:01 2025 user.notice omr-schedule-010-services: Set firewall on server vps

lhc1993, 2025-04-08:
The operation is eth2

Ysurac, 2025-04-09:
@lhc1993 Can you also give the result of "multipath" and "ip mptcp endpoint" output via SSH on the router when you have the issue ? (on latest snapshot)

Ysurac, 2025-04-10:
Can you try latest snapshot ?

lhc1993, 2025-04-10:
**The result after unplugging eth2**
**root@OpenMPTCProuter:~# multipath**
br-lan is deactivated
erspan0 is deactivated
eth0 is deactivated
eth1 is deactivated
eth2 is in default mode
eth3 is deactivated
eth4 is deactivated
gre0 is deactivated
gretap0 is deactivated
ifb4eth1 is deactivated
ifb4eth2 is deactivated
ifb4eth3 is deactivated
ifb4eth4 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
miireg is deactivated
pcie_mhi0 is in default mode
pcie_mhi1 is deactivated
phy0-ap0 is deactivated
phy1-ap0 is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
wwan0 is deactivated
wwan1 is in default mode

**root@OpenMPTCProuter:~# ip mptcp endpoint**
10.230.105.155 id 1 subflow fullmesh dev pcie_mhi0 
192.168.100.220 id 2 subflow fullmesh dev eth2 
10.220.52.107 id 3 subflow fullmesh dev wwan1

lhc1993, 2025-04-10:
**The result after plugging in eth2**
**root@OpenMPTCProuter:~# multipath**
br-lan is deactivated
erspan0 is deactivated
eth0 is deactivated
eth1 is deactivated
eth2 is in default mode
eth3 is deactivated
eth4 is deactivated
gre0 is deactivated
gretap0 is deactivated
ifb4eth1 is deactivated
ifb4eth2 is deactivated
ifb4eth3 is deactivated
ifb4eth4 is deactivated
ip6gre0 is deactivated
ip6tnl0 is deactivated
lo is deactivated
miireg is deactivated
pcie_mhi0 is in default mode
pcie_mhi1 is deactivated
phy0-ap0 is deactivated
phy1-ap0 is deactivated
sit0 is deactivated
teql0 is deactivated
tun0 is deactivated
wwan0 is deactivated
wwan1 is in default mode

**root@OpenMPTCProuter:~# ip mptcp endpoint**
10.230.105.155 id 1 subflow fullmesh dev pcie_mhi0
192.168.100.220 id 2 subflow fullmesh dev eth2
10.220.52.107 id 3 subflow fullmesh dev wwan1

Ysurac, 2025-04-11:
It's with latest snapshot ? What is the result of ```ip r``` ?

lhc1993, 2025-04-12:
root@OpenMPTCProuter:~# ip r
default via 10.255.252.1 dev tun0 
default via 192.168.100.1 dev eth2 metric 12 
default via 10.201.245.16 dev pcie_mhi0 metric 15 
default via 10.220.52.108 dev wwan1 metric 18 
10.201.245.0/27 dev pcie_mhi0 scope link metric 15 
10.201.245.16 dev pcie_mhi0 scope link metric 15 
10.220.52.104/29 dev wwan1 scope link metric 18 
10.220.52.108 dev wwan1 scope link metric 18 
10.255.252.0/24 dev tun0 proto kernel scope link src 10.255.252.2 
123.57.0.66 via 10.201.245.16 dev pcie_mhi0 metric 15 
123.57.0.66 via 10.220.52.108 dev wwan1 metric 18 
192.168.9.0/24 dev br-lan scope link metric 9999 
192.168.100.0/24 dev eth2 scope link metric 12 
192.168.100.1 dev eth2 scope link metric 12

lhc1993, 2025-04-12:
The same is true when using the latest version. When streaming, the newly added eth2 will not pass traffic.

Ysurac, 2025-04-12:
I would need ```ip r``` from snapshot, this one is not from latest snapshot.

lhc1993, 2025-04-12:
latest snapshot is v0.63-snapshot?

Ysurac, 2025-04-12:
Yes, from https://snapshots.openmptcprouter.com/6.6/x86_64/targets/x86/64/ latest snapshot is from 11 april for now.

lhc1993, 2025-04-12:
Is there a record of the changed PR submission?

Ysurac, 2025-04-12:
Here: https://github.com/Ysurac/openmptcprouter-feeds/commits/develop/ and here: https://github.com/Ysurac/openmptcprouter/commits/develop/

lhc1993, 2025-04-12:
I compiled it locally, how can I merge these new commits into my local ^_^

Ysurac, 2025-04-12:
Run ```build.sh``` script again, it should update all repos.

lhc1993, 2025-04-12:
I have some local driver adaptations and page changes, will they be overwritten?

lhc1993, 2025-04-12:
https://github.com/Ysurac/openmptcprouter/commits/develop/ This repository is the latest one locally, but https://github.com/Ysurac/openmptcprouter-feeds/commits/develop/ This repository has not been merged in, so it needs to be merged

Ysurac, 2025-04-12:
Develop branch is merged in master branch only for releases. Snapshots are based on develop branch.

lhc1993, 2025-05-14:
Need to study the principle here, still looking at the code.