Ysurac/openmptcprouter #3880: 0.62rc and 0.63 snapshot kernel crash R5S

hle5128, 2025-03-28:
## Expected Behavior

Expect to run normal

## Current Behavior

Crashed after restarting: kern.err kernel
## Possible Solution

has no problem 0.61

## Steps to Reproduce the Problem

  1. running on r5s
  2. fresh installation
  3. after entering vps booth up and crashed

## Context (Environment)
[openmptcprouter v0.63-snapshot-6.6 r0+28431-92e020b50f]
and 
0.62rc1 

## Specifications

  - OpenMPTCProuter version: 0.62rc1 and v0.63-snapshot-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: failed to run vps 0.62rc1 so run (wget -O - https://www.openmptcprouter.com/server-test/debian-x86_64.sh | KERNEL="6.12" sh)
  - OpenMPTCProuter VPS provider: buyvm
  - OpenMPTCProuter platform: r5s
  - Country: IP

https://pastebin.com/xEFzGgaQ

Ysurac, 2025-03-28:
Do you run another MPTCP scheduler than default ?

hle5128, 2025-03-28:
I tested the default as well rr

Ysurac, 2025-03-28:
Here the log is with a MPTCP Scheduler ? Can you give the kernel error with default scheduler ?

hle5128, 2025-03-28:
It looks like when we plug in 2nd eth, then it crashes after. If we remove the 2nd ETH cable and reboot, the crash stops, but no, I didnt change the motco scheduler. It was the default when it crashed

Ysurac, 2025-03-30:
Because the crash log was speaking about bpf and MPTCP. But if it's a problem with ethernet can be related to the driver. I will check this part.

closed 2025-07-03 by github-actions[bot]