Ysurac/openmptcprouter #3881: Router crash on heavy upload

kameleon25, 2025-03-29:
## Expected Behavior
The router should not randomly crash/reboot

## Current Behavior
I have latest OMR router x86_64 6.12 kernel snapshot image running in a proxmox environment. It connects to the VPS just fine and traffic flows. When I run a speedtest from speedtest.net the on prem OMR router vm crashes and reboots itself. When it does this all logs are lost from that session and internet connectivity is lost. It also crashes once more upon boot but stays up until whatever is triggering it happens again after the second boot. I can consistently make this happen in the test environment. However oddly enough speedof.me nor fast.com trigger this behavior. If I put a physical machine on the OMR LAN and do any uploads it seems to crash instantly also. The router vm had 2 cores and 2GB ram so I updated that to 4 cores and 8GB ram thinking it may be a resource issue but the problem persists. The host is proxmox running on an Intel NUC w/ i5 and NVMe backed storage so I do not believe it is resources however I do not have other equipment to test. Even though no logs can be obtained, I have collected some screen grabs (video) of the crashes if that would be of any assistance.

## Possible Solution
Honestly, no clue

## Steps to Reproduce the Problem

  1. Install latest snapshot and configure single or multiple WAN
  2. Try to run speedtest.net or another heavy upload
  3. Profit ...... err crash.

## Context (Environment)
## Specifications

  - OpenMPTCProuter version: v0.63-snapshot-6.12 r0+28753-f480fa60d3
  - OpenMPTCProuter VPS version: OpenMPTCProuter VPS 0.1032
  - OpenMPTCProuter VPS provider: Racknerd
  - OpenMPTCProuter platform: x86_64 vm
  - Country: USA

kameleon25, 2025-03-29:
*update*

I have added a third cellular connection and all seems to be stable so far. Although the third connection is not passing much traffic. I will continue to monitor this.

Ysurac, 2025-03-29:
I have also a crash problem with current kernel 6.12. I will update and see...

kameleon25, 2025-03-29:
@Ysurac is there an official or preferred way to downgrade to non-snapshot or an older snapshot for testing?

Ysurac, 2025-03-30:
Only use a kernel 6.6 even on snapshot. 6.12 is only a test.

kameleon25, 2025-03-30:
> Only use a kernel 6.6 even on snapshot. 6.12 is only a test.

I was using 6.12 following the wiki. I will roll back to 6.6. I am assuming it will require a complete reinstall of the router and vps, correct?

Ysurac, 2025-03-31:
There is no informations to use kernel 6.12 on router part. The 6.12 kernel can be used on VPS.
No need to reinstall anything, only an update and keep settings on router.

closed 2025-07-04 by github-actions[bot]