Ysurac/openmptcprouter #3882: The network connection is unstable. There is no other log when ss is disconnected. I don't know what happened.

lhc1993, 2025-03-29:
## Expected Behavior

Sat Mar 29 10:43:53 2025 authpriv.info dropbear[22169]: Child connection from 121.36.124.34:2930
Sat Mar 29 10:44:05 2025 authpriv.notice dropbear[22169]: Password auth succeeded for 'root' from 121.36.124.34:2930
Sat Mar 29 11:02:01 2025 daemon.info omr-tracker-ss: Shadowsocks rust sss0 is down (can't contact via http 198.27.92.1, 151.101.129.164, 77.88.55.77)
## Current Behavior

as above
## Possible Solution

no
## Steps to Reproduce the Problem

  1.
  2.
  3.

## Context (Environment)

## Specifications

  - OpenMPTCProuter version: v0.62rc1
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform: ipq8074
  - Country: cn

Ysurac, 2025-03-29:
Please update at least to a stable version.

lhc1993, 2025-03-29:
Does it mean to use v0.62, v0.61 without rc?

Ysurac, 2025-03-29:
Latest release os v0.62

lhc1993, 2025-03-30:
get, tks!

closed 2025-03-30 by lhc1993