Ysurac/openmptcprouter #3883: Multi IP troubles

VaKKuumDEV, 2025-03-29:
## Expected Behavior
I am trying to set up multiple white IP addresses as mentioned in the article [https://github.com/Ysurac/openmptcprouter/wiki/VPS-Multi-IPs](url). I have several computers on my network. I want all devices on the network to connect to the network from the first IP by default, and one of the computers to connect to the network from the second IP.

## Current Behavior
Here are my current settings.

[image]

[image]

[image]

[image]

[image]

[image]

I've read previous reports about similar issues. Here are also the VDS network settings.

[image]

Also, here is the output result of the command `netstat -laputen | grep ss-server`
[image]

For some reason that I don't really understand, the computer I need still accesses the Internet from the first IP, and not from the one I expect. I suspect that the problem is with configuring the network interface on the VDS side. I tried to set it up correctly, but it didn't work out.

## Specifications

  - OpenMPTCProuter version: 0.62-6.6
  - OpenMPTCProuter VPS version: 0.1032-6.12.15
  - OpenMPTCProuter VPS provider: FirstVDS
  - OpenMPTCProuter platform: RPI5

Ysurac, 2025-03-30:
The IP on the VPS part should be correctly configured first. You can check with a ```ping -B -I <ipyouwant> 1.1.1.1``` or with a traceroute.
But I will check if it's still working, I don't use this so possible it's now working anymore.

VaKKuumDEV, 2025-03-30:
I was trying to ping. It turns out that the computer goes online from the first IP and knocks on the second IP.

Ysurac, 2025-03-30:
Not sure to really understand what you mean by knocks. VPS IPs answer to ping at least from external ?
If not, they need to be configured on VPS following provider docs (nothing generic here).

VaKKuumDEV, 2025-03-30:
By "knocks" I mean sending packets. I want to understand if the VDS needs to be configured somehow in order for multiple IP to work. I saw that `shadowsocks-libev` creates one server. I was thinking about adding the `local_address` field (according to the Shadowsocks-libev documentation), but after some time, synchronization occurred, and the router rolled back my changes.

Ysurac, 2025-03-31:
I'm not sure to understand, if multi IPs are available and working on first VPS script install, shadowsocks-libev is configured on VPS and when server key is set on router it's also configured on router.

VaKKuumDEV, 2025-03-31:
Yes, I realized after a while. I first installed MPTCP on a VDS with a single IP. Later, I added another IP to the server. And the reconfiguration did not happen. I have now reinstalled the operating system on VDS and run the installation script. Today I'm going to the garage and try to set it up directly on the router. I'll write it off based on the results. It might be worth making a system reconfiguration tool.