Ysurac/openmptcprouter #3886: SYSTEME DE FICHIERS EN LECTURE SEULE

fabiofar27, 2025-03-31:
My configuration said
Systeme de fichiers en lecture seule

I tried the command e2fsck -y /dev/mmcblk0P2 and no results
whenn i use mount command i have this results
/dev/mmcblk0p2 on /rom type ext4 (ro,noatime)
devtmpfs on /rom/dev type devtmpfs (rw,relatime,size=1909832k,nr_inodes=477458,mode=755)
proc on /proc type proc (rw,nosuid,nodev,noexec,relatime)
sysfs on /sys type sysfs (rw,nosuid,nodev,noexec,relatime)
cgroup2 on /sys/fs/cgroup type cgroup2 (rw,nosuid,nodev,noexec,relatime,nsdelegate)
tmpfs on /tmp type tmpfs (rw,nosuid,nodev,noatime)
/dev/mmcblk0p2 on / type ext4 (rw,relatime)
/dev/mmcblk0p1 on /boot type vfat (rw,noatime,fmask=0022,dmask=0022,codepage=437,iocharset=iso8859-1,shortname=mixed,errors=remount-ro)
tmpfs on /dev type tmpfs (rw,nosuid,noexec,noatime,size=512k,mode=755)
devpts on /dev/pts type devpts (rw,nosuid,noexec,noatime,mode=600,ptmxmode=000)
debugfs on /sys/kernel/debug type debugfs (rw,nosuid,nodev,noexec,noatime)
bpffs on /sys/fs/bpf type bpf (rw,nosuid,nodev,noexec,noatime,mode=700)
mountd(pid5562) on /tmp/run/blockd type autofs (rw,relatime,fd=7,pgrp=5562,timeout=30,minproto=5,maxproto=5,indirect)

i moun remount /dev/mmcblk0p2 in rw
because in each case i have to restart using INTERFACE OPENMPTCPROUTER this is coming back to ro status

i changed the SD card cloning the previous using BALENA ETCHER but no changes
it seems /dev/mmcblck0p2 is on /rom and type ext4. is it normal?

this trouble cannot permit any changes or upgrade on my system

Do you have an idea please?
thanks in advance

## Specifications

  - OpenMPTCProuter version: <0.62RC1-6,6>
  - OpenMPTCProuter VPS version: <0.1032 6.6.36-x64v3-xanmod1>
  - OpenMPTCProuter VPS provider: OVH
  - OpenMPTCProuter platform:

Ysurac, 2025-04-01:
You didn't completed the template but I would say switch to squashfs image.

Ysurac, 2025-04-03:
And use release instead of RC1

fabiofar27, 2025-04-03:
Thank you Ysurac
I will try
Question: how can i charge my actual backup if i charge à squashfs image now?

Ysurac, 2025-04-04:
You can update image in System->Backup/Flash firmware and keep settings or import a config at same place.
Config is the same on ext or squashfs.

fabiofar27, 2025-04-11:
thanks @Ysurac  with the squashfs, it solved this topic. But still upgrading keeping my configurations....it seems there is a lot of things mixing. Now i have a good downloading but not uploading speed (i chek on my freebox and it doesn't correspond)
So this weekend i'll flash a new installation on my system and put new parameters .

Ysurac, 2025-04-11:
Thanks for feedback
If you have still issues on speed, open a new issue please.