Ysurac/openmptcprouter #3887: OOM occurs when using v2ray

lhc1993, 2025-03-31:
## Expected Behavior

Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.582422] v2ray invoked oom-killer: gfp_mask=0x140cca(GFP_HIGHUSER_MOVABLE|__GFP_COMP), order=0, oom_score_adj=0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.582484] CPU: 3 PID: 32384 Comm: v2ray Tainted: G        W  O       6.6.73 #0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.591680] Hardware name: YJ-H100
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.599228] Call trace:
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.603482]  dump_backtrace+0xa0/0xe0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.605735]  show_stack+0x18/0x24
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.609552]  dump_stack_lvl+0x48/0x60
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.612852]  dump_stack+0x18/0x24
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.616497]  dump_header+0x48/0x1b0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.619796]  oom_kill_process+0x16c/0x284
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.623096]  out_of_memory+0xc8/0x2f0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.627261]  __alloc_pages+0x90c/0xe04
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.630916]  __folio_alloc+0x14/0x20
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.634559]  __filemap_get_folio+0x104/0x2c8
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.638286]  filemap_fault+0x128/0x8f4
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.642538]  __do_fault+0x3c/0x114
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.646098]  __handle_mm_fault+0xa90/0x1500
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.649483]  handle_mm_fault+0x160/0x334
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.653564]  do_page_fault+0x1cc/0x498
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.657730]  do_translation_fault+0x88/0x9c
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.661290]  do_mem_abort+0x44/0x94
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.665369]  el0_da+0x24/0x60
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.668847]  el0t_64_sync_handler+0xb4/0x12c
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.671967]  el0t_64_sync+0x178/0x17c
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.676358] Mem-Info:
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867] active_anon:12296 inactive_anon:15117 isolated_anon:0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867]  active_file:0 inactive_file:590 isolated_file:0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867]  unevictable:0 dirty:0 writeback:0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867]  slab_reclaimable:2878 slab_unreclaimable:19502
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867]  mapped:134 shmem:273 pagetables:784
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867]  sec_pagetables:0 bounce:0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867]  kernel_misc_reclaimable:0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.679867]  free:3118 free_pcp:108 free_cma:0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.698148] Node 0 active_anon:644kB inactive_anon:109008kB active_file:0kB inactive_file:2360kB unevictable:0kB isolated(anon):0kB isolated(file):0kB mapped:536kB dirty:0kB writeback:0kB shmem:1092kB writeback_tmp:0kB kernel_stack:3552kB pagetables:3136kB sec_pagetables:0kB all_unreclaimable? yes
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.724584] DMA free:12244kB boost:0kB min:16384kB low:20480kB high:24576kB reserved_highatomic:0KB active_anon:109008kB inactive_anon:644kB active_file:1344kB inactive_file:588kB unevictable:0kB writepending:0kB present:1048576kB managed:882408kB mlocked:0kB bounce:0kB free_pcp:352kB local_pcp:0kB free_cma:0kB
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.752140] lowmem_reserve[]: 0 0 0 0
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.774254] DMA: 1801*4kB (UME) 360*8kB (UM) 62*16kB (UME) 0*32kB 0*64kB 0*128kB 0*256kB 0*512kB 0*1024kB 0*2048kB 0*4096kB = 11076kB
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.777800] 601 total pagecache pages
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.789773] 262144 pages RAM
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.793420] 0 pages HighMem/MovableOnly
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.796358] 41542 pages reserved
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.799917] Tasks state (memory values in pages):
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.803401] [  pid  ]   uid  tgid total_vm      rss pgtables_bytes swapents oom_score_adj name
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.808031] [   1519]    81  1519      379       32    36864        0             0 ubusd
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.816615] [   1629]     0  1629     1390      160    36864        0             0 rngd
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.824778] [   1702]     0  1702      272        0    28672        0             0 urngd
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.832932] [   5572]   514  5572      516       83    32768        0             0 logd
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.840988] [   5638]     0  5638     1089      256    36864        0             0 rpcd
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.849163] [   5902]     0  5902      283        0    36864        0             0 dropbear
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.857253] [   6696]     0  6696      713       32    32768        0             0 hostapd
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.865649] [   6697]     0  6697      713       32    32768        0             0 wpa_supplicant
Mon Mar 31 16:58:31 2025 kern.err kernel: [  393.872202] [E][mhi_qmap_rx_fixup] drop qmap unknow pkt, len=1916, pkt_len=31393
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.873722] [   6741]   101  6741     2308      256    49152        0             0 wpa_supplicant
Mon Mar 31 16:58:31 2025 kern.err kernel: [  393.882761] [E][mhi_qmap_rx_fixup] mhi_qmap_rx_fixup error
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.890293] [   6742]   101  6742     2498      352    53248        0             0 hostapd
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.904554] [   6762]     0  6762      702       96    32768        0             0 netifd
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.906517] ksoftirqd/1: page allocation failure: order:0, mode:0x820(GFP_ATOMIC), nodemask=(null)
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.912802] [   7026]     0  7026      665      128    32768        0             0 fibocom-dial-co
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.912811] [   7090]     0  7090      355        0    32768        0             0 crond
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.912817] [   7148]     0  7148     2470      160    49152        0             0 mqtt_c
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.921112] ,cpuset=
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.930086] [   7149]     0  7149      415        0    32768        0             0 xzchapo
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.939004] /
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.947327] [   7162]     0  7162      619       96    32768        0             0 yanjing.sh
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.955475] ,mems_allowed=0
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.957827] [   7174]     0  7174     2458      192    49152        0             0 yanjing
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  393.965890]
Mon Mar 31 16:58:31 2025 kern.info kernel: [  393.967637] [   7195]     0  7195      327       32    32768        0             0 fibocom-dial
[... 140 lines cut ...]
Mon Mar 31 16:58:31 2025 kern.err kernel: [  394.808702] [E][mhi_qmap_rx_fixup] netdev_alloc_skb_ip_align fail
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  394.816591] ,cpuset=/
Mon Mar 31 16:58:31 2025 kern.err kernel: [  394.822760] [E][mhi_qmap_rx_fixup] mhi_qmap_rx_fixup error
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  394.831261] ,mems_allowed=0
Mon Mar 31 16:58:31 2025 kern.err kernel: [  394.836745] [E][mhi_qmap_rx_fixup] netdev_alloc_skb_ip_align fail
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  394.844629] ,global_oom,task_memcg=/
Mon Mar 31 16:58:31 2025 kern.err kernel: [  394.850794] [E][mhi_qmap_rx_fixup] mhi_qmap_rx_fixup error
Mon Mar 31 16:58:31 2025 kern.warn kernel: [  394.859301] ,task=v2ray,pid=31927,uid=0
[... cut ...]
Mon Mar 31 16:58:31 2025 kern.err kernel: [  394.966181] [E][mhi_qmap_rx_fixup] mhi_qmap_rx_fixup error
Mon Mar 31 16:58:41 2025 daemon.info v2ray[27014]: V2Ray 5.26.0 (V2Fly, a community-driven edition of V2Ray.) OpenWrt (go1.23.4 linux/arm64)
Mon Mar 31 16:58:41 2025 daemon.info v2ray[27014]: A unified platform for anti-censorship.

## Current Behavior

## Specifications

  - OpenMPTCProuter version: v0.62
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider:
  - OpenMPTCProuter platform:

Ysurac, 2025-03-31:
I would say not enough memory.
What is the platform ?

lhc1993, 2025-03-31:
platform：ipq8074

lhc1993, 2025-04-01:
aw1000

Ysurac, 2025-04-03:
It as 1GB memory that is enough. Check memory usage with "htop" command via SSH on the router.
I can't help really more as it's not a supported device.

lhc1993, 2025-04-06:
tks

closed 2025-04-06 by lhc1993