Ysurac/openmptcprouter #3889: Thunderbird Emails not being sent with Hetzner VPN

3dzug, 2025-04-01:
## Expected Behavior

When I send an email with Thunderbird the mail should be send regardless which VPN I use.

## Current Behavior

When I use a Hetzner VPN the email is not sent and the transmission message is visible until it is failed. If I use a Avoro VPN it works fine. I made no changes other than chaning the VPS vrom Hetzner to Avoro.

## Possible Solution

Use a different VPN provider

## Steps to Reproduce the Problem

  1. Setup a Hetzner VPN
  2. Send a mail with Thunderbird.
 

## Context (Environment)

Hetzner VPN is far more stable than avoro.

## Specifications

  - OpenMPTCProuter version: [v0.61rc3-6.6 r0+27346-c7ba5574f5](https://github.com/ysurac/openmptcprouter)
  - OpenMPTCProuter VPS version: 
  - OpenMPTCProuter VPS provider: 0.1032.6.12.15-x64v3-xanmod1
  - OpenMPTCProuter platform:  x86_64
  - Country:  Germany
  - 
Please note as I am stuck in the v.62 bootloop I cant test newer versions.

Ysurac, 2025-04-01:
I think you mean VPS and yes some provider are blocking some ports. If they didn't change the rules, port 25 is blocked for 1 month for new user.
It's not related to OpenMPTCProuter.

3dzug, 2025-04-01:
Thanks,You are right.Matthias

I found this on Hetzner website: https://docs.hetzner.com/cloud/servers/faq/#why-can-i-not-send-any-mails-from-my-server

closed 2025-04-02 by 3dzug