Ysurac/openmptcprouter #3890: Problem with internet access

mhdalhelaly, 2025-04-01:
OMR works fine when I connect to it directly, but when I connect a mikrotik router internet slows down then gradually stops.

Ysurac, 2025-04-02:
Please use the good category and bugs template.

mhdalhelaly, 2025-04-02:
I have a MikroTik CCR1036 with 1000 users. When I switch the internet source to OpenMPTCPRouter, the internet becomes slow and frequently disconnects. However, when I use OpenMPTCPRouter directly without the MikroTik, the internet works perfectly.

mhdalhelaly, 2025-04-03:
up

mhdalhelaly, 2025-04-03:
I am using version v61 and kernel 5.4. Do you recommend a specific version?

Where can I find the error log template? Can you explain it to me? I am a beginner in this system.

Ysurac, 2025-04-03:
Please can you explain to me why you can't change issue category and use bug template ? There is a bug on Github I don't see ?

Ysurac, 2025-04-03:
It's in "New issue" and "Bug report" on Github, like you opened this issue.
I tried to put an explanation below each category, but maybe not enough clear as many are opening in wrong category or remove template. I will try to improve bug reports asap.

mhdalhelaly, 2025-04-05:
The problem was solved by changing the proxy from Shadowsocks-Rust 2022 to Shadowsocks, and the issue was resolved.

closed 2025-04-05 by mhdalhelaly