Ysurac/openmptcprouter #3893: Access from outside is coming and going

MRedZac, 2025-04-02:
## Expected Behavior
Access from outside to websites and locally hosted services possible and constantly available, once setup.

## Current Behavior
Access from outside is only available part time. Sometimes it is working fine, other times it is not possible, without anything changed on the network, the router or the vps. It is working for hours and all of a sudden, access is refused, websites are shown as down in the external monitoring tool uptime robot. Local hosted services are also not always down the same time as the websites are: Sometimes the websites which are hosted locally are down, but services like security cams on blueiris or vaultwarden server are still working fine, even though they are both routed through the same nginx instance on a docker host server. 

## Possible Solution
I have IPv6 in suspicion, as in the cloudflare dns there are entries for IPv6, but IPv6 is deactivated for the router, from what I understood...
I have deleted these entries from the cloudflare dns now. For now everything is working. Let's see for how long...

## Steps to Reproduce the Problem

1. basic installation of openmptcprouter. 
2. port forwarding for port 80 and 443 to internal ip 192.168.178.246

## Context (Environment)
This bug report is related to #3892
I have sucessfully up and running now the openmptcprouter and even managed to setup the firewall, so that websites behind the router can be addressed successfully, as well as some other services, I am hosting locally. I have a starlink, a FTTH and a 4G connection bonded together, all ticks are green.
Problem now is, that the access from outside is somehow non-permanent. Sometime (for hours) it is all working fine, and without changing anything on the network, the router or the vps, all of a sudden access from outside is refused, websites are not reachable, services are not available, even though outgoing traffic is still fine, internet access is fully available.
What are the possible points of failure here ? - What to look for ? - Is that a common issue ?

## Specifications

  - OpenMPTCProuter version: 0.62-6.6
  - OpenMPTCProuter VPS version: 0.1032 6.12 15-x64v3 xanmod1
  - OpenMPTCProuter VPS provider: IONOS
  - OpenMPTCProuter platform: x86_64
  - Country: Germany

Ysurac, 2025-04-02:
Thanks for fixing #3892 with correct template.
I would need router Status->log with timestamp when this happend and a check in ```/etc/shorewall/rules``` that entries are here and ```/etc/shorewall/params.vpn``` that vpn IP it the same as what you get with ```ip a show dev tun0``` (for OpenVPN as VPN) on the router via SSH.
And check also ```ip r``` (should have VPN as default on first place)

MRedZac, 2025-04-04:
[Log.txt](https://github.com/user-attachments/files/19598436/Log.txt)

Here is the log file, starting from 01.04.2025. The following dates and times (+/- 5 min) I am able to track where it changed from up to down or vice versa:

04.04.25, 02.54 UP
03.04.25, 22.33 DOWN
03.04.25, 19.31 UP
02.04.25, 21.17 DOWN
02.04.25, 11.12 UP
01.04.25, 21.35 DOWN
01.04.25, 05.13 UP

[image]

[image]

[image]

This morning it is again working, as you can see from the last entry at 02.54 am... 
I hope you are able to find, what I am desperately looking for... ;-)
Thanks in advance !

Cheers
Marco

Ysurac, 2025-04-04:
Sorry I don't download attached files, can you copy/paste ?

MRedZac, 2025-04-04:
OK, have to do this partially, as it is obviously too long to paste in once:

Tue Apr  1 00:05:54 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:09:40 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:11:33 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:12:29 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:12:57 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:13:00 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Tue Apr  1 00:13:00 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.33, lease time 1800
Tue Apr  1 00:17:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Tue Apr  1 00:17:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Tue Apr  1 00:17:27 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Tue Apr  1 00:17:27 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Tue Apr  1 00:28:00 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:35:32 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:39:17 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:41:10 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:42:06 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:42:34 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 00:42:37 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Tue Apr  1 00:42:38 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.33, lease time 1800
Tue Apr  1 00:47:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Tue Apr  1 00:47:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Tue Apr  1 00:47:27 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Tue Apr  1 00:47:27 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Tue Apr  1 00:57:38 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:05:09 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:08:55 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:10:48 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:11:44 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:12:12 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:12:15 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Tue Apr  1 01:12:16 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.33, lease time 1800
Tue Apr  1 01:16:23 2025 user.notice post-tracking-020-status: Check API configuration...
Tue Apr  1 01:16:23 2025 user.notice post-tracking-020-status: Check API configuration... Done
Tue Apr  1 01:17:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Tue Apr  1 01:17:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Tue Apr  1 01:17:27 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Tue Apr  1 01:17:27 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Tue Apr  1 01:27:16 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:34:47 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:38:33 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:40:26 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:41:22 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:41:50 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Tue Apr  1 01:41:53 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Tue Apr  1 01:41:53 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.33, lease time 1800
Tue Apr  1 01:47:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Tue Apr  1 01:47:23 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Tue Apr  1 01:47:27 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
[... cut ...]
Tue Apr  1 23:55:34 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Tue Apr  1 23:56:30 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Tue Apr  1 23:56:58 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Tue Apr  1 23:57:01 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Tue Apr  1 23:57:01 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.32, lease time 1800

MRedZac, 2025-04-04:
Wed Apr  2 00:12:01 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:16:58 2025 user.notice post-tracking-020-status: Check API configuration...
Wed Apr  2 00:16:58 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Apr  2 00:17:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Wed Apr  2 00:17:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Wed Apr  2 00:17:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Wed Apr  2 00:17:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Wed Apr  2 00:19:32 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:23:18 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:25:11 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:26:07 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:26:35 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:26:38 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Wed Apr  2 00:26:38 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.32, lease time 1800
Wed Apr  2 00:41:39 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:47:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Wed Apr  2 00:47:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Wed Apr  2 00:47:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Wed Apr  2 00:47:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Wed Apr  2 00:49:10 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:52:55 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:54:48 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:55:44 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:56:12 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 00:56:15 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Wed Apr  2 00:56:15 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.32, lease time 1800
Wed Apr  2 01:11:15 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:17:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Wed Apr  2 01:17:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Wed Apr  2 01:17:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Wed Apr  2 01:17:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Wed Apr  2 01:18:46 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:22:32 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:24:25 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:25:22 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:25:50 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:25:53 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Wed Apr  2 01:25:53 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.32, lease time 1800
Wed Apr  2 01:27:51 2025 user.notice post-tracking-020-status: Check API configuration...
Wed Apr  2 01:27:51 2025 user.notice post-tracking-020-status: Check API configuration... Done
Wed Apr  2 01:40:53 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:47:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: sending renew to server 192.168.1.1
Wed Apr  2 01:47:28 2025 daemon.notice netifd: wan1 (12613): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Wed Apr  2 01:47:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Wed Apr  2 01:47:33 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Wed Apr  2 01:48:24 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:52:10 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:54:03 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:54:59 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
Wed Apr  2 01:55:27 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.32
[... cut ...]
Wed Apr  2 23:47:37 2025 daemon.notice netifd: wan3 (7806): udhcpc: sending renew to server 192.168.51.1
Wed Apr  2 23:47:37 2025 daemon.notice netifd: wan3 (7806): udhcpc: lease of 192.168.51.178 obtained from 192.168.51.1, lease time 3600
Wed Apr  2 23:49:15 2025 daemon.notice netifd: wan1 (14592): udhcpc: sending renew to server 192.168.1.1
Wed Apr  2 23:49:15 2025 daemon.notice netifd: wan1 (14592): udhcpc: lease of 192.168.1.88 obtained from 192.168.1.1, lease time 3600
Wed Apr  2 23:55:50 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33

MRedZac, 2025-04-04:
Thu Apr  3 00:03:21 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Thu Apr  3 00:04:50 2025 user.notice post-tracking-020-status: Check API configuration...
Thu Apr  3 00:04:50 2025 user.notice post-tracking-020-status: Check API configuration... Done
Thu Apr  3 00:07:07 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Thu Apr  3 00:09:00 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Thu Apr  3 00:09:57 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Thu Apr  3 00:10:25 2025 daemon.notice netifd: wan2 (8347): udhcpc: sending renew to server 212.88.155.33
Thu Apr  3 00:10:28 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Thu Apr  3 00:10:30 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Thu Apr  3 00:10:34 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting renew
Thu Apr  3 00:10:36 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease lost, entering init state
Thu Apr  3 00:10:36 2025 daemon.notice netifd: Interface 'wan2' has lost the connection
Thu Apr  3 00:10:37 2025 daemon.notice ttyd[758]: [2025/04/03 00:10:37:0001] N: rops_handle_POLLIN_netlink: DELADDR
Thu Apr  3 00:10:40 2025 daemon.notice netifd: wan2 (8347): udhcpc: broadcasting select for 31.220.32.31, server 212.88.155.33
Thu Apr  3 00:10:40 2025 daemon.notice netifd: wan2 (8347): udhcpc: lease of 31.220.32.31 obtained from 212.88.155.33, lease time 1800
Thu Apr  3 00:10:40 2025 daemon.notice netifd: Interface 'wan2' is now up
Thu Apr  3 00:10:40 2025 user.notice firewall: Reloading firewall due to ifup of wan2 (eth4.132)
Thu Apr  3 00:10:41 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Apr  3 00:10:41 2025 user.notice mptcp: Reloading mptcp config due to ifup of wan2 (eth4.132)
Thu Apr  3 00:10:43 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Apr  3 00:10:43 2025 user.notice omr-bypass: Starting OMR-ByPass...
Thu Apr  3 00:10:51 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Apr  3 00:10:51 2025 user.notice omr-bypass: Restart dnsmasq...
Thu Apr  3 00:10:51 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Thu Apr  3 00:10:52 2025 user.notice omr-bypass: OMR-ByPass is running
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Thu Apr  3 00:10:52 2025 daemon.warn dnsmasq[1]: warning: interface omrvpn does not currently exist
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.178.100 -- 192.168.178.249, lease time 12h
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5355
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using only locally-known addresses for test
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using only locally-known addresses for onion
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using only locally-known addresses for localhost
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using only locally-known addresses for local
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using only locally-known addresses for invalid
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using only locally-known addresses for bind
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: using only locally-known addresses for lan
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: read /etc/hosts - 12 names
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq[1]: read /tmp/hosts/dhcp.cfg01411c - 4 names
Thu Apr  3 00:10:52 2025 daemon.info dnsmasq-dhcp[1]: read /etc/ethers - 0 addresses
Thu Apr  3 00:10:52 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Apr  3 00:10:56 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Apr  3 00:10:56 2025 user.notice omr-bypass: Starting OMR-ByPass...
Thu Apr  3 00:11:00 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Thu Apr  3 00:11:00 2025 user.notice omr-bypass: Restart dnsmasq...
Thu Apr  3 00:11:00 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Thu Apr  3 00:11:00 2025 user.notice omr-schedule-010-services: Can't resolve via dnsmasq, restart it...
Thu Apr  3 00:11:00 2025 user.notice omr-bypass: OMR-ByPass is running
Thu Apr  3 00:11:00 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Thu Apr  3 00:11:00 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth cryptohash DNSSEC no-ID loop-detect inotify dumpfile
Thu Apr  3 00:11:00 2025 daemon.info dnsmasq[1]: UBus support enabled: connected to system bus
Thu Apr  3 00:11:00 2025 daemon.warn dnsmasq[1]: warning: interface omrvpn does not currently exist
Thu Apr  3 00:11:00 2025 daemon.info dnsmasq-dhcp[1]: DHCP, IP range 192.168.178.100 -- 192.168.178.249, lease time 12h
Thu Apr  3 00:11:00 2025 daemon.info dnsmasq[1]: using nameserver 127.0.0.1#5355
[... cut ...]
Fri Apr  4 02:53:10 2025 user.notice firewall.omr-server: Firewall reload, set server part firewall reloading
Fri Apr  4 02:53:10 2025 user.notice omr-bypass: Restart dnsmasq...
Fri Apr  4 02:53:10 2025 daemon.info dnsmasq[1]: exiting on receipt of SIGTERM
Fri Apr  4 02:53:11 2025 daemon.info dnsmasq[1]: started, version 2.90 cachesize 1000
Fri Apr  4 02:53:11 2025 daemon.info dnsmasq[1]: compile time options: IPv6 GNU-getopt no-DBus UBus no-i18n no-IDN DHCP DHCPv6 no-Lua TFTP conntrack no-ipset nftset auth crypto

Ysurac, 2025-04-04:
I think I found where is the error, now only need to find why...
But in the log I see many disconnection from wan2, do you know why ?

MRedZac, 2025-04-04:
I am looking at this right at the moment. It looks like, WAN2 gets some kind of timeout while requesting a new dhcp lease from the provider. I am trying to find a pattern and for now it looks like mostly, that after 6 times recieving the lease successfully, the 7th time it doesn't work. But this is only 80% of the time like this. For the 20% rest, these times are shorter or longer. The provider only provides leases with 1800 seconds, so every half an hour, there's the possibility of disconnect. I am thinking already - as the IP on this internet line is fix - to fix the IP in settings for this interface, instead of DHCP... This might fix the problem of not recieving a new lease in time... But today, someone from the provider will come here to my home to check the physical fiber line... Let's wait, what he has to say...

MRedZac, 2025-04-07:
Can you explain, what you think is the issue ? - Situation is becoming critical for us these days...

Ysurac, 2025-04-07:
You can try latest snapshot but I don't think it's fully solved yet. I only need times to tests...

MRedZac, 2025-04-07:
OK, thank you. Will try tonight to install it. Does it need an Update on VPS as well ? - And if yes, how to do it there ?

Ysurac, 2025-04-07:
https://github.com/Ysurac/openmptcprouter/wiki/Snapshots

MRedZac, 2025-04-08:
OK, I would have tested latest snapshot, but already the Server script stopped running:

[image]

I hope, you'll find the time for solving & testing soon... 👍

Ysurac, 2025-04-08:
As indicated, You have no more space left on device.

MRedZac, 2025-04-11:
@Ysurac - OK, have setup another VPS with more space at IONOS and will also setup the router part today with Snapshot. But can you explain, where/what you think, the problem is ? Maybe that would enable me to find some kind of workaround until you had time for testing and fixing the issue...

Ysurac, 2025-04-18:
I'm not able to reproduce the connect issue on snapshot.

MRedZac, 2025-04-18:
So does that mean, that you believe the problem is fixed in the latest Snapshot ?

MRedZac, 2025-04-19:
OK, I finally was able to connect Router to VPS. But what about the initial problem of this threat now ? Any news on this ? - It doesn't make sense to waste time on trying to setup snapshots, if the initial but reported here hasn't been solved, yet, neither a concrete description of what is the problem has been given...

Ysurac, 2025-04-21:
To fix issues, first this need to be tested on latest snapshot (because lots of changes are made from release). I need new logs from the snapshot with disconnection timestamps.
If I know the problem, this would be solved...

closed 2025-07-25 by github-actions[bot]