Ysurac/openmptcprouter #3895: mt7615

jihadismail8, 2025-04-03:
Ok so I installed openmptcprouter on my SD card on banana pi br-4 , latest version 13-march, and I putted a wifi  banana mt7615 module into my mpcie slot .
Now the device is working, and I was able to create AP on 2.4 and 5ghz . But the tx power limit is 6db , which is extremely low . 
I did read in other places that there is patches and it's all related to eeprom file . Could you help please in creating a patch ? Or if there a guide on how to create one ? If it's too hard , which wifi module do you recommend as AP for the banana pi br4 ? Thanks mate

Ysurac, 2025-04-03:
BPI-R4 is already supported, so please can you create a bug issue and complete template ?

jihadismail8, 2025-04-03:
The issue is not the support of br4 , the issue is in supporting a driver for mt7615 wifi module

Ysurac, 2025-04-03:
Exactly, it's why I say it's not in good category (and need bug template).
And if you have also links about this issue this can help.

jihadismail8, 2025-05-06:
sorry for the long delay , but i was waiting for the issue to be resolved on openwrt , and seems now its resolved , could you help please in resolving it on openmptcprouter as well ? 
https://github.com/openwrt/mt76/pull/954