Ysurac/openmptcprouter #3898: Shadowsocks connectivity

falahati, 2025-04-05:
## Expected Behavior
Being able to connect to shadowsocks, with both the rust and libev 

## Current Behavior
It does not connect. One crashed and the other one failed to connect.

## Possible Solution
I do not have a solution

## Steps to Reproduce the Problem
  1. Trying to use shadowsocks proxy, rust or go

## Context (Environment)
There are a few issues that I have noticed:

### Shadowsocks 2022
When using this option, the log shows the following error:
```plain
omr-tracker-ss: Shadowsocks rust sss0 is down (can't contact via http 1.0.0.1, xxx.xxx.xxx.xxx, xxx.xxx.xxx.xxx)
```

### Shadowsocks
When using this option, the Shadowsocks service crashes:
```plain
Sat Apr  5 10:13:01 2025 daemon.info procd: Instance shadowsocks-rust::ss_local.tracker_sss0 s in a crash loop 6 crashes, 0 seconds since last crash
Sat Apr  5 10:13:01 2025 daemon.info procd: Instance shadowsocks-rust::ss_redir.hi1 s in a crash loop 6 crashes, 0 seconds since last crash
```

### API
I don't know if this is related to the issue, or just something nonimportant. But when calling the server status endpoint I get this:
```json
  "proxy": {
    "available": [
      "shadowsocks",
      "shadowsocks-go",
      "v2ray",
      "v2ray-vmess",
      "v2ray-socks",
      "v2ray-trojan",
      "xray",
      "xray-vless-reality",
      "xray-vmess",
      "xray-socks",
      "xray-trojan",
      "xray-shadowsocks"
    ],
    "current": "shadowsocks-rust"
  },
```

This is for Shadowsocks 2022, as you can see the current proxy is set to `shadowsocks-rust`, but this option is not available in the list above. when switching to Shadowsocks, the value becomes `shadowsocks`. In other words, it is as if we can not select `shadownsocks-go` option at all.

## Specifications

  - OpenMPTCProuter version: v0.62-6.6 r0+28431-92e020b50f
  - OpenMPTCProuter VPS version: 0.1032
  - OpenMPTCProuter VPS provider: Two separate operators, both local, different hardware, different datacenter
  - OpenMPTCProuter platform: x86_64
  - Country: Not related to country

Ysurac, 2025-04-05:
Shadowsocks Rust is using shadowsocks 2022 protocol on shadowsocks go server.
Here I would say it doesn't connect to shadowsocks server at all.
In most case it's wrong key or provider server.
Can you check shadowsocks-go log on VPS to see if it try to connect or not at all ? ```journalctl -u shadowsocks-go```

vecjh, 2025-04-26:
Hi there, I confirm I have the same problem on the latest install

```
Sat Apr 26 17:26:12 2025 daemon.info procd: Instance shadowsocks-rust::ss_local.tracker_sss0 s in a crash loop 6 crashes, 1 seconds since last crash
Sat Apr 26 17:26:12 2025 daemon.info procd: Instance shadowsocks-rust::ss_redir.hi1 s in a crash loop 6 crashes, 1 seconds since last crash
```

No issues on the WPS server:
```
sudo journalctl -u shadowsocks-go | fold -w 80
Apr 26 10:03:31 omp-server systemd[1]: Stopped shadowsocks-go.service - shadowso
cks-go Service.
Apr 26 10:03:31 omp-server systemd[1]: Started shadowsocks-go.service - shadowso
cks-go Service.
Apr 26 10:03:38 omp-server systemd[1]: Stopping shadowsocks-go.service - shadows
ocks-go Service...
Apr 26 10:03:38 omp-server systemd[1]: shadowsocks-go.service: Deactivated succe
ssfully.
Apr 26 10:03:38 omp-server systemd[1]: Stopped shadowsocks-go.service - shadowso
cks-go Service.
Apr 26 10:03:38 omp-server systemd[1]: Started shadowsocks-go.service - shadowso
cks-go Service.
```

[image]

vecjh, 2025-04-26:
ok, it seems like a figured it out. Looks like I can't use libev and rust at the same time. if enable one at a time then all works. Not sure if it's intended to work that way

Ysurac, 2025-04-26:
Yes you can't use both at the same time. Why would you ?