Ysurac/openmptcprouter #3900: A difference in the size of incoming and outgoing data.

mhdalhelaly, 2025-04-05:
The data input into the OpenMPTCPRouter system from the internet service provider is 100 GB, while the data output is only 50 GB. There is a significant difference. I want a solution to this problem.

Ysurac, 2025-04-05:
I don't know to understand where is the issue.

mhdalhelaly, 2025-04-05:
The incoming traffic from the service provider is much higher than the outgoing traffic from the OpenMPTCPRouter system — there is a data leak.

Ysurac, 2025-04-06:
It seems normal that we receive more data than what we send. Why this would be equal ?

mhdalhelaly, 2025-04-06:
The incoming traffic from the WAN port, whether download or upload, is higher than the outgoing traffic from the LAN port.

Ysurac, 2025-04-06:
It's not the same, but there is ping sent via the router to check WAN status. Configurable in Services->OMR-Tracker Manager.
There is also a check every hours to VPS API.
Also VPN & Proxy, due to encryption and protocol used, also use some traffic.

mhdalhelaly, 2025-04-06:
[image]
[image]

No, it's not a ping being sent. It seems there's a leak or a process that's continuously sending packets. Look at the images and see the big difference between the incoming and outgoing ports — something is not normal.

Ysurac, 2025-04-06:
I don't from where the image come, this doesn't help to identify a problem. In network->MPTCP bandwidth there is a graph that show current usage, and in Status->Traffic Monitor you can add LAN interface clear data and this will give you all traffic (I think it's best way to check where can be the problem).
And as it's seems to be lot of data, you can check Status->Realtime graphs, Connections, first entry should be the proxy or VPN and it's normal.

mhdalhelaly, 2025-04-07:
I think there is an error in installing the system on the computer. The version of OpenMPTCPRouter that I burned onto the flash drive has a .vhdx extension, and I ran it on the computer, and it worked. Is the poor performance due to this?

Ysurac, 2025-04-07:
This mean it's a VM in Hyper-V. I don't really test this hypervisor, I can't say anything about it. But in a hypervisor you need one interface for each WAN it's better, and this need enough memory and vCPU allocated.

mhdalhelaly, 2025-04-07:
I have a separate computer for the OMR system. I tried burning the version with the extension .img.gz to a USB flash drive and booted it, but after booting, the system stops and doesn't work. I then tried burning the version with the extension .vhdx to the USB flash drive, and it worked on the computer. I thought everything was fine, but I'm not sure if I made a big mistake. I will attach a picture of the installation process for the .img.gz version and where it stops.

[image]

mhdalhelaly, 2025-04-07:
The problem was solved and the IMG version is now running. The issue turned out to be with the USB flash drive — I replaced it, and it worked.

closed 2025-07-11 by github-actions[bot]